openwrt/packages #12805: ImageMagick: Compilation failed with glibc GCC 10

DazzyWalkman, 2020-07-13:
Maintainer: @val-kulkov
 
package version: 7.0.9-5, x86-64, snapshot r13828-cef9e5a49f

In fact I am not quite sure it's related to GCC 10, however, nothing changed except I tried switching to the lastest GCC 10 toolchain.

Seemed like there's one step that failed silently, making an empty part between "|"
in command `libtool: link: x86_64-openwrt-linux-gnu-gcc-nm  coders/.libs/aai_la-aai.o   |  | /opt/openwrt/staging_dir/host/bin/sed 's/.* //' | sort | uniq > coders/.libs/aai.exp`
, thus throwing out an error.

I haven't got hold of a previously successful copy of buildlog to see what actually is.

Bumping the IM version to 7.0.10-23 doesn't help, either.

Here is the related part of the faillog:
```
make  all-am

make[4]: Entering directory '/opt/openwrt/build_dir/target-x86_64_glibc/ImageMagick-7.0.9-5'
/bin/bash ./libtool  --tag=CC   --mode=link x86_64-openwrt-linux-gnu-gcc  -I/opt/openwrt/staging_dir/target-x86_64_glibc/usr/include/libpng16 -I/opt/openwrt/staging_dir/target-x86_64_glibc/usr/include  -I/opt/openwrt/staging_dir/target-x86_64_glibc/usr/include/freetype2 -I/opt/openwrt/staging_dir/target-x86_64_glibc/usr/include -I/opt/openwrt/staging_dir/target-x86_64_glibc/usr/include/libpng16  -I/opt/openwrt/staging_dir/target-x86_64_glibc/usr/include    -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -ffile-prefix-map=/opt/openwrt/build_dir/target-x86_64_glibc/ImageMagick-7.0.9-5=ImageMagick-7.0.9-5 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -flto  -Wall -fexceptions -pthread -DMAGICKCORE_HDRI_ENABLE=0 -DMAGICKCORE_QUANTUM_DEPTH=8 -no-undefined -export-symbols-regex ".*" -shared -module -avoid-version -L/opt/openwrt/staging_dir/target-x86_64_glibc/usr/lib -L/opt/openwrt/staging_dir/target-x86_64_glibc/lib -L/opt/openwrt/staging_dir/toolchain-x86_64_gcc-10.1.0_glibc/usr/lib -L/opt/openwrt/staging_dir/toolchain-x86_64_gcc-10.1.0_glibc/lib -znow -zrelro  -o coders/aai.la -rpath /usr/lib/ImageMagick-7.0.9/modules-Q8/coders coders/aai_la-aai.lo MagickCore/libMagickCore-7.Q8.la

libtool: link: x86_64-openwrt-linux-gnu-gcc-nm  coders/.libs/aai_la-aai.o   |  | /opt/openwrt/staging_dir/host/bin/sed 's/.* //' | sort | uniq > coders/.libs/aai.exp

./libtool: eval: line 1737: syntax error near unexpected token `|'

./libtool: eval: line 1737: `x86_64-openwrt-linux-gnu-gcc-nm  coders/.libs/aai_la-aai.o   |  | /opt/openwrt/staging_dir/host/bin/sed 's/.* //' | sort | uniq > coders/.libs/aai.exp'

make[4]: *** [Makefile:7022: coders/aai.la] Error 2

make[4]: Leaving directory '/opt/openwrt/build_dir/target-x86_64_glibc/ImageMagick-7.0.9-5'

make[3]: *** [Makefile:5990: all] Error 2`

```

val-kulkov, 2020-07-13:
There was a recent change in libfreetype in #12633 that might or might not have broken IM. I have not yet had a chance to investigate whether #12633 breaks IM compilation.

Other than that, no idea what might be causing the problem. I'll try to find time to see what's going on in the near future.

neheb, 2020-07-13:
A bunch of autotools packages fail in this way.  No idea why.

DazzyWalkman, 2020-07-13:
I looked at the official repo in hope of finding a successful or failing buildlog of IM, but unable to find one. Nevertheless, seeing that the official buildbot is generating IM x86-64 package just fine, I presume ongoing shellcheck cleanups have no negative impacts on the IM building. 
Maybe I'll look at my building host,  Ubuntu 20.04, x64, for unwelcome changes.

DazzyWalkman, 2020-07-15:
I switched back to gilbc gcc 9.30, and tried clean build IM again. Compilation was successful.

I diff 2 build logs side by side, find that aside from nelegible time stamp differences, one line is different prior to the failure and I think it might be the contributing factor:

`checking command to parse x86_64-openwrt-linux-gnu-gcc-nm output from x86_64-openwrt-linux-gnu-gcc object`

The result of GCC 10 toolchain is 'failed', while the GCC 9 one is 'ok'

Regrettably, my current limited time and knowledge prevent me from digging deeper.

Hopefully someone could shed some lights on this, or provide some pointers for me about where to look at.

Thanks.

DazzyWalkman, 2020-09-22:
Strangely, I clean build , including the toolchain today with gcc 10.2.0, glibc and binutil 2.35, without the disable-flto patch, the issue had gone. Compilation completed successful. 
Thank you all anyway. I am closing this.

closed 2020-09-22 by DazzyWalkman