openwrt/packages #12806: mwan3: basename help prompt being displayed in console with mwan3track

jamesmacwhite, 2020-07-13:
Maintainer: @feckert 
Environment: 19.07.3 mwan3_2.8.8-1
Description:

When using commands line `mwan3 restart`, the following is being output in console. I believe this is coming from `mwan3track` and occurring for each interface being tracked

```
BusyBox v1.30.1 () multi-call binary.

Usage: basename FILE [SUFFIX]

Strip directory path and .SUFFIX from FILE
uci: Entry not found
uci: Entry not found
```

This seems to have started happening either on 2.8.7 or 2.8.8, I don't remember it occurring on 2.8.6.

feckert, 2020-07-14:
This can actually only come from these two lines.

https://github.com/openwrt/packages/blob/master/net/mwan3/files/usr/sbin/mwan3rtmon#L6
https://github.com/openwrt/packages/blob/master/net/mwan3/files/usr/sbin/mwan3track#L6

I can't reproduce this on my system.

The following outputs
```
uci: Entry not found
uci: Entry not found
```

maybe comes from this lines
https://github.com/openwrt/packages/blob/master/net/mwan3/files/lib/mwan3/mwan3.sh#L48
https://github.com/openwrt/packages/blob/master/net/mwan3/files/lib/mwan3/mwan3.sh#L60

if you execude them do you get the same output -> Entry not found? (You have to replace idx with the valid entry number in the config file)

But I see the following recurring output in the terminal when `mwan3 restart` is executed?
`Dump terminated`

jamesmacwhite, 2020-07-14:
Thanks for your reply @feckert.

It would appear the basename console output is from the `LOG="logger -t $(basename "$0")[$$] -p"` line. 

It appears to be the idx value of 8 that throws the Entry not found errors, all others 1-7 return valid responses.

How do I find what the idx value of 8 is in relation to?

feckert, 2020-07-14:
> It would appear the basename console output is from the `LOG="logger -t $(basename "$0")[$$] -p"` line.

That's what I wrote.
But the question is why?
I don't get this message.
Maybe you could output the $0 value because this is empty.
I'd like to know why is this value not defined on your device.

> It appears to be the idx value of 8 that throws the Entry not found errors, all others 1-7 return valid responses.

I think the values **enabled** and **family** are not defined in your config section 8?
If I remove the family option from my config, I get also the following output `uci: Entry not found`.

jamesmacwhite, 2020-07-14:
Regarding the basename issue. Echoing $0 shows both paths `/usr/sbin/mwan3track` and `/usr/sbin/mwan3rtmon`, so it is there. So not sure why that's happening, unless it isn't coming from the LOG variable at all.

All my interfaces have both family and enabled set, or does this also apply to rules as well?

```
config interface 'wan'
    option enabled '1'
    list track_ip '8.8.8.8'
    list track_ip '8.8.4.4'
    list track_ip '208.67.222.222'
    list track_ip '208.67.220.220'
    option family 'ipv4'
    option reliability '2'
    option count '1'
    option timeout '2'
    option failure_latency '1000'
    option recovery_latency '500'
    option failure_loss '20'
    option recovery_loss '5'
    option interval '5'
    option down '3'
    option up '8'

config interface 'wan6'
    option enabled '1'
    list track_ip '2001:4860:4860::8888'
    list track_ip '2001:4860:4860::8844'
    list track_ip '2620:0:ccc::2'
    list track_ip '2620:0:ccd::2'
    option family 'ipv6'
    option reliability '2'
    option count '1'
    option timeout '2'
    option interval '5'
    option down '3'
    option up '8'

config interface 'wanb'
    option enabled '1'
    list track_ip '8.8.8.8'
    list track_ip '8.8.4.4'
    list track_ip '208.67.222.222'
    list track_ip '208.67.220.220'
    option family 'ipv4'
    option reliability '1'
    option count '1'
    option timeout '2'
    option failure_latency '1000'
    option recovery_latency '500'
    option failure_loss '20'
    option recovery_loss '5'
    option interval '5'
    option down '3'
    option up '8'

config interface 'wanb6'
    option enabled '1'
    list track_ip '2001:4860:4860::8888'
    list track_ip '2001:4860:4860::8844'
    list track_ip '2620:0:ccc::2'
    list track_ip '2620:0:ccd::2'
    option family 'ipv6'
    option reliability '1'
    option count '1'
    option timeout '2'
    option interval '5'
    option down '3'
    option up '8'
    
config interface 'wg'
    option enabled '1'
    list track_ip '8.8.8.8'
    list track_ip '8.8.4.4'
    option family 'ipv4'
    option reliability '1'
    option count '1'
    option timeout '2'
    option interval '5'
    option down '3'
    option up '8'

config interface 'wg6'
    option enabled '1'
    list track_ip '2001:4860:4860::8888'
    list track_ip '2001:4860:4860::8844'
    list track_ip '2620:0:ccc::2'
    list track_ip '2620:0:ccd::2'
    option family 'ipv6'
    option reliability '1'
    option count '1'
    option timeout '2'
    option interval '5'
    option down '3'
    option up '8'

config interface 'wgb'
    option enabled '1'
    list track_ip '8.8.8.8'
    list track_ip '8.8.4.4'
    option family 'ipv4'
    option reliability '1'
    option count '1'
    option timeout '2'
    option interval '5'
    option down '3'
    option up '8'

config interface 'wgb6'
    option enabled '1'
    list track_ip '2001:4860:4860::8888'
    list track_ip '2001:4860:4860::8844'
    list track_ip '2620:0:ccc::2'
    list track_ip '2620:0:ccd::2'
    option family 'ipv6'
    option reliability '1'
    option count '1'
    option timeout '2'
    option interval '5'
    option down '3'
    option up '8'
```

I don't have a family option defined on the default rules here:

```
config rule 'https'
    option sticky '1'
    option dest_port '443'
    option proto 'tcp'
    option use_policy 'wan_wanb'

config rule 'default_rule'
    option dest_ip '0.0.0.0/0'
    option proto 'all'
    option use_policy 'wan_wanb'
```

Is this the problem? These are the same as the default config that ships, with the exception of changing the policy from balanced.

aaronjg, 2020-07-14:
The default rule should have a family of 'ipv4' since the dest_ip is 0.0.0.0/0. This looks like a bug in the mwan3 default config.

You can add "set -x" to the start of the mwan3.sh or mwan3track scripts to have it write all the commands to the console. Be prepared for a lot of output, but it will help pinpoint exactly what statement is causing the issue.

jamesmacwhite, 2020-07-15:
@aaronjg I think it has been like that for some time and I haven't got an error on that rule before. I don't think the uci errors are related to it either, looks to be something else. It was my understanding that the family value if not specified on rules defaults to `all` anyway. It seems to be doing that based on the iptables output and covers both IPv4 and IPv6, even though the default rule is using a IPv4 0.0.0.0/0 mask.

```
 1750  102K S https  tcp  --  *      *       0.0.0.0/0            0.0.0.0/0            multiport dports 443
32543 5023K - wan_wanb  all  --  *      *       0.0.0.0/0            0.0.0.0/0
```

```
6469  514K S https  tcp      *      *       ::/0                 ::/0                 multiport dports 443
 1472  787K - wan_wanb  all      *      *       ::/0                 ::/0
```

aaronjg, 2020-07-15:
If no ip is specified, then it will default to 0.0.0.0/0 for the ipv4 and ::/0 for ipv6. If the IP is specified, it will create the entry for ipv4 then silently fail for ipv6. 

It looks like that is just the partial output for the https rule, no? Is there another output for the default_rule in the ipv4 table?

anyway, I don't think think this is the issue - the ip6tables command will just fail, but not cause any further issues in the script. I just pointed it out because you asked about the family option...

Did `set -x` turn up anything on the `basename` issue?

jamesmacwhite, 2020-07-15:
No worries. Thanks for clarifying.

The basename issue is weird but seems a minor issue, however the uci errors are more my focus currently, I believe I have some explanation as to why they are happening, I found this from the debug output:

```
+ uci get 'mwan3.@interface[8]'
+ rm -f /tmp/mwan3rtmon/ipv4.1 /tmp/mwan3rtmon/ipv4.3 /tmp/mwan3rtmon/ipv4.main
+ return 0
+ ret=0
+ mwan3_unlock service mwan3rtmon
+ lock -u /var/run/mwan3.lock
+ '[' 0 '=' 0 ]
+ '[' 5 '=' 0 ]
+ wait
+ sleep 5
BusyBox v1.30.1 () multi-call binary.

Usage: basename FILE [SUFFIX]

Strip directory path and .SUFFIX from FILE
uci: Entry not found
uci: Entry not found
```

Calling `uci get 'mwan3.@interface[8]'` does indeed throw a `uci: Entry not found` error, but I don't know where it is getting the value of 8 from. I do have 8 interfaces configured in my mwan3 config, but from an array point of view, it looks like the valid values are 0-7.

feckert, 2020-07-15:
Maybe the whole idx calculation is wrong https://github.com/openwrt/packages/blob/335efed8cb56bcc2c140efcb42a3b70fb4b4a14c/net/mwan3/files/lib/mwan3/mwan3.sh#L46-L61
The way I see it right now, it looks weird.
I think we start there with an 1 and not with an 0 that`s wrong!
The line idx=$((idx+1)) must go to the end of the funktion?

feckert, 2020-07-15:
[0001-mwan3-fix-idx-calculation.txt](https://github.com/openwrt/packages/files/4924349/0001-mwan3-fix-idx-calculation.txt)
Does this fix your issue. I did not a run test.

jamesmacwhite, 2020-07-15:
@feckert Thank you for looking into it. I tried to apply that patch but testing with --dry-run shows it will fail at two points.

```
root@linksys-wrt3200acm:/lib/mwan3# patch --dry-run < /tmp/0001-mwan3-fix-idx-calculation.txt
checking file mwan3.sh
Hunk #1 FAILED at 38.
Hunk #2 succeeded at 57 (offset -2 lines).
Hunk #3 FAILED at 72.
Hunk #4 succeeded at 91 (offset -2 lines).
2 out of 4 hunks FAILED
```

aaronjg, 2020-07-15:
> Calling uci get 'mwan3.@interface[8]' does indeed throw a uci: Entry not found error, but I don't know where it is getting the value of 8 from. I do have 8 interfaces configured in my mwan3 config, but from an array point of view, it looks like the valid values are 0-7.

It should go one past, and then fail to get a uci entry and return. Not sure why it is outputting to the console though.

The line in the script is actually has an output redirect in it:
https://github.com/openwrt/packages/blob/335efed8cb56bcc2c140efcb42a3b70fb4b4a14c/net/mwan3/files/lib/mwan3/mwan3.sh#L45

Does `uci get 'mwan3.@interface[8]' >/dev/null 2>&1` have any output?

I agree with @feckert  that the the way it is currently doing the indexing is weird, though.

feckert, 2020-07-15:
> The line in the script is actually has an output redirect in it:

That only redirects the stdout and not the stderr
> 
> https://github.com/openwrt/packages/blob/335efed8cb56bcc2c140efcb42a3b70fb4b4a14c/net/mwan3/files/lib/mwan3/mwan3.sh#L45
> 
> Does `uci get 'mwan3.@interface[8]' >/dev/null 2>&1` have any output?

No this does not have any output alle is redirected (stdout and stderr) to /dev/null

> I agree with @feckert that the the way it is currently doing the indexing is weird, though.

I think the problem with my patch file is that it is based on the last merged P/R from this day morning.
Please rebase your working copy and then try to apply my patch again.

jamesmacwhite, 2020-07-15:
@feckert Thanks. I have patched mwan3.sh now and tested it. I didn't fix the error and broke other things by the looks of it. Interesting there is some ip6tables error output.

```
BusyBox v1.30.1 () multi-call binary.

Usage: basename FILE [SUFFIX]

Strip directory path and .SUFFIX from FILE
uci: Entry not found
uci: Entry not found
BusyBox v1.30.1 () multi-call binary.

Usage: basename FILE [SUFFIX]

Strip directory path and .SUFFIX from FILE
uci: Entry not found
uci: Entry not found
Error: Invalid gateway address.
uci: Invalid argument
uci: Invalid argument
uci: Invalid argument
uci: Invalid argument
uci: Invalid argument
uci: Invalid argument
uci: Invalid argument
uci: Invalid argument
BusyBox v1.30.1 () multi-call binary.

Usage: basename FILE [SUFFIX]

Strip directory path and .SUFFIX from FILE
uci: Entry not found
uci: Entry not found
BusyBox v1.30.1 () multi-call binary.

Usage: basename FILE [SUFFIX]

Strip directory path and .SUFFIX from FILE
uci: Entry not found
uci: Entry not found
BusyBox v1.30.1 () multi-call binary.

Usage: basename FILE [SUFFIX]

Strip directory path and .SUFFIX from FILE
uci: Entry not found
uci: Entry not found
BusyBox v1.30.1 () multi-call binary.

Usage: basename FILE [SUFFIX]

Strip directory path and .SUFFIX from FILE
uci: Entry not found
uci: Entry not found
uci: Invalid argument
uci: Invalid argument
uci: Invalid argument
uci: Invalid argument
uci: Invalid argument
uci: Invalid argument
uci: Invalid argument
uci: Invalid argument
BusyBox v1.30.1 () multi-call binary.

Usage: basename FILE [SUFFIX]

Strip directory path and .SUFFIX from FILE
uci: Entry not found
uci: Entry not found
BusyBox v1.30.1 () multi-call binary.

Usage: basename FILE [SUFFIX]

Strip directory path and .SUFFIX from FILE
uci: Entry not found
uci: Entry not found
BusyBox v1.30.1 () multi-call binary.

Usage: basename FILE [SUFFIX]

Strip directory path and .SUFFIX from FILE
uci: Entry not found
uci: Entry not found
BusyBox v1.30.1 () multi-call binary.

Usage: basename FILE [SUFFIX]

Strip directory path and .SUFFIX from FILE
uci: Entry not found
uci: Entry not found
Error: Invalid gateway address.
uci: Invalid argument
uci: Invalid argument
uci: Invalid argument
uci: Invalid argument
uci: Invalid argument
uci: Invalid argument
uci: Invalid argument
uci: Invalid argument
BusyBox v1.30.1 () multi-call binary.

Usage: basename FILE [SUFFIX]

Strip directory path and .SUFFIX from FILE
uci: Entry not found
uci: Entry not found
BusyBox v1.30.1 () multi-call binary.

Usage: basename FILE [SUFFIX]

Strip directory path and .SUFFIX from FILE
uci: Entry not found
uci: Entry not found
BusyBox v1.30.1 () multi-call binary.

Usage: basename FILE [SUFFIX]

Strip directory path and .SUFFIX from FILE
uci: Entry not found
uci: Entry not found
iptables v1.8.3 (legacy): Invalid match name " statistic --mode random --probability 0.500" (28 chars max)
Try `iptables -h' or 'iptables --help' for more information.
BusyBox v1.30.1 () multi-call binary.

Usage: basename FILE [SUFFIX]

Strip directory path and .SUFFIX from FILE
uci: Entry not found
uci: Entry not found
Error: Invalid gateway address.
uci: Invalid argument
uci: Invalid argument
uci: Invalid argument
uci: Invalid argument
uci: Invalid argument
uci: Invalid argument
uci: Invalid argument
uci: Invalid argument
BusyBox v1.30.1 () multi-call binary.

Usage: basename FILE [SUFFIX]

Strip directory path and .SUFFIX from FILE
uci: Entry not found
uci: Entry not found
iptables v1.8.3 (legacy): Invalid match name " statistic --mode random --probability 0.500" (28 chars max)
Try `iptables -h' or 'iptables --help' for more information.
ip6tables v1.8.3 (legacy): Invalid match name " statistic --mode random --probability 0.500" (28 chars max)
Try `ip6tables -h' or 'ip6tables --help' for more information.
BusyBox v1.30.1 () multi-call binary.

Usage: basename FILE [SUFFIX]

Strip directory path and .SUFFIX from FILE
uci: Entry not found
uci: Entry not found
root@linksys-wrt3200acm:/lib/mwan3# iptables v1.8.3 (legacy): Invalid match name " statistic --mode random --probability 0.500" (28 chars max)
Try `iptables -h' or 'iptables --help' for more information.
ip6tables v1.8.3 (legacy): Invalid match name " statistic --mode random --probability 0.500" (28 chars max)
Try `ip6tables -h' or 'ip6tables --help' for more information.
BusyBox v1.30.1 () multi-call binary.

Usage: basename FILE [SUFFIX]

Strip directory path and .SUFFIX from FILE
uci: Entry not found
uci: Entry not found
BusyBox v1.30.1 () multi-call binary.

Usage: basename FILE [SUFFIX]

Strip directory path and .SUFFIX from FILE
uci: Entry not found
uci: Entry not found
```

jamesmacwhite, 2020-07-15:
I even reverted to the default opkg config and ran `mwan3 restart` just to make sure I'm not going insane, I still get the same error output, but only twice. I guess because the default config only has one interface enabled by default.

Why this this so horribly broken all of a sudden specifically for my router. Seems bizarre. I've completely uninstalled and reinstalled mwan3 to make sure none of the files have been modified while I've been debugging and left something in.

```
root@linksys-wrt3200acm:/lib/mwan3# mwan3 restart
BusyBox v1.30.1 () multi-call binary.

Usage: basename FILE [SUFFIX]

Strip directory path and .SUFFIX from FILE
uci: Entry not found
uci: Entry not found
BusyBox v1.30.1 () multi-call binary.

Usage: basename FILE [SUFFIX]

Strip directory path and .SUFFIX from FILE
uci: Entry not found
uci: Entry not found
```

aaronjg, 2020-07-15:
> That (ie https://github.com/openwrt/packages/blob/335efed8cb56bcc2c140efcb42a3b70fb4b4a14c/net/mwan3/files/lib/mwan3/mwan3.sh#L45 ) only redirects the stdout and not the stderr

> No this (ie `uci get 'mwan3.@interface[8]' >/dev/null 2>&1`)  does not have any output alle is redirected (stdout and stderr) to /dev/null

I'm missing something - what's the difference here other than the $idx substitution?

aaronjg, 2020-07-15:
> Seems bizarre. I've completely uninstalled and reinstalled mwan3 to make sure none of the files have been modified while I've been debugging and left something in.

Yeah, does seem very strange, but I'm sure we can get to the bottom of this. The basename thing also seems weird. I wonder if these are related?

Can you add the echo commands to the rtmon script? Maybe that is where the basename issue is coming from. If that doesn't turn anything up, could you do the `set -x` diagnostics on this script as well?

feckert, 2020-07-15:
I found the problem. 
The line https://github.com/openwrt/packages/blob/master/net/mwan3/files/lib/mwan3/mwan3.sh#L766 has more argument and like your other one I didn't pay attention and put that in quotes. This must not be placed in quotation marks https://github.com/openwrt/packages/blob/master/net/mwan3/files/lib/mwan3/mwan3.sh#L770

I think kids and home office drive me crazy and don't get along.

Thanks @aaronjg for your testing :+1:

feckert, 2020-07-15:
This should fix your issue I have added with my last P/R.

[0001-mwan3-fix-quotation-error.txt](https://github.com/openwrt/packages/files/4926003/0001-mwan3-fix-quotation-error.txt)

aaronjg, 2020-07-15:
@feckert - yep, that looks like the issue with the iptables error @jamesmacwhite was having. That one looks like it could actually be cleaned up to just contain the probability number and move the other arguments out of the `probability` varibale and into the $IPT line.

feckert, 2020-07-15:
@aaronjg What is with the other patch that should fix the calculation?
The way I see it, the calculation is wrong now and the patch fixed that.
Can you test that again?
If you give me a thumbs up, I will merge the two patches tomorrow.

aaronjg, 2020-07-15:
Our comments crossed in the ethers. I missed your patch earlier :)

You actually don't need the `${probability:+...}` since probability will always be set.

I actually like the thinking on this though - with the conditional evaluation of the probability, you can move the $IPT lines outside of the `metric` conditional logic and cut out some repeated code. 

I didn't test this part of the code (which is also why I didn't try to consolidate the IPv4 and IPv6 stuff here). I can set up a test for this and also do the IPv4/IPv6 consolidation here in my PR, but may not be for a couple days. It looks to me like the patch you propose will fix the issue though.

jamesmacwhite, 2020-07-15:
Thanks. This fixes the ip6tables error but the original uci and basename errors are still present. I'm currently on 2.8.8-1. I can update to 2.8.9 from the snapshot packages if it helps?

aaronjg, 2020-07-15:
@jamesmacwhite 

Did adding additional logging the rtmon script turn up anything?

jamesmacwhite, 2020-07-15:
I have a very verbose log where I set xtrace on all three sh scripts, that might have been a bad idea given just how much info it spits out, but I can provide the verbose log if you want to have look.

The basename and uci errors seems to appear at random points, but I'm slowly going a bit mad with what is causing it. Maybe someone else's eyes can help me! Here's the full verbose info of a `mwan3 restart` all redirected to a log file

https://jmwhite.co.uk/mwan3/mwan3_debug.log.txt

aaronjg, 2020-07-15:
Thanks for sharing the log. It is indeed bizarre. The basename error shows up much more often than basename is actually called. Perhaps basename is called by some other subprocess from the script, but I don't see any likely candidates.

Perhaps your busybox installation is somehow corrupted?

jamesmacwhite, 2020-07-15:
Yeah I don't get it at all. It is possible, but not sure how busybox could have been broken, you can't exactly reinstall it either without probably breaking a lot of things.

It happens with the default configuration so it can't be any of my rules or interface configuration. It wasn't happening on 2.8.6 or 2.8.7 I don't believe. I'd downgrade to confirm, but I can't find an ipk available of the older versions, so I might have to compile from source from a specific commit to test that.

I'm looking at everything that might have changed since, whether mwan3 related or not, because I'm out of ideas!

jamesmacwhite, 2020-07-15:
I'm wondering is this is to do with the recent init.d to rpcd changes, could the basename calls be coming from something like rc.common?

aaronjg, 2020-07-16:
Interesting idea.

Looks like it might be this line:
```procd_kill "$(basename ${basescript:-$initscript})" "$1"```
The other calls to basename have quotes around the argument to ensure an  empty argument goes through if basename is empty. Not sure why basename would be empty though. 

Perhaps there is some other script that gets restarted when mwan3 is restarted?

feckert, 2020-07-16:
> You actually don't need the `${probability:+...}` since probability will always be set.

@aaronjg I know but I think we should have a uniform code style. Also this makes it easier for me to see if the variable has any more arguments. The advantage would be that the quotation marks issue i introduced in may last merge, would not happen again! I have to update the [ipset](https://github.com/openwrt/packages/blob/master/net/mwan3/files/lib/mwan3/mwan3.sh#L1001) variable too.

jamesmacwhite, 2020-07-16:
Now I'm pretty sure if must be another script or process erroring out when mwan3 is restarted. Here's why:

1. No one else seems to be experiencing this problem, so it looks localised to my specific router
2. Upgrading to mwan3 2.8.9 does not resolve the issue
3. It must be a specific package I have installed that others do not

jamesmacwhite, 2020-07-16:
OMG. FINALLY FOUND THE PROBLEM.

miniupnpd........ Remove it entirely. Both the basename and uci errors go away. WTF. Even if it's not enabled or running it still causes problems. If it is removed entirely the errors are gone!

jamesmacwhite, 2020-07-16:
@feckert 

Just an FYI /lib/mwan3/mwan3.sh in 2.8.9 needs more testing. It broke my Wireguard routing entirely, I noticed several "Error: Invalid gateway address." errors being output when using the more recent changes, reverting mwan3.sh to 5147dfc resolves the problem.

closed 2020-07-16 by jamesmacwhite

aaronjg, 2020-07-16:
@jamesmacwhite Glad you were finally able to get to the bottom of it! 

@feckert - Yeah, in general, I think that style makes sense - just for the `probability` case we don't need to check if probability is set because it will certainly be set in one of the above lines. We can do it with quoting and without those checks here:

https://github.com/openwrt/packages/pull/12814/files#diff-ccaf6dffccf2fc134d63b5c33d5ee8e1R746

I agree overall that it makes sense to explicitly do the checks rather than to let the empty variables disappear. `ipset` could certainly use this treatment as well.  It's too bad `ash` doesn't support array variables, or a lot of this could be cleaned up even further.

jamesmacwhite, 2020-08-07:
The cause of the basename errors was found to be this!

https://github.com/openwrt/packages/pull/13051

caylorme, 2020-08-20:
> The cause of the basename errors was found to be this!
> 
> #13051
Thanks for the debugging session @jamesmacwhite. Prevented me from spending as much time as you in the crazy state.
I also happen to be using mwan3 with miniupnpd with a 3200ACM and getting this exact same thing.

Was expecting this to be the cause of ipv6 functionality breaking in my set up, but guessing maybe not?