openwrt/packages #12807: mwan3: switching to procd init script causes status/overview, system/startup loading failure

qweaszxcdf, 2020-07-13:
Maintainer: @feckert 
Environment: snapshot mwan3_2.8.8-1
Description:
switching to procd init script causes status/overview, system/startup loading failure

openwrt/luci#4228

feckert, 2020-07-14:
I can't reproduce this on my system. I use the latest master and the latest openwrt-19.07

jamesmacwhite, 2020-07-17:
It is hard to reproduce but I have experienced this as well. When I was debugging and switching between 2.8.8 and 2.8.9 where the init.d to procd change is, I had LuCI hang on me, stopping mwan3 resolved it, I assume there was an deadlock somewhere.

jow-, 2020-07-17:
This bug instance might be similar to the one fixed in https://github.com/openwrt/openwrt/commit/a03d6d2fab13c478a0f6cfc3082bec141f2adcf1

One of the processes started by mwan3 (I guess mwan3track?) might inherit the open file descriptor 1000 which holds the exclusive concurrency lock for init script execution.

@jamesmacwhite - when you can reproduce the hang, could you check via SSH which open descriptors are held by the mwan3track process? To check, get the pid from `ps` or similar, then check `ls -lh /proc/$pid/fd/`

qweaszxcdf, 2020-07-17:
```
root@OpenWrt:~# ps -ef|grep mwan3
root       522     1  0 15:42 ?        00:00:00 /bin/sh /etc/rc.common /etc/init.d/mwan3 enabled
root      3058     1  0 11:23 ?        00:00:16 /bin/sh /usr/sbin/mwan3track suda vpn-suda online 42.244.58.5 114.114.114.114
root      3385     1  0 11:23 ?        00:02:30 /bin/sh /usr/sbin/mwan3rtmon
root      3811     1  0 11:23 ?        00:00:15 /bin/sh /usr/sbin/mwan3track wan pppoe-wan online 121.236.89.8 www.qq.com www.baidu.com www.taobao.com 114.114.114.114 119.29.29.29 223.5.5.5
root      5146     1  0 11:23 ?        00:00:14 /bin/sh /usr/sbin/mwan3track vwan1 pppoe-vwan1 online 121.224.85.22 www.baidu.com 114.114.114.114 119.29.29.29
root      6110     1  0 11:23 ?        00:00:14 /bin/sh /usr/sbin/mwan3track vwan2 pppoe-vwan2 online 121.224.85.23 www.baidu.com 114.114.114.114 119.29.29.29
root      7140     1  0 11:23 ?        00:00:14 /bin/sh /usr/sbin/mwan3track vwan3 pppoe-vwan3 online 221.225.233.26 www.baidu.com 114.114.114.114 119.29.29.29
root      8120     1  0 11:23 ?        00:00:14 /bin/sh /usr/sbin/mwan3track vwan4 pppoe-vwan4 online 221.225.233.23 www.baidu.com 114.114.114.114 119.29.29.29
root     19832 19806  0 15:51 pts/2    00:00:00 grep mwan3
root@OpenWrt:~# ls -lh /proc/522/fd/
lr-x------    1 root     root          64 Jul 17 15:49 0 -> pipe:[49278443]
l-wx------    1 root     root          64 Jul 17 15:49 1 -> /dev/null
lr-x------    1 root     root          64 Jul 17 15:49 10 -> /etc/rc.common
l-wx------    1 root     root          64 Jul 17 15:49 1000 -> /tmp/lock/procd_mwan3.lock
lr-x------    1 root     root          64 Jul 17 15:49 11 -> /lib/functions/procd.sh
l-wx------    1 root     root          64 Jul 17 15:49 2 -> pipe:[36039876]
root@OpenWrt:~# ls -lh /proc/3058/fd/
lr-x------    1 root     root          64 Jul 17 15:43 0 -> /dev/null
lrwx------    1 root     root          64 Jul 17 15:43 1 -> /dev/pts/7 (deleted)
lr-x------    1 root     root          64 Jul 17 15:43 10 -> /usr/sbin/mwan3track
lrwx------    1 root     root          64 Jul 17 11:23 2 -> /dev/pts/7 (deleted)
root@OpenWrt:~# ls -lh /proc/3385/fd/
lr-x------    1 root     root          64 Jul 17 15:44 0 -> /dev/null
lrwx------    1 root     root          64 Jul 17 15:44 1 -> /dev/pts/7 (deleted)
lr-x------    1 root     root          64 Jul 17 15:44 10 -> /usr/sbin/mwan3rtmon
lrwx------    1 root     root          64 Jul 17 11:23 2 -> /dev/pts/7 (deleted)

```
@jow-

jow-, 2020-07-17:
Thank you @qweaszxcdf - that confirms my suspicion.

feckert, 2020-07-17:
@jow- thanks for the help and for your clearing words.
In the meantime, shall we reverse the change in the stable?

@jamesmacwhite Thanks for your tests.
In my setup I have not seen this?
I have only seen that the process lock did not get removed!
```
root@test ~ # ps | grep mwan3
 4178 root      1128 S    vi /etc/config/mwan3
32026 root      1108 S    lock /var/run/mwan3.lock
32047 root      1116 S    grep mwan3
```

I'll get right on it.
And will add @jow- suggestion and see if this helps ` /usr/sbin/mwan3 start 1000>&-`.

feckert, 2020-07-17:
@jamesmacwhite I have added @jow- suggestion and also extend `mwan3_lock_clean` to kill running mwan3 lock on mwan3 stop.

[0001-mwan3-add-workaround-for-procd-change.txt](https://github.com/openwrt/packages/files/4936634/0001-mwan3-add-workaround-for-procd-change.txt)
[0002-mwan3-also-cleanup-lock-on-mwan3-stop.txt](https://github.com/openwrt/packages/files/4936636/0002-mwan3-also-cleanup-lock-on-mwan3-stop.txt)

jamesmacwhite, 2020-07-17:
@feckert Applied to the two patches and did `mwan3 stop` and `mwan3 start` sequence. Looks OK, no deadlock, LuCI is OK and mwan3 itself is working. Maybe worth @qweaszxcdf testing as well. I wasn't able to reliably reproduce it, as I nearly always do `mwan3 restart` anyway.

qweaszxcdf, 2020-07-17:
@jamesmacwhite what's your output when run `/etc/init.d/mwan3 enable` or `/etc/init.d/mwan3 enabled`
@feckert According to my test, Luci still hangs when visiting status/overview or system/startup after patch applied.

feckert, 2020-07-17:
@qweaszxcdf I can't confirm that. Have you rebooted?
Could it be another problem?
I have to say I only tested my change on the master Branch.
[image]

qweaszxcdf, 2020-07-17:
@feckert Luci works fine after reboot.
Found another issue: get `iptables v1.8.4 (legacy): Invalid match name " statistic --mode random --probability 0.500" (28 chars max)` from output and only one interface shown
 ```
Current ipv4 policies:
balanced:
 wan (100%)
```
it shoud be 
```
Current ipv4 policies:
balanced:
 vwan4 (20%)
 vwan3 (20%)
 vwan2 (20%)
 vwan1 (20%)
 wan (20%)
```
checking recent comimit https://github.com/openwrt/packages/commit/feae9e57423897e14dfb9f45f2defb97f480a731#diff-ccaf6dffccf2fc134d63b5c33d5ee8e1 and found 
```
				$IPT4 -I mwan3_policy_$policy \
					-m mark --mark 0x0/$MMX_MASK $probability \
				$IPT4 -I "mwan3_policy_$policy" \
					-m mark --mark 0x0/$MMX_MASK "$probability" \

```
```
				$IPT6 -I mwan3_policy_$policy \
				$IPT6 -I "mwan3_policy_$policy" \
					-m mark --mark 0x0/$MMX_MASK \
					$probability \
					"$probability" \
```
fix this issue by undoublequoting "$probability"

jamesmacwhite, 2020-07-17:
@qweaszxcdf Yeah, a couple of regressions sunk in on 2.8.9, there is active PR currently which fixes it, along with various other improvements:

https://github.com/openwrt/packages/pull/12814

2.8.10 should fix that issue when it is merged into master.

aaronjg, 2020-07-17:
I had this issue as well and can confirm that the fd patch resolves the issue.

closed 2020-07-20 by neheb

aaronjg, 2020-07-28:
@jow- Could you describe a bit more about the 'proper' way to do this using procd. If it makes sense to move things in that direction, I can work on it.

It seems relatively straightforward for the mwan3rtmon process, which runs as long as mwan3 is running. However the mwan3track processes seem a bit more challenging.

Right now they are launched when mwan3 started, than if an interface goes offline (eg through ifdown) the process stops running. When the interface comes back, they are relaunched from the hotplug script.

Is it possible to replicate this sort of behavior using procd? If you can outline how this would work, I could implement it.

PR #12879 (2020-07-20): mwan3: update to version 2.8.11
Maintainer: me
Compile tested: only script changes
Run tested: x86_64, APU3, OpenWrt master

Description:

- [Commit](https://github.com/openwrt/packages/commit/058a2b6f3057d849cbe4f2c4fda258d32d6da2fd) fixes #12807 
- Add online time and uptime to detail output
- Remove pending lock process on mwan3 stop
- [Commit](https://github.com/openwrt/packages/commit/bcd13ba95cd5b96d04e490b576faab54404cbf69) fixes output `uci: Entry not found`