openwrt/packages #12813: Upgrade bridge-utils to 1.7

Neustradamus, 2020-07-15:
Can you upgrade bridge-utils to 1.7?
- https://git.kernel.org/cgit/linux/kernel/git/shemminger/bridge-utils.git/
- https://wiki.linuxfoundation.org/networking/bridge

-> https://github.com/openwrt/packages/tree/master/net/bridge-utils

Thanks in advance.

neheb, 2020-07-15:
The README says:

```
This utility is DEPRECATED. No new features will ever be added.
Instead use the bridge command from the iproute2 package which
supports more features.
```

Neustradamus, 2020-07-16:
@neheb: Yes, it is the last official build.
Several years after the 1.6.

In this case, it is needed to upgrade to the last last bridge-utils version and plan a migration to iproute2...

neheb, 2020-07-16:
I'll merge and remove after the next release assuming to objections.

closed 2020-07-16 by neheb

Neustradamus, 2020-07-16:
@neheb: Thanks!
You will remove after the next release but it will be completely replaced with iproute2 in the same time?
- https://mirrors.edge.kernel.org/pub/linux/utils/net/iproute2/
- https://git.kernel.org/pub/scm/network/iproute2/iproute2.git
- https://github.com/shemminger/iproute2

neheb, 2020-07-16:
Hmm? Isn't that already available?

Neustradamus, 2024-11-03:
Note: After this update, a new one (1.7.1) has been update:
- https://github.com/openwrt/packages/pull/17753