openwrt/packages #12815: meson: Some packages fail to build with 'No module named pathlib' Python traceback

detly, 2020-07-15:
Maintainer: @neheb
Environment: Defaults (ath79), ramips, 18.06.5 and 19.07.3

Some packages that use `PKG_BUILD_DEPENDS:=meson/host` fail to build, ending with a Python traceback that says:

```text
/json-glib-1.4.4/openwrt-build /home/jason/Code/clean-openwrt/build_dir/target-mips_24kc_musl/json-glib-1.4.4/openwrt-build/..
Traceback (most recent call last):
  File "/home/jason/Code/clean-openwrt/staging_dir/hostpkg/lib/meson/meson.py", line 18, in <module>
    from pathlib import Path
ImportError: No module named pathlib
```

I actually discovered this with a package I'm creating, but you can see it in the `json-glib` package as well.

To reproduce on 19.07.3, clone the OpenWrt repo and do:

```bash
git co v19.07.3
sed -i -e 's/\(src-git packages.\+^\).\+$/\1335efed8cb56bcc2c140efcb42a3b70fb4b4a14c/' feeds.conf.default 
scripts/feeds update packages
scripts/feeds install json-glib
echo 'CONFIG_PACKAGE_json-glib=y' > .config
make defconfig
make download
make tools/{compile,install} toolchain/{compile,install} target/linux/compile
make -j 1 V=sc package/json-glib/compile
```

Note that `json-glib` build-depends on `meson/host` and has this problem; but I also tried `mpc` and it doesn't. I did notice that installing `mpc` pulled in Python3 as well, although I don't see any Python-related deps in its Makefile.

This also seems superficially related to #12443, but perhaps not really because:
- that was fixed by patching the `zstd` package, and
- that mentions it's a problem on Debian with the `python-minimal` package installed; I am using Ubuntu 20.04 with a full Python 3 installation (indeed, running Python 3 and doing `import pathlib` on the build machine works fine).

I suspect, but haven't proven, that it's related to using Meson's `subdir(...)` function in the build configuration. `mpc` coincidentally has any `subdir(...)` calls guarded by options that its Makefile disables; whereas `json-glib` (and my package) rely on them for normal building.

Let me know if you want the full log.

detly, 2020-07-15:
I was wrong, it happens to `mpc` in current `master`, I must have forgotten to bump the ref in `feeds.conf.default` when I tried that.

detly, 2020-07-15:
If you don't want to wait for `mpc` or whatever to compile, here's the most minimal example I could make. Unzip it to `package` ie. so you have `package/localtest/mesonhello/...`, and do:

```test
scripts/feeds update -i packages
scripts/feeds install mesonhello
echo 'CONFIG_PACKAGE_mesonhello=y' > .config
make defconfig
make -j 1 V=sc package/mesonhello/compile
```

[localtest.zip](https://github.com/openwrt/packages/files/4925268/localtest.zip)

detly, 2020-07-15:
I set a breakpoint (`import pdb; pdb.set_trace()`) before the offending import, and it looks like Meson is using... Python 2?

```text
> /home/jason/Code/clean-openwrt/staging_dir/hostpkg/lib/meson/meson.py(19)<module>()
-> from pathlib import Path
(Pdb) import sys
(Pdb) p sys.version
'2.7.18rc1 (default, Apr  7 2020, 12:05:55) \n[GCC 9.3.0]'
(Pdb) p __name__
'__main__'
```

This is despite the clear `#!/usr/bin/env python3` at the top of that very file. Here's some more:

```text
(Pdb) p sys.argv
['/home/jason/Code/clean-openwrt/staging_dir/hostpkg/lib/meson/meson.py', '--buildtype', 'plain', '--native-file', '/home/jason/Code/clean-openwrt/build_dir/target-mips_24kc_musl/mesonhello/openwrt-native.txt', '--cross-file', '/home/jason/Code/clean-openwrt/build_dir/target-mips_24kc_musl/mesonhello/openwrt-cross.txt', '/home/jason/Code/clean-openwrt/build_dir/target-mips_24kc_musl/mesonhello/openwrt-build', '/home/jason/Code/clean-openwrt/build_dir/target-mips_24kc_musl/mesonhello/openwrt-build/..']
(Pdb) p sys.executable
'/home/jason/Code/clean-openwrt/staging_dir/host/bin/python'
(Pdb) 
```

There is no `python3` in `staging_dir/host/bin/python` either.

jefferyto, 2020-07-15:
Ping meson maintainer @dhewg

detly, 2020-07-15:
One more thing, sorry for the barrage — I was suspicious of the version number there, and indeed OpenWrt's buildroot is using the system Python (ie. the one installed on the build machine via `apt`):

```text
~ • python
Python 2.7.18rc1 (default, Apr  7 2020, 12:05:55) 
[GCC 9.3.0] on linux2
~ • file /home/jason/Code/clean-openwrt/staging_dir/host/bin/python
/home/jason/Code/clean-openwrt/staging_dir/host/bin/python: symbolic link to /usr/bin/python2.7
```

dhewg, 2020-07-15:
I never tried on anything but master, where the build system itself uses python3:
```
staging_dir/host/bin/python -> /usr/bin/python3.8
staging_dir/host/bin/python3 -> /usr/bin/python3.8
```

Last time I checked 19.x their buildsystem didn't use python3. I thought meson got a `python3/host` dependency there for exactly that reason?

jefferyto, 2020-07-15:
I would suggest you do a clean checkout and try the build again (and you shouldn't need to edit feeds.conf.default).

jefferyto, 2020-07-15:
> git co v19.07.3
> sed -i -e 's/\(src-git packages.\+^\).\+$/\1335efed8cb56bcc2c140efcb42a3b70fb4b4a14c/' feeds.conf.default 

You're checking out v19.07.3 of the main OpenWrt repo but 335efed8cb56bcc2c140efcb42a3b70fb4b4a14c is on the master branch of the packages repo, not the openwrt-19.07 branch. Seems like your problem is here :joy:

detly, 2020-07-15:
Yes, that's deliberate — I'm stuck with 18.07 OpenWrt due to some not-yet-mainlined kernel drivers and patches. However I want to use Meson to build a package. I didn't realise there was a branching scheme that matched up OpenWrt releases with the package repo, oops.

Although I notice 125f6ff9ec01c57e0cf57b042b95884f160cf48f might have the changes I need, so perhaps I can set my package feed ref to that and have it work.

neheb, 2020-07-15:
Most likely.

I could backport is to 18.06. But nothing in 18.06 uses meson.

neheb, 2020-07-15:
Also read the comments in https://github.com/openwrt/packages/pull/11843

detly, 2020-07-16:
@neheb Meson on the 19.07 branch works as-is under 18.06, and if using 19.07 packages under 18.06 causes problems, I have my own private repo for prototyping stuff that I can port it into.

Sorry for the noise everyone, and thanks for the info.

closed 2020-07-16 by detly