openwrt/packages #12835: docker-ce: security issue, open ports from WAN

dimonb, 2020-07-16:
Maintainer: G.M0N3Y.2503@gmail.com
Environment: snapshot docker-ce_19.03.12-2_x86_64

Description:

If tunnel port from docker container to host (-P option), it will be accessible from WAN. By default access from WAN should be prohibited for docker containers I think

Example:
`docker run --rm -ti -p 65341:8800 krys/echo-server`

From wan:
```
~ » nmap -p65341  xxxx
Starting Nmap 7.80 ( https://nmap.org ) at 2020-07-16 09:57 MSK
Nmap scan report for 172.17.0.79
Host is up (0.15s latency).

PORT      STATE SERVICE
65341/tcp open  unknown
```

dimonb, 2020-07-16:
If block this port from "Firewall - Traffic Rules" it remains blocked until reboot :(

feckert, 2020-07-17:
We are working on an generic solution for this issue with docker-ce:

Please have a look at the comment from  @jmarcet as an suggestion howto integrate this in openwrt.
https://github.com/openwrt/packages/pull/12556#issuecomment-648866347

This includes changes to luci-app-dockerman that need to be considered.
https://github.com/openwrt/packages/pull/12804#issuecomment-660017871

G-M0N3Y-2503, 2021-05-24:
Looks like this can be closed as of https://github.com/openwrt/packages/pull/12897

closed 2021-06-07 by feckert