openwrt/packages #12837: miniupnpd: Causes basename command output to be shown when mwan3 is restarted

jamesmacwhite, 2020-07-16:
Maintainer: @neheb (Sorry I don't know who's the true maintainer?)
Environment: 19.07.3 Linksys WRT3200ACM

Description:

The summary of #12806 is that when I use mwan3 and use a command like `mwan3 restart` I was getting the the following output below constantly in console. Originally it looked like it was mwan3 itself, but it is actually coming from miniupnpd. I determined this after echoing the initscript $1 variable in /etc/rc.common and watched the output. This led me to discover it was miniupnpd causing the output and confirmed this when I completely remove the package with opkg.

```
BusyBox v1.30.1 () multi-call binary.

Usage: basename FILE [SUFFIX]

Strip directory path and .SUFFIX from FILE
uci: Entry not found
uci: Entry not found
```

I'm not sure why a basename call is essentially empty, but I now know it is tied to miniupnpd and procd maybe?

neheb, 2020-07-17:
The miniupnpd init script does not use basename. No idea what your issue is.

neheb, 2020-07-17:
Try:

```
--- a/net/miniupnpd/files/miniupnpd.init
+++ b/net/miniupnpd/files/miniupnpd.init
@@ -164,8 +164,7 @@ upnpd() {
 
 		[ -z "$uuid" ] && {
 			uuid="$(cat /proc/sys/kernel/random/uuid)"
-			uci set upnpd.config.uuid="$uuid"
-			uci commit upnpd
+			config_set uuid config uuid
 		}
 
 		[ "$uuid" = "nocli" ] || echo "uuid=$uuid"
```

jamesmacwhite, 2020-07-17:
@neheb Yes it is very strange, I still don't know exactly what is causing this, updating the init with that change does not resolve it. Perhaps miniupnpd is triggering something else, but it goes away when I remove the package.

neheb, 2020-08-07:
Fixed.

closed 2020-08-07 by neheb