openwrt/packages #12844: libudev-fbsd: Proposal

illiliti, 2020-07-16:
Maintainer: @dangowrt
Description:

Hello everyone!

I wrote my own replacement for libudev called libudev-zero. Mainly, it's intended to work with xorg-server/wayland but can be extended to provide more functionality. It's written in c99 and doesn't depend on external libraries unlike libudev-fbsd and also uses POSIX Makefile instead of CMake. Most of functions are implemented except for hotplugging. For now you need to restart your application to hotplug devices. Сurrenly, i'm working on hotplugging support which can be easily integrated with any device manager(idea is similar to vdev libudev-compat) or used without device manager at all. I propose considering replacing libudev-fbsd with libudev-zero because libudev-fbsd not actively maintained, originally designed for BSD's, limited, used CMake and depend on libevdev. What do you think ?

https://github.com/illiliti/libudev-zero

neheb, 2020-07-17:
libudev is what prevents usbutils from being updated further. Does this work with the latest version?

libudev-fbsd does not as it targets an older udev version.

illiliti, 2020-07-17:
Probably no, because usbutils uses deprecated udev_hwdb

https://www.freedesktop.org/software/systemd/man/libudev.html

_Furthermore, libudev also exports legacy APIs that should not be used by new software (and as such are not documented as part of this manual). **This includes the hardware database known as udev_hwdb** (please use the new sd-hwdb(3) API instead) and the udev_queue object to query the udev daemon (which should not be used by new software at all)._

I can try to do no-op udev_hwdb and see if it works

neheb, 2020-07-17:
Please do. I tried doing an update to usbutils by reverting the hwdb patch, but that broke with the release after that. I don't think that's the long term solution.

illiliti, 2020-07-17:
i did it. it just works

neheb, 2020-07-20:
Just realized libudev belongs here. It's needed for libfuse3, which is used for sshfs.

If ready, please post a pull request.

illiliti, 2020-07-20:
Are you sure that libfuse3 depends on libudev ? I don't see any udev_* functions in libfuse source code, only udev rules

neheb, 2020-07-20:
Yes for the mount utility. That's why I have it disabled.

illiliti, 2020-07-20:
There is no actual udev dependency, only udev rules

neheb, 2020-08-02:
I fixed fuse3. Turns out it's just a meson oddity.

dangowrt, 2020-08-02:
Thank you for that great work @illiliti !
That's all very good news. When I imported libudev-fbsd I was hoping for something like that to happen. 
Let's drop it right away and switch to libudev-zero!

diizzyy, 2020-08-03:
Uhm... No? libusb doesn't work (according to upstream) which is a showstopper

neheb, 2020-08-03:
Doesn't sound right. Maybe confused with usbutils mentioned above?

illiliti, 2020-08-03:
> Uhm... No? libusb doesn't work (according to upstream) which is a showstopper

libusb doesn't work if compiled with udev support. It works fine without udev

illiliti, 2020-08-03:
To be clear, libusb will be fixed with upcoming hotplugging support. For now we don't lose any functionality if libusb compiled with --disable-udev

dangowrt, 2020-10-15:
@illiliti Will you produce a pull request adding libudev-zero package to this repository our would you like me to do that? I think this is great work and can do the job much better than our current in-tree libudev-fbsd fork, hece I'd want to switch to libudev-zero asap.

illiliti, 2020-10-17:
@dangowrt So sorry for long delay. I'm very busy at the moment, so i can't do pull request in the near time. Would be awesome if you could do that yourself.

dangowrt, 2020-10-19:
I've just tried popping it in instead of libudev-fbsd and libinput as well as usbip still fail to build against libudev-zero... Maybe I'm doing something wrong, also don't have much time to take care of this right now...

closed 2020-11-03 by dangowrt