openwrt/packages #12847: squid can not be compiled with enabled ASLR PIE

BKPepe, 2020-07-17:
Maintainer: @ratkaj 
cc: @neheb 
Environment: aarch64_cortex-a53, Turris MOX, OpenWrt 19.07 and OpenWrt master

Description:

OpenWrt master
```
OpenWrt-libtool: link: ccache_cxx -Wall -Wpointer-arith -Wwrite-strings -Wcomments -Wshadow -Woverloaded-virtual -Werror -Wno-deprecated-register -pipe -D_REENTRANT -I/serverpath/build/staging_dir/target-aarch64_cortex-a53_musl/usr/include -Os -pipe -mcpu=cortex-a53 -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -ffile-prefix-map=/serverpath/build/build_dir/target-aarch64_cortex-a53_musl/squid-4.6=squid-4.6 -Wformat -Werror=format-security -fpic -fstack-protector-strong -D_FORTIFY_SOURCE=2 -Wl,-z -Wl,now -Wl,-z -Wl,relro -Wno-error .libs/squidS.o -fpic -specs=/serverpath/build/include/hardened-ld-pie.specs -znow -zrelro -o squid AclRegs.o AuthReg.o AccessLogEntry.o AsyncEngine.o cache_cf.o CacheDigest.o cache_manager.o CachePeer.o carp.o cbdata.o client_db.o client_side.o client_side_reply.o client_side_request.o BodyPipe.o clientStream.o CollapsedForwarding.o CommandLine.o CompletionDispatcher.o ConfigOption.o ConfigParser.o CpuAffinity.o CpuAffinityMap.o CpuAffinitySet.o debug.o delay_pools.o DelayId.o DelayBucket.o DelayConfig.o DelayPool.o DelaySpec.o DelayTagged.o DelayUser.o DelayVector.o ClientDelayConfig.o fs_io.o dlink.o dns_internal.o Downloader.o errorpage.o ETag.o event.o EventLoop.o external_acl.o ExternalACLEntry.o FadingCounter.o fatal.o fd.o fde.o filemap.o fqdncache.o FwdState.o gopher.o helper.o htcp.o http.o HttpHdrCc.o HttpHdrRange.o HttpHdrSc.o HttpHdrScTarget.o HttpHdrContRange.o HttpHeader.o HttpHeaderTools.o HttpBody.o HttpControlMsg.o HttpMsg.o HttpReply.o RequestFlags.o HttpRequest.o icp_v2.o icp_v3.o Instance.o int.o internal.o ipc.o ipcache.o LogTags.o main.o MasterXaction.o mem_node.o MemBuf.o MemObject.o mime.o mime_header.o multicast.o neighbors.o Notes.o Parsing.o pconn.o peer_digest.o peer_proxy_negotiate_auth.o peer_select.o peer_sourcehash.o peer_userhash.o PeerPoolMgr.o Pipeline.o redirect.o refresh.o RemovalPolicy.o send-announce.o SBufStatsAction.o SquidMath.o SquidNew.o stat.o StatCounters.o StatHist.o String.o StrList.o stmem.o store.o StoreFileSystem.o store_io.o StoreIOState.o store_client.o store_digest.o store_key_md5.o store_log.o store_rebuild.o store_swapin.o store_swapmeta.o store_swapout.o StoreMetaUnpacker.o StoreMeta.o StoreMetaMD5.o StoreMetaSTD.o StoreMetaSTDLFS.o StoreMetaURL.o StoreMetaVary.o StoreStats.o StoreSwapLogData.o Transients.o MemStore.o time.o tools.o tunnel.o urn.o wccp.o wccp2.o whois.o wordlist.o XactionInitiator.o LoadableModule.o LoadableModules.o err_type.o err_detail_type.o globals.o hier_code.o icp_opcode.o lookup_t.o repl_modules.o swap_log_op.o -Wl,--export-dynamic  -L/serverpath/build/staging_dir/target-aarch64_cortex-a53_musl/usr/lib -L/serverpath/build/staging_dir/target-aarch64_cortex-a53_musl/lib -L/serverpath/build/staging_dir/toolchain-aarch64_cortex-a53_gcc-8.4.0_musl/usr/lib -L/serverpath/build/staging_dir/toolchain-aarch64_cortex-a53_gcc-8.4.0_musl/lib auth/.libs/libacls.a ident/.libs/libident.a acl/.libs/libacls.a acl/.libs/libstate.a auth/.libs/libauth.a acl/.libs/libapi.a clients/.libs/libclients.a servers/.libs/libservers.a ftp/.libs/libftp.a helper/.libs/libhelper.a http/.libs/libhttp.a dns/.libs/libdns.a base/.libs/libbase.a ./.libs/libsquid.a ip/.libs/libip.a fs/.libs/libfs.a DiskIO/.libs/libdiskio.a -lrt -lpthread comm/.libs/libcomm.a anyp/.libs/libanyp.a security/.libs/libsecurity.a ssl/.libs/libsslsquid.a ssl/.libs/libsslutil.a ipc/.libs/libipc.a mgr/.libs/libmgr.a parser/.libs/libparser.a eui/.libs/libeui.a icmp/.libs/libicmp.a log/.libs/liblog.a format/.libs/libformat.a sbuf/.libs/libsbuf.a repl/liblru.a -lcrypt mem/.libs/libmem.a store/.libs/libstore.a ../lib/.libs/libmisccontainers.a ../lib/.libs/libmiscencoding.a ../lib/.libs/libmiscutil.a -lssl -lcrypto ../compat/.libs/libcompatsquid.a /serverpath/build/staging_dir/toolchain-aarch64_cortex-a53_gcc-8.4.0_musl/lib/libatomic.so /serverpath/build/staging_dir/toolchain-aarch64_cortex-a53_gcc-8.4.0_musl/lib/libstdc++.so -lm -ldl -L.. -lltdl -pthread -Wl,-rpath -Wl,/serverpath/build/staging_dir/toolchain-aarch64_cortex-a53_gcc-8.4.0_musl/lib -Wl,-rpath -Wl,/serverpath/build/staging_dir/toolchain-aarch64_cortex-a53_gcc-8.4.0_musl/lib
/serverpath/build/staging_dir/toolchain-aarch64_cortex-a53_gcc-8.4.0_musl/lib/gcc/aarch64-openwrt-linux-musl/8.4.0/../../../../aarch64-openwrt-linux-musl/bin/ld: AclRegs.o: in function `Acl::Init()::{lambda(char const*)#1}::_FUN(char const*)':
AclRegs.cc:(.text+0xc): relocation truncated to fit: R_AARCH64_LD64_GOTPAGE_LO15 against symbol `Acl::AllOf::Pool()' defined in .text._ZN3Acl5AllOf4PoolEv[_ZN3Acl5AllOf4PoolEv] section in AclRegs.o
/serverpath/build/staging_dir/toolchain-aarch64_cortex-a53_gcc-8.4.0_musl/lib/gcc/aarch64-openwrt-linux-musl/8.4.0/../../../../aarch64-openwrt-linux-musl/bin/ld: AclRegs.cc:(.text+0xc): warning: too many GOT entries for -fpic, please recompile with -fPIC
/serverpath/build/staging_dir/toolchain-aarch64_cortex-a53_gcc-8.4.0_musl/lib/gcc/aarch64-openwrt-linux-musl/8.4.0/../../../../aarch64-openwrt-linux-musl/bin/ld: final link failed: symbol needs debug section which does not exist
collect2: error: ld returned 1 exit status
OpenWrt-libtool: link: rm -f ".libs/squidS.o"
```

OpenWrt 19.07:
```
/serverpath/build/staging_dir/toolchain-aarch64_cortex-a53_gcc-7.5.0_musl/lib/gcc/aarch64-openwrt-linux-musl/7.5.0/../../../../aarch64-openwrt-linux-musl/bin/ld: cache_cf.o: in function `dump_SBufList(StoreEntry*, std::__cxx11::list<SBuf, std::allocator<SBuf> > const&)':
cache_cf.cc:(.text+0x220): relocation truncated to fit: R_AARCH64_LD64_GOTPAGE_LO15 against symbol `SBuf::rawContent() const' defined in .text section in sbuf/.libs/libsbuf.a(SBuf.o)
/serverpath/build/staging_dir/toolchain-aarch64_cortex-a53_gcc-7.5.0_musl/lib/gcc/aarch64-openwrt-linux-musl/7.5.0/../../../../aarch64-openwrt-linux-musl/bin/ld: cache_cf.cc:(.text+0x220): warning: too many GOT entries for -fpic, please recompile with -fPIC
/serverpath/build/staging_dir/toolchain-aarch64_cortex-a53_gcc-7.5.0_musl/lib/gcc/aarch64-openwrt-linux-musl/7.5.0/../../../../aarch64-openwrt-linux-musl/bin/ld: final link failed: symbol needs debug section which does not exist
collect2: error: ld returned 1 exit status
OpenWrt-libtool: link: rm -f ".libs/squidS.o"
make[6]: *** [Makefile:6181: squid] Error 1
make[6]: Leaving directory '/serverpath/build/build_dir/target-aarch64_cortex-a53_musl/squid-4.6/src'
make[5]: *** [Makefile:6945: install-recursive] Error 1
```

neheb, 2020-07-18:
Is this also the case with your version bump?

BKPepe, 2020-07-19:
I will. Thanks for the hint.

BKPepe, 2020-07-20:
I tried to add `--with-pic`  as suggested, but it fails with the same output.