openwrt/packages #12853: zstd: Buildbot errors ("Permission denied")

jefferyto, 2020-07-17:
Maintainer: none
Environment: arm_cortex-a7, powerpc_8540

Description:
Looks like errors are only happening for these two architectures:

* https://downloads.openwrt.org/snapshots/faillogs/arm_cortex-a7/packages/zstd/compile.txt
* https://downloads.openwrt.org/snapshots/faillogs/powerpc_8540/packages/zstd/compile.txt

And it's not an actual build error, but something wrong with the tarball:

```
bash: /store/buildbot/slave/shared-workdir/build/sdk/dl/zstd-1.4.5.tar.zst: Permission denied
/store/buildbot/slave/shared-workdir/build/sdk/staging_dir/host/bin/tar: This does not look like a tar archive
/store/buildbot/slave/shared-workdir/build/sdk/staging_dir/host/bin/tar: Exiting with failure status due to previous errors
Makefile:101: recipe for target '/store/buildbot/slave/shared-workdir/build/sdk/build_dir/target-arm_cortex-a7_musl_eabi/zstd-1.4.5/.prepared_5723a7cca025fe5af4ae2f6be93f29e0_18f1e190c5d53547fed41a3eaa76e9e9' failed
make[3]: *** [/store/buildbot/slave/shared-workdir/build/sdk/build_dir/target-arm_cortex-a7_musl_eabi/zstd-1.4.5/.prepared_5723a7cca025fe5af4ae2f6be93f29e0_18f1e190c5d53547fed41a3eaa76e9e9] Error 2
time: package/feeds/packages/zstd/compile#0.26#0.05#0.83
```

It looks like this error is persisting for every build run. Not sure how to correct this.

neheb, 2020-07-18:
The problem is the outdated SDK used by the buildbots.

tools/zstd provides the way to extract the tarball. I have no idea what's going wrong here.

neheb, 2020-07-18:
I'm looking at http://buildbot.openwrt.org/master/images/ to see why zstd is not being built but I can't find anything.

jefferyto, 2020-07-20:
You're right - the SDKs are too old. The SDKs used for arm_cortex-a7 ([mediatek/mt7629](https://downloads.openwrt.org/snapshots/targets/mediatek/mt7629/)) and powerpc_8540 ([mpc85xx/generic](https://downloads.openwrt.org/snapshots/targets/mpc85xx/generic/)) were last built on 3 May and 18 May, respectively.

zstd support (openwrt/openwrt@258dc0d0fd3aae47add9b7dca40848a92d03a4ea, openwrt/openwrt@4696112ea28299a805ef7de5aff32451adfb2fc3, openwrt/openwrt@4bd7990488b0ca7b5cae16f0a9147a4146759053) landed on 31 May.

Looks like [mediatek/mt7629](https://buildbot.openwrt.org/master/images/builders/mediatek%2Fmt7629) is failing to build because of a swig error, [mpc85xx/generic](https://buildbot.openwrt.org/master/images/builders/mpc85xx%2Fgeneric) is failing to build because of a weird error during kmodprepare. I don't think these are things I can help with :joy:

neheb, 2020-07-20:
Very strange. ping @jow- . Any idea what's happening?

neheb, 2020-07-22:
This is the actual error for mt7629:

```
make[1]: *** [package/boot/uboot-mediatek/compile] Error 2
```

That can be solved I think. Some of the other u-boot packages have patches that patch out swig.

jow-, 2020-07-23:
Yeah, I think the way towards a solution would be to fix the failing architectures. Due to severe time constraints I cannot be of much help atm unfortunately.

neheb, 2020-07-23:
I sent a patch fixing the uboot issue on the mailing list.

No idea what's wrong with powerpc.

neheb, 2020-07-24:
mt7629 builds now. Absolutely no idea what’s wrong with PowerPC.

jow-, 2020-07-24:
I think mpc85xx is "broken" because its "generic" subtarget got renamed to "p1010". That kmodprepare step just happens to be the first one tripping over this discrepancy because the expected directory layout in `bin/` differs.

We do have a simple sanity check in place to assert that the target matches what we configured after defconfig, but no check for the subtarget. I'll kick the buildmaster now to refresh its target list.

jefferyto, 2020-07-24:
Does that mean the [powerpc_8540 builder](https://buildbot.openwrt.org/master/packages/builders/powerpc_8540) will start using the mpc85xx/p1010 sdk automatically or does it need to be adjusted somewhere?

neheb, 2020-07-25:
Closing. Both targets have been fixed.

closed 2020-07-25 by neheb