openwrt/packages #12862: Softflowd

ruralroots, 2020-07-18:
Description:
Upgraded to  davidc502  build OpenWrt SNAPSHOT, r13342-e35e40ad82

I am now receiving:

``` Fri Jul 17 22:08:38 2020 daemon.info procd: Instance softflowd::instance1 s in a crash loop 6 crashes, 0 seconds since last crash ```
``` Fri Jul 17 22:08:38 2020 daemon.info procd: Instance softflowd::instance2 s in a crash loop 6 crashes, 0 seconds since last crash ```

Same behaviour for single or multiple instances in /etc/config/softflowd

Runs OK from rc.local, command line, and custom init script. Works without issue on previous davidc502 build OpenWrt SNAPSHOT r12570-f68c48f7dd

neheb, 2020-09-20:
Is this still an issue?

ruralroots, 2020-10-01:
This continues to be an issue on SNAPSHOT r14524. 

With the same /etc/config file, /etc/init.d/softflowd worked as advertised on r12570, failed on r13342, and now exhibits the same behaviour on r14524

The init script does parse the config file properly and builds the correct command line encapsulated within {instance1}. . .{end instance}{instance2}. . .{end instance} tags.

The only anomaly I see with the official script is the appending of a duplicate  -p pid parameter at the end of the generated command line (-p /var/run/vpn.pid)
As a workaround I  came up with a functional, working, but very clunky custom script as below so it seems not a matter of the binary.

`#! /bin/sh /etc/rc.common

USE_PROCD=1
START=99
#PROCD_DEBUG=1

TUNIFACE=vpn0				# VPN interface
LANIFACE=br-lan				# LAN interface
WANIFACE=wan				# WAN interface
TUNPORT=5556                       	# <listen port of PRTG VPN Netflow collector>
LANPORT=5555				# <listen port of PRTG LAN Netflow collector>
WANPORT=5557				# <listen port of PRTG WAN Netflow collector>
SERVER=10.10.1.100              	# <ip of PRTG Netflow collector>
TUNPID=/var/run/vpn.pid			# location of VPN pid file
LANPID=/var/run/lan.pid			# location of LAN pid file
WANPID=/var/run/wan.pid			# location of WAN pid file
TUNCTL=/var/run/vpn.ctl			# location of VPN ctl file
LANCTL=/var/run/lan.ctl			# location of LAN ctl file
WANCTL=/var/run/wan.ctl			# location of WAN ctl file
SOFTFLOWD=softflowd			# location of softflowd binary
SOFTFLOWCTL=/usr/sbin/softflowctl	# location of softflowctl binary (statistics/dump flows)

EXTRA_COMMANDS="statistics dump_flows pause resume"
EXTRA_HELP="        statistics  ->  Show the Statistics for All Collectors        
	dump_flows  ->  Dump All Flows
	pause       ->  Pause All Flow Collection        
	resume      ->  Resume All Flow Collection"

#		*********** RELOAD SERVICES ***********  
reload_service()
{
	stop_service
	start_service
}
#		*********** RESTART SERVICES ***********
restart()
{
	stop_service
	start_service
}

#		*********** START SERVICES  ************			
start_service()
{
	procd_open_instance
	procd_set_param command "${SOFTFLOWD}" -d -6 -v 9 -i "${TUNIFACE}" -n "${SERVER}:${TUNPORT}" -p "${TUNPID}" -c "${TUNCTL}" -T full
	echo "${SOFTFLOWD}" -d -i "${TUNIFACE}" -6 -v 9 -n "${SERVER}:${TUNPORT}" -p "${TUNPID}" -c "${TUNCTL}" -T full
	procd_close_instance

	procd_open_instance
	procd_set_param command "${SOFTFLOWD}" -d -6 -v 9 -i "${LANIFACE}" -n "${SERVER}:${LANPORT}" -p "${LANPID}" -c "${LANCTL}" -T full
	echo "${SOFTFLOWD}" -d -i "${LANIFACE}" -6 -v 9 -n "${SERVER}:${LANPORT}" -p "${LANPID}" -c "${LANCTL}" -T full
	procd_close_instance

	procd_open_instance
	procd_set_param command "${SOFTFLOWD}" -d -6 -v 9 -i "${WANIFACE}" -n "${SERVER}:${WANPORT}" -p "${WANPID}" -c "${WANCTL}" -T full
	echo "${SOFTFLOWD}" -d -i "${WANIFACE}" -6 -v 9 -n "${SERVER}:${WANPORT}" -p "${WANPID}" -c "${WANCTL}" -T full
	echo "${IFACE}"
	procd_close_instance
}

#		********** SERVICE TRIGGERS **********
service_triggers()
{
	procd_add_interface_trigger "interface.*" "$WANIFACE" /etc/init.d/sflow restart
	procd_add_interface_trigger "interface.*" "$LANIFACE" /etc/init.d/sflow restart
	procd_add_interface_trigger "interface.*" "$TUNIFACE" /etc/init.d/sflow restart
}

#		********* SERVICE STATUS ***********
statistics()
{
	${SOFTFLOWCTL} -c ${TUNCTL} statistics
	${SOFTFLOWCTL} -c ${WANCTL} statistics
	${SOFTFLOWCTL} -c ${LANCTL} statistics
}

#               ********** DUMP FLOWS **************
dump_flows()
{
	${SOFTFLOWCTL} -c ${TUNCTL} dump flows
	${SOFTFLOWCTL} -c ${WANCTL} dump flows
	${SOFTFLOWCTL} -c ${LANCTL} dump flows
}

#		********* STOP SERVICE *************
stop_service()
{
	${SOFTFLOWCTL} -c ${TUNCTL} shutdown
		rm -f ${TUNPID} 2> /dev/null
		rm -f ${TUNCTL} 2> /dev/null

	${SOFTFLOWCTL} -c ${LANCTL} shutdown
		rm -f ${LANPID} 2> /dev/null
                rm -f ${LANCTL} 2> /dev/null

	${SOFTFLOWCTL} -c ${WANCTL} shutdown
		rm -f ${WANPID} 2> /dev/null
                rm -f ${WANCTL} 2> /dev/null
}

#		********** PAUSE COLLECTION *************
pause()
{       
	${SOFTFLOWCTL} -c ${TUNCTL} stop-gather
	${SOFTFLOWCTL} -c ${LANCTL} stop-gather
	${SOFTFLOWCTL} -c d{WANCTL} stop-gather
}

#             	**********  RESUME COLLECTION  **********
resume()
{
	${SOFTFLOWCTL} -c ${TUNCTL} start-gather
	${SOFTFLOWCTL} -c ${LANCTL} start-gather
	${SOFTFLOWCTL} -c ${WANCTL} start-gather
}
`

ruralroots, 2020-10-19:
**This issue is resolved.** 

From softflowd documentation I understood that softflowd will expire quiescent flows after user-configurable periods using the
-t option. I assumed from this and the provided /etc/config/softflowd file that using a numeric value in the -t option would provide 
a general flow timeout value applied to all traffic unless overridden by one of the other timeouts. Not so.

When used, the -t option value must identify the specific flow type & default timeout thus:
timeout -t tcp 3600s
timeout -t tcp.rst 120s
timeout -t tcp.fin 300s
timeout -t udp 300s
timeout -t icmp 300s
timeout -t general 3600s
timeout -t maxlife 604800s
timeout -t expint 60s
 
NULLing the value from the timeout option in the config file has resolved the crash issue.

I did however find some superfluous code as identified in the diff that follows.

```
--- /rom/etc/init.d/softflowd
+++ /etc/init.d/softflowd
@@ -28,8 +28,6 @@
 	config_get_bool enabled "$section" 'enabled' '0'
 	[ "$enabled" -gt 0 ] || return 1
 
-	config_get pid_file "$section" 'pid_file'    # superfluous code - not needed
-
 	args=""
 	append_string "$section" 'interface' '-i'
 	append_string "$section" 'pcap_file' '-r'
@@ -45,7 +43,7 @@
 	append_bool "$section" track_ipv6 '-6'
 
 	procd_open_instance
-	procd_set_param command /usr/sbin/softflowd -d $args${pid_file:+ -p $pid_file}
+	procd_set_param command /usr/sbin/softflowd -d $args
 	procd_set_param respawn
 	procd_close_instance
 }
@@ -55,5 +53,5 @@
         chmod 0755 /var/empty
 
 	config_load 'softflowd'
-	config_foreach start_instance 'softflowd'
+	config_foreach start_instance
 }
```

closed 2020-10-19 by ruralroots