openwrt/packages #12867: travelmate - Captive portal on mikrotik router not detected

ChristianKuehnel, 2020-07-18:
Maintainer: @dibdot 
Environment:  AVM FRITZ!Box 7320, OpenWrt 19.07.3 r11063-85e04e9f46

Description:
Thank you for creating this package. It's exactly what I needed! 👍 

I have some issues with the automatic login in to the captive portal on a Mikrotik router. The debug log shows:
```
Sat Jul 18 13:08:33 2020 user.debug travelmate-1.5.4[17067]: f_net     ::: fetch: /bin/uclient-fetch, timeout: 5, url: http://captive.apple.com, result: net ok
```

From what I understand "net ok" means that we have internet access. But we do not, I still get prompted to manually login to the captive portal.

When running the portal detection commands on the Openwrt shell, I get this output:
```
root@OpenWrt:~# uclient-fetch http://captive.apple.com
Downloading 'http://captive.apple.com'
Connecting to 17.253.109.203:80
Redirected to /login?dst=http%3A%2F%2Fcaptive.apple.com%2F on 172.16.0.1
Writing to 'login?dst=http%3A%2F%2Fcaptive.apple.com%2F'
login?dst=http%3A%2F 100% |*******************************|  3329   0:00:00 ETA
Download completed (3329 bytes)
```

When printing the contents of the `raw` variable in `f_net()`, I get  `Download completed (69 bytes)`. So it does not say anything about the `Redirect` your looking for in the `awk` command. So I suppose the `tail -n 1` removed the data we actually needed to detect the redirection.

However my shell knowledge is not sufficient to actually offer solution here...

I worked around the issue by always returning `"net cp '172.16.0.1'"` from `f_net()`, which is not a very elegant solution. However I get the router to login to the captive portal...

dibdot, 2020-07-18:
Hi,

thanks for your feedback. I know that the current cp code is far from being bullet proof ... ;-))
Currently I'm working on travelmate 2.0 and I've changed the cp part heavily. Could you please provide the output of, with active captive portal of course:

`curl -w %{json} connectivity-check.ubuntu.com`

ChristianKuehnel, 2020-07-19:
Sorry, I traveled on and can't test any more.

dibdot, 2020-07-20:
Please re-test once travelmate 2.0 has been released and re-open this ticket if the issue persists.

closed 2020-07-20 by dibdot

ChristianKuehnel, 2021-03-24:
Now I am am back at the place where I had the issues.
`curl -w %{json} connectivity-check.ubuntu.com`  results in 
```
curl: unknown --write-out variable: 'json'
```
Maybe the json output is not available in this curl version? 

I guess you want to see the header information?
*Before* login on the captive portal:
```
root@OpenWrt:~# curl -D - connectivity-check.ubuntu.com
HTTP/1.1 302 Hotspot login required
Cache-Control: no-cache
Content-Length: 135
Content-Type: text/html
Date: Wed, 24 Mar 2021 14:56:59 GMT
Expires: 0
Location: http://172.16.0.1/login?dst=http%3A%2F%2Fconnectivity-check.ubuntu.com%2F

<html>
<head><title>Error 302: Hotspot login required</title></head>
<body>
<h1>Error 302: Hotspot login required</h1>
</body>
</html>
```
after a successful login:
```
root@OpenWrt:~# curl -D - connectivity-check.ubuntu.com
HTTP/1.1 204 No Content
Date: Wed, 24 Mar 2021 14:57:18 GMT
Server: Apache/2.4.18 (Ubuntu)
X-NetworkManager-Status: online
```

ChristianKuehnel, 2021-03-24:
If I run a `uclient-fetch http://captive.apple.com -O /dev/null` as travelmate does in `f_net()` it returns:
```
Downloading 'http://captive.apple.com'
Connecting to 17.253.109.202:80
Redirected to /login?dst=http%3A%2F%2Fcaptive.apple.com%2F on 172.16.0.1
Writing to '/dev/null'
/dev/null            100% |*******************************|  3400   0:00:00 ETA
Download completed (3400 bytes)
```

So just getting the last line `tail -n 1` will always return a `Download completed`. If you want to check for a `Redirected` statement, you need to scan the entire output...

dibdot, 2021-03-24:
Hi, thanks for the follow up. 
As said before, the CP code has been fully changed in travelmate 2.x and relies on curl > 7.7.0 (with json output support), therefore tests with uclient-fetch or something like that doesn't help. Hopefully travelmate 2.x will just work for you, once you've updated to forthcoming OpenWrt 21.x

ChristianKuehnel, 2021-03-25:
Hi @dibdot ,

I understand that you're focusing on the next generation of the package. However I need a solution for the current 19.07 release now. So I fixed it for my setup and created a PR #15249 for it. It's probably not the most elegant solution, however it works so far. 

I also created PR #15250 for the mikrotik login script.

Feel free to use or ignore these if they are not useful to you...