openwrt/packages #12868: mwan3: flock deadlock trying to restart

brianjmurrell, 2020-07-18:
Maintainer: @feckert 
Environment: 19.03.7

Description:
Trying to restart mwan3 with
```
/etc/init.d/mwan3 restart
```
hangs.  This is apparently due to a deadlock.

```
28355 pts/2    S+     0:00  |       \_ /bin/sh /etc/rc.common /etc/init.d/mwan3 restart
28359 pts/2    S+     0:00  |           \_ flock 1000
```

deadlocks with:

```
14256 ?        S      0:00 /bin/sh /etc/rc.common /etc/init.d/mwan3 running
14264 ?        S      0:00  \_ flock 1000
```

Should that `running` instance be hanging around so long?

Package info:
```
Package: mwan3
Version: 2.8.8-1
Depends: libc, ip, ipset, iptables, iptables-mod-conntrack-extra, iptables-mod-ipopt, jshn
Status: install user installed
Section: net
Architecture: all
Size: 14405
Filename: mwan3_2.8.8-1_all.ipk
Conffiles:
 /etc/config/mwan3 e24ba32a5ce88023830684f00bf5dbe18615b6c8645c81cca4f2e8d83cc186d5
 /etc/mwan3.user 6226c1286a55bbdee865a24321e8a4ca80ca7553b5d81adb41ed774eddc4d192
Description: Hotplug script which makes configuration of multiple WAN interfaces simple
 and manageable. With loadbalancing/failover support for up to 250 wan
 interfaces, connection tracking and an easy to manage traffic ruleset.
Installed-Time: 1595074387

```

brianjmurrell, 2020-07-18:
This seems to be even more serious than just re-starting mwan3.  I now have the following processes all deadlocked on the same lock:

```
 9085 ?        S      0:00 /bin/sh /etc/rc.common /etc/init.d/mwan3 running
 9089 ?        S      0:00  \_ flock 1000
17989 ?        S      0:00 /bin/sh /etc/rc.common /etc/init.d/mwan3 enabled
17993 ?        S      0:00  \_ flock 1000
23489 ?        S      0:00 /bin/sh /etc/rc.common /etc/init.d/mwan3 enabled
23493 ?        S      0:00  \_ flock 1000
15479 ?        S      0:00 /bin/sh /etc/rc.common /etc/init.d/mwan3 enabled
15488 ?        S      0:00  \_ flock 1000
15830 ?        S      0:00 /bin/sh /etc/rc.common /etc/init.d/mwan3 enabled
15834 ?        S      0:00  \_ flock 1000
16285 ?        S      0:00 /bin/sh /etc/rc.common /etc/init.d/mwan3 enabled
16289 ?        S      0:00  \_ flock 1000
```

I suspect this is going to continue to grow.

jamesmacwhite, 2020-07-18:
Have you tried the patches here? It is to do with the switch to procd.

https://github.com/openwrt/packages/issues/12807

feckert, 2020-07-18:
Fix is on the way.
@brianjmurrell thanks for reporting.

closed 2020-07-18 by feckert