openwrt/packages #12878: ModemManager: Allow using network.wwan.device=any, instead of sysfspath

jonesmz, 2020-07-19:
Maintainer: @nickberry17 
Environment: ModemManager 14

Description:

Allow setting device=any instead of device=$sysfspath

This is useful for systems which are known to have only one modem, ever. In this situation, modemmanager can locate the modem automatically using the special device "any".

This requires only this patch to modemmanager.proto, tested with the latest modemmanager.proto from github.com/openwrt/packages/ as of today.

index 07147b1..4c6acb9 100644
--- a/package/modemmanager/files/modemmanager.proto
+++ b/package/modemmanager/files/modemmanager.proto
@@ -367,7 +367,7 @@ proto_modemmanager_setup() {
                proto_set_available "${interface}" 0
                return 1
        }
-       [ -e "${device}" ] || {
+       [ "${device}" = "any" ] || [ -e "${device}" ] || {
                echo "Device not found in sysfs"
                proto_set_available "${interface}" 0
                return 1

aleksander0m, 2020-07-20:
The idea is not bad, but the logic behind the change in the proto handler is not fully right. This change would make the interface set as "available" even if there is no device in the system. Not sure what to say :/

jonesmz, 2020-07-21:
The workflow that netifd uses is pretty opaque to me.

Any hints on how to make this better? Happy to investige further if more information can be shared.

feckert, 2023-09-13:
I'd rather close this issue before it hangs around any longer. From my point of view it is also better to specify an exact syspath in the configuration than to use the 'any'.

closed 2023-09-13 by feckert