openwrt/packages #12888: nginx: mismatch in docs and reality

majkrzak, 2020-07-20:
Maintainer: @\<github-user> (find it by checking history of the package Makefile/21 people)
Environment: ARMv7 Processor rev 0 (v7l); TP-Link Archer C2600; OpenWrt 19.07.3 r11063-85e04e9f46 / LuCI openwrt-19.07 branch git-20.136.49537-fb2f363

Description:

When installed nginx on my OpenWrt instance, I've noticed that `nginx.conf` and other files like `lan...` are not like those described in: https://openwrt.org/docs/guide-user/services/webserver/nginx#openwrt_s_defaults

My:
```

user  root;
worker_processes  2;

#error_log  logs/error.log;
#error_log  logs/error.log  notice;
#error_log  logs/error.log  info;

pid        /var/run/nginx.pid;

events {
    worker_connections  1024;
}

http {
    include       mime.types;
    default_type  application/octet-stream;

    sendfile on;
    keepalive_timeout 0;

    client_body_buffer_size 10K;
    client_header_buffer_size 1k;
    client_max_body_size 1G;
    large_client_header_buffers 2 1k;

    gzip on;
    gzip_http_version 1.1;
    gzip_vary on;
    gzip_comp_level 1;
    gzip_proxied any;
    gzip_types text/plain text/css application/json application/x-javascript text/xml application/xml application/xml+rss text/javascript application/javascript image/svg+xml;

    root /www;

    server {
        listen 80 default_server;
        listen [::]:80 default_server;
        server_name _;
        return 301 https://$host$request_uri;
    }

    server {
        listen 443 ssl default_server;
        listen [::]:443 ssl default_server;
        server_name  localhost;

        ssl_protocols TLSv1 TLSv1.1 TLSv1.2;
        ssl_prefer_server_ciphers on;
        ssl_ciphers "EECDH+ECDSA+AESGCM:EECDH+aRSA+AESGCM:EECDH+ECDSA+SHA384:EECDH+ECDSA+SHA256:EECDH+aRSA+SHA384:EECDH+aRSA+SHA256:EECDH:DHE+AESGCM:DHE:!RSA!aNULL:!eNULL:!LOW:!RC4:!3DES:!MD5:!EXP:!PSK:!SRP:!DSS:!CAMELLIA:!SEED";
        ssl_session_tickets off;

        ssl_certificate /etc/nginx/nginx.cer;
        ssl_certificate_key /etc/nginx/nginx.key;

        location ~* .(jpg|jpeg|png|gif|ico|css|js)$ {
            expires 365d;
        }

        include luci_uwsgi.conf;

    }

    include /etc/nginx/conf.d/*.conf;
}
```

"Default"
```
# Please consider creating files in /etc/nginx/conf.d/ instead of editing this.
# For details see https://openwrt.org/docs/guide-user/services/webserver/nginx
 
user root;
 
events {}
 
http {
	access_log off;
	log_format openwrt
		'$request_method $scheme://$host$request_uri => $status'
		' (${body_bytes_sent}B in ${request_time}s) <- $http_referer';
 
	include mime.types;
	default_type application/octet-stream;
	sendfile on;
 
	client_max_body_size 17M;
	large_client_header_buffers 2 1k;
 
	gzip on;
	gzip_vary on;
	gzip_proxied any;
 
	root /www;
 
	include conf.d/*.conf;
}
```

neheb, 2020-07-20:
ping @Ansuel @peter-stadler

peter-stadler, 2020-07-21:
`opkg` does not replace a modified config (`nginx.conf`), but it should put the package file to `nginx.conf-opkg` and write a message about it, too. Didn't it?

(I am working on draft PRs #11456 and #11726 using UCI config with another file (`uci.conf`), this would solve the conflict with `nginx.conf`; but it will take some time: we want to drop the nossl package of Nginx first and to make that smooth, a [patch to opkg](https://patchwork.ozlabs.org/project/openwrt/patch/20200716143444.77470-1-peter.stadler@student.uibk.ac.at/) is needed imho …)

The other files should be the same (I am creating the wiki with a shell script that includes the content of the files, but maybe I made an error), can you post the differences, too?

majkrzak, 2020-07-21:
At least I did not touch the `nginx` config yet. I was trying to set up some simple reverse TLS proxy.

Anyway, if I understand correctly, it wont be anymore valid after your work will be done?

P.S. I've also noticed that `acme` script is rather not updating configs correctly:
https://github.com/openwrt/packages/blob/openwrt-19.07/net/acme/files/run.sh#L280-L290

peter-stadler, 2020-07-21:
After these changes, there will be a template that is used to create the config from UCI settings by default. There will be a UCI setting to turn this off and use the `/etc/nginx/nginx.conf` again (right now in the draft I have `uci set nginx.global.uci_enable=false`).

The idea would be to separate the configuration:
* The main file (now `nginx.conf`, then `uci.conf.template`) contain some presets and `include conf.d/*.conf` in the `http` part
* Further configuration (especially server parts) should go into `conf.d/${name}.conf` files (for now, then server parts could go into UCI config sections, too)
* The `conf.d` folder would also be the place for `.locations` files that are included into different `server` parts and ssl certificates (`.crt` and `.key` files)

Other packages could install different `.locations` files (e.g. luci and etesync-server) or edit server parts more easily, I will look into `acme` …

peter-stadler, 2020-07-21:
In your case it should be safe to replace the `nginx.conf` with `nginx.conf-opkg` (I am not sure where `worker_processes  2;` come from). It should use the `_redirect2ssl.conf` and `_lan.conf` in the `/etc/nginx/conf.d/` folder as default servers and you can add more files to that folder (e.g. for setting up a reverse TLS proxy) …

majkrzak, 2020-07-21:
I have cleaned up, and installed explicitly `nginx` package:
```
root@archer:/etc# opkg install nginx
Installing nginx (1.17.7-2) to root...
Downloading http://downloads.openwrt.org/releases/19.07.3/packages/arm_cortex-a15_neon-vfpv4/packages/nginx_1.17.7-2_arm_cortex-a15_neon-vfpv4.ipk
```

content of the `/etc/nginx/nginx.conf` file is now:
```
root@archer:/etc/nginx# cat nginx.conf 

user nobody nogroup;
worker_processes  1;

#error_log  logs/error.log;
#error_log  logs/error.log  notice;
#error_log  logs/error.log  info;

#pid        logs/nginx.pid;

events {
    worker_connections  1024;
}

http {
    include       mime.types;
    #default_type  application/octet-stream;

    #log_format  main  '$remote_addr - $remote_user [$time_local] "$request" '
    #                  '$status $body_bytes_sent "$http_referer" '
    #                  '"$http_user_agent" "$http_x_forwarded_for"';

    #access_log  logs/access.log  main;

    sendfile        on;
    #tcp_nopush     on;

    #keepalive_timeout  0;
    keepalive_timeout  65;

    #gzip  on;

    server {
        listen       80;
        server_name  localhost;

        #charset koi8-r;

        #access_log  logs/host.access.log  main;

        location / {
            root   html;
            index  index.html index.htm;
        }

        #error_page  404              /404.html;

        # redirect server error pages to the static page /50x.html
        #
        error_page   500 502 503 504  /50x.html;
        location = /50x.html {
            root   html;
        }

        # proxy the PHP scripts to Apache listening on 127.0.0.1:80
        #
        #location ~ \.php$ {
        #    proxy_pass   http://127.0.0.1;
        #}

        # pass the PHP scripts to FastCGI server listening on 127.0.0.1:9000
        #
        #location ~ \.php$ {
        #    root           html;
        #    fastcgi_pass   127.0.0.1:9000;
        #    fastcgi_index  index.php;
        #    fastcgi_param  SCRIPT_FILENAME  /scripts$fastcgi_script_name;
        #    include        fastcgi_params;
        #}

        # deny access to .htaccess files, if Apache's document root
        # concurs with nginx's one
        #
        #location ~ /\.ht {
        #    deny  all;
        #}
    }

    # another virtual host using mix of IP-, name-, and port-based configuration
    #
    #server {
    #    listen       8000;
    #    listen       somename:8080;
    #    server_name  somename  alias  another.alias;

    #    location / {
    #        root   html;
    #        index  index.html index.htm;
    #    }
    #}

    # HTTPS server
    #
    #server {
    #    listen       443 ssl;
    #    server_name  localhost;

    #    ssl_certificate      cert.pem;
    #    ssl_certificate_key  cert.key;

    #    ssl_session_cache    shared:SSL:1m;
    #    ssl_session_timeout  5m;

    #    ssl_ciphers  HIGH:!aNULL:!MD5;
    #    ssl_prefer_server_ciphers  on;

    #    location / {
    #        root   html;
    #        index  index.html index.htm;
    #    }
    #}

}
```

Which is still not what in the repo and docs :thinking:

majkrzak, 2020-07-21:
Ok, I found out:
**Docs are about current master, which is not in sync with current release: 19.07**

peter-stadler, 2020-07-21:
Yes, I changed the wiki to resemble the current master. We could add a link to the [old revision](https://openwrt.org/docs/guide-user/services/webserver/nginx?rev=1520109017) at the beginning, but that does not contain file of the current release 19.07 neither (it is about setting up Nginx for PHP) …

majkrzak, 2020-07-21:
What about adding a note that it is about current master only?

peter-stadler, 2020-07-21:
Yes, I will add that. Thank you :-)

closed 2020-07-21 by majkrzak