openwrt/packages #12890: kmod-usb-storage-uas: UASP is not working

bcanterac, 2020-07-21:
Maintainer: @mans0n 
Environment: Openwrt 19.07.3, Linksys WRT3200ACM

Description:
UASP is not working with an USB 3.0 external HDD, it is only possible to have around 40MBps write speed on HDD while on a PC the same HDD hits 110MBps write speed. Package kmod-usb-storage-uas is installed for UASP support but kernel reveals the following warning:

```
[    1.830001] usb 3-1: new SuperSpeed USB device number 2 using xhci-hcd
[    1.860632] usb 3-1: USB controller f10f8000.usb3 does not support streams, which are required by the UAS driver.
[    1.870949] usb 3-1: Please try an other USB controller if you wish to use UAS.
[    1.878291] usb-storage 3-1:1.0: USB Mass Storage device detected
```
Therefore, UASP is not working.

neheb, 2020-07-21:
This has nothing to do with the packages feed.

The Marvell USB controller does not support UAS. The message is pretty clear.

bcanterac, 2020-07-21:
Isn't it supposed that installing kmod-usb-storage-uas should enable USB 3.0 normal operation? Including controllers adjustments, which make it a package issue.

To clear that there is no package problem it could be nice to check if this package can make devices archieve more than 40MBps speeds.

neheb, 2020-07-22:
UAS != USB3

It uses bulk only transfer instead of UAS. marvell USB3 works fine.

bcanterac, 2020-07-22:
Package is "kmod-usb-storage-uas". As noticed, UAS isn't working, right?

neheb, 2020-07-22:
You seem to have a problem with comprehension. I'm done.

diizzyy, 2020-07-22:
Wrong repo and not a bug

closed 2020-07-22 by diizzyy