openwrt/packages #12892: make error!

snakwu, 2020-07-21:
Version: 0.54.3                                                                        
Source dir: /home/snakwu/openwrt-master/openwrt/build_dir/target-x86_64_musl/lz4-1.9.2/
contrib/meson                                                                          
Build dir: /home/snakwu/openwrt-master/openwrt/build_dir/target-x86_64_musl/lz4-1.9.2/c
ontrib/meson/openwrt-build                                                             
Build type: cross build                                                                
Using 'PKG_CONFIG_PATH' from environment with value: '/home/snakwu/openwrt-master/openw
rt/staging_dir/target-x86_64_musl/usr/lib/pkgconfig:/home/snakwu/openwrt-master/openwrt
/staging_dir/target-x86_64_musl/usr/share/pkgconfig'                                   
Project name: lz4                                                                      
Project version: DUMMY                                                                 
C compiler for the build machine: gcc (gcc 9.3.0 "gcc (Ubuntu 9.3.0-10ubuntu2) 9.3.0") 
C linker for the build machine: gcc ld.bfd 2.34                                        
Using 'CFLAGS' from environment with value: ''                                         
Using 'LDFLAGS' from environment with value: ''                                        
C compiler for the host machine: x86_64-openwrt-linux-musl-gcc (gcc 8.4.0 "x86_64-openw
rt-linux-musl-gcc (OpenWrt GCC 8.4.0 r13831-52cdd6185e) 8.4.0")                        
C linker for the host machine: x86_64-openwrt-linux-musl-gcc ld.bfd 2.31.1             
Build machine cpu family: x86_64                                                       
Build machine cpu: x86_64                                                              
Host machine cpu family: x86_64                                                        
Host machine cpu: 64                                                                   
Target machine cpu family: x86_64                                                      
Target machine cpu: 64                                                                 
Program python3 found: YES (/home/snakwu/openwrt-master/openwrt/staging_dir/host/bin/py
thon)                                                                                  
                                                                                       
../../../../build_dir/target-x86_64_musl/lz4-1.9.2/contrib/meson/meson.build:19:0: ERRO
R: <ExternalProgram 'python3' -> ['/home/snakwu/openwrt-master/openwrt/staging_dir/host
/bin/python']> is not a valid python or it is missing setuptools                       
                                                                                       
A full log can be found at /home/snakwu/openwrt-master/openwrt/build_dir/target-x86_64_
musl/lz4-1.9.2/contrib/meson/openwrt-build/meson-logs/meson-log.txt                    
make[3]: *** [Makefile:109: /home/snakwu/openwrt-master/openwrt/build_dir/target-x86_64
_musl/lz4-1.9.2/.configured_a3f0024ff73a97e449b569fb2abce425] Error 1                  
make[3]: Leaving directory '/home/snakwu/openwrt-master/openwrt/feeds/packages/libs/lib
lz4'                                                                                   
time: package/feeds/packages/liblz4/compile#0.59#0.16#0.98                             
make[2]: *** [package/Makefile:113: package/feeds/packages/liblz4/compile] Error 2     
make[2]: Leaving directory '/home/snakwu/openwrt-master/openwrt'                       
make[1]: *** [package/Makefile:107: /home/snakwu/openwrt-master/openwrt/staging_dir/tar
get-x86_64_musl/stamp/.package_compile] Error 2                                        
make[1]: Leaving directory '/home/snakwu/openwrt-master/openwrt'                       
make: *** [/home/snakwu/openwrt-master/openwrt/include/toplevel.mk:235: world] Error 2

snakwu, 2020-07-21:
x86

jefferyto, 2020-07-21:
What OS/version are you using?

ping @dhewg

snakwu, 2020-07-21:
ubuntu server 20  @dhewg

dhewg, 2020-07-21:
It's right there: "is not a valid python or it is missing setuptools"

snakwu, 2020-07-21:
snakwu@hassio-server:~$ python -V                                                      Python 2.7.18rc1
snakwu@hassio-server:~$

neheb, 2020-07-21:
Looks like it. I'll fix when I can.

neheb, 2020-07-21:
Fixed with https://github.com/openwrt/packages/commit/cb44cd60c499d78592170d88a3c9f03991071e10

neheb, 2020-07-21:
@jefferyto the package you're thinking of is zstd. Both have the same author.

closed 2020-07-21 by neheb

snakwu, 2020-07-21:
@neheb  Same error, compilation failed

neheb, 2020-07-21:
Patch was incomplete. Fixed: https://github.com/openwrt/packages/commit/2860cf169a7f54f4061500052b0ea9f0301c5325

snakwu, 2020-07-21:
@neheb ok，thx！！！