openwrt/packages #12895: [19.07] fbzmq for powerpc is not compiled

BKPepe, 2020-07-21:
Maintainer: @ammubhave (maybe ping @neheb)
Environment: Turris 1.x, powerpc 8540, OpenWrt 19.07

Description:

Fbzmq fails to compile.
Faillog: https://downloads.openwrt.org/releases/faillogs-19.07/powerpc_8540/packages/fbzmq/compile.txt

Snip:
```
CMake Error at /builder/shared-workdir/build/sdk/staging_dir/host/share/cmake-3.15/Modules/CMakeFindDependencyMacro.cmake:47 (find_package):
  By not providing "Findpython-six.cmake" in CMAKE_MODULE_PATH this project
  has asked CMake to find a package configuration file provided by
  "python-six", but CMake did not find one.

  Could not find a package configuration file provided by "python-six" with
  any of the following names:

    python-sixConfig.cmake
    python-six-config.cmake

  Add the installation prefix of "python-six" to CMAKE_PREFIX_PATH or set
  "python-six_DIR" to a directory containing one of the above files.  If
  "python-six" provides a separate development package or SDK, be sure it has
  been installed.
Call Stack (most recent call first):
  /builder/shared-workdir/build/sdk/staging_dir/hostpkg/lib/cmake/fbthrift/FBThriftConfig.cmake:69 (find_dependency)
  CMakeLists.txt:56 (find_package)

-- Configuring incomplete, errors occurred!
See also "/builder/shared-workdir/build/sdk/build_dir/target-powerpc_8540_musl/fbzmq-2019.10.07.00/CMakeFiles/CMakeOutput.log".
Makefile:48: recipe for target '/builder/shared-workdir/build/sdk/build_dir/target-powerpc_8540_musl/fbzmq-2019.10.07.00/.configured_68b329da9893e34099c7d8ad5cb9c940' failed
make[3]: *** [/builder/shared-workdir/build/sdk/build_dir/target-powerpc_8540_musl/fbzmq-2019.10.07.00/.configured_68b329da9893e34099c7d8ad5cb9c940] Error 1
time: package/feeds/packages/fbzmq/compile#0.60#0.18#1.62
```

Python-six is available to download: https://downloads.openwrt.org/releases/19.07.3/packages/powerpc_8540/packages/python-six_1.13.0-1_powerpc_8540.ipk

neheb, 2020-07-21:
I can't test this as my 19.07 VM is running CentOS 7, which can only compile C++11 host packages (this needs C++14).

neheb, 2020-07-22:
Fixed with the zstd fixes.

closed 2020-07-22 by neheb