openwrt/packages #12898: ModemManager

jonesmz, 2020-07-21:
Maintainer: @nickberry17 @aleksander0m 
Environment: (put here arch, model, OpenWrt version)

Description:/etc/hotplug.d/net/25-modemmanager-net causes misleading error messages.

Specifically, it's saying in sequence:
hotplug: add network interface $itfhere: event processed
hotplug: event reported: action=add, name=br-lan, subsystem=net
hotplug: error: parent device sysfspath not found

/etc/hotplug.d/net/25-modemmanager-net is being called with the environment variable "${INTERFACE}"
That then calls 
```
mm_report_event "${ACTION}" "${INTERFACE}" "net" "/sys${DEVPATH}"
```
the ACTION in this case is "add" which gets processed by a switch-case inside of mm_report_event
which adds the event in question to a CACHE file of some kind

Then, after the switch case for dealing with the cache the event gets reported to 

1) logread
2) modem manager daemon via mmcli --report-kernel-event="action=${action} ......."

At this point in the script, we have not bothered trying to determine if the device being hotplugged is actually a modem at all the scripts are, apparently, just expecting modemmanager daemon to get it right
and then it checks "Is the provided syspath blank or provided?"
if it's there, then it calls "mm_report_modem_wait" which calls "mm_find_physdev_sysfs_path" with the sysfspath variable
mm_find_physdev_sysfs_path recursively walks up from the provided sysfspath until it finds a folder that contains the two files "idVendor" and "idProduct"
mm_find_physdev_sysfs_path returns empty string, because the device in question does not have an idVendor, or idProduct file.
and finally, mm_report_modem_wait prints an error to the log explaining that it can't find the parent path.

Questions:

- Is it appropriate to do anything in this script file without checking that DEVPATH is defined.???
- Why no locking on the cache file? is it safe to assume the hotplug scripts are only ever called one at a time system wide?
- Why notify modem mananger daemon of a network event for something that is known to not be a modem?
- Why cache devices that are known to not be modems?

aleksander0m, 2020-07-22:
>     Is it appropriate to do anything in this script file without checking that DEVPATH is defined.???

That can probably be fine-tuned, yes, there are a lot of network interfaces (e.g. virtual ones) that don't have a DEVPATH, and we can safely ignore those in the hotplug scripts. As you pointed out the only problem is anyway misleading error messages, though, no functionality is affected.

>     Why no locking on the cache file? is it safe to assume the hotplug scripts are only ever called one at a time system wide?

I believe they are, yes; one by one, one after the other. This was actually "kind" of a problem in MM because it was assumed the events for the different ports would be reported very very quickly (as in udev) but in this case I actually had to made updates in MM to accommodate with ports of a single device being reported slowly.

>     Why notify modem mananger daemon of a network event for something that is known to not be a modem?

This can probably be improved, because of course virtual network interfaces or wlan devices are not modems, but the line is not so clear with ethX or usbX devices, a lot of modems may be reported like that if the kernel doesn't flag the network interfaces as WWAN. The outcome, though, is not really problematic, MM will just ignore those non-modem network devices. So again, may be improved but only if we have a clear way to identify non-modem network interfaces.

>     Why cache devices that are known to not be modems?

Same reason as above; the hotplug script doesn't currently make any distinction between an Ethernet network interface of the platform or an Ethernet network interface exposed by a modem. The logic deciding finally what is a modem or what is not a modem is in ModemManager. This is primarily because that same thing happens when udev is used; MM gets notified of all ports and then decides what is a modem and what is not a modem. Additional fine tuning can definitely added in the hotplug scripts to avoid this; and also in ModemManager itself if the logic is clear enough (i.e. so that in the udev case the same logic is applied).

wulfy23, 2020-12-23:
I think it's a good idea to fine tune the hotplug scripts... they descend way too far down the codepath un-necessarily on IFB teql, wlan etc. etc.

sourcing mm.common and spamming the log for each event for the above interfaces is a waste of system resources...

feckert, 2023-09-13:
For the reasons mentioned above, we cannot do more. Therefore, at least for now, all [virtual devices are not processed](https://github.com/openwrt/packages/commit/e66fead6d479b5870280918ea42160c7dd0fc532).

closed 2023-09-13 by feckert