openwrt/packages #12900: Php5.6.32 Makefile:1387: recipe for target 'ext/mysql/php_mysql.lo' failed   build 19.07.2

predators46, 2020-07-22:
i tried to make php5.6.32 on openwrt 19.07.2 but failed. I included an error log

`-c /home/open/19/build_dir/target-mips_24kc_musl/php-5.6.32/ext/mysql/php_mysql.c -o ext/mysql/php_mysql.lo 
/home/open/19/build_dir/target-mips_24kc_musl/php-5.6.32/ext/mysql/php_mysql.c: In function 'zm_info_mysql':
/home/open/19/build_dir/target-mips_24kc_musl/php-5.6.32/ext/mysql/php_mysql.c:693:46: error: 'MYSQL_UNIX_ADDR' undeclared (first use in this function); did you mean 'PDO_MYSQL_UNIX_ADDR'?
  php_info_print_table_row(2, "MYSQL_SOCKET", MYSQL_UNIX_ADDR);
                                              ^~~~~~~~~~~~~~~
                                              PDO_MYSQL_UNIX_ADDR
/home/open/19/build_dir/target-mips_24kc_musl/php-5.6.32/ext/mysql/php_mysql.c:693:46: note: each undeclared identifier is reported only once for each function it appears in
/home/open/19/build_dir/target-mips_24kc_musl/php-5.6.32/ext/mysql/php_mysql.c: In function 'php_mysql_do_connect':
/home/open/19/build_dir/target-mips_24kc_musl/php-5.6.32/ext/mysql/php_mysql.c:725:36: error: 'MYSQL_PORT' undeclared (first use in this function); did you mean 'MYSQL_STMT'?
  int hashed_details_length, port = MYSQL_PORT;
                                    ^~~~~~~~~~
                                    MYSQL_STMT
Makefile:1387: recipe for target 'ext/mysql/php_mysql.lo' failed
make[3]: *** [ext/mysql/php_mysql.lo] Error 1
make[3]: Leaving directory '/home/open/19/build_dir/target-mips_24kc_musl/php-5.6.32'
Makefile:535: recipe for target '/home/open/19/build_dir/target-mips_24kc_musl/php-5.6.32/.built' failed
make[2]: *** [/home/open/19/build_dir/target-mips_24kc_musl/php-5.6.32/.built] Error 2`

neheb, 2020-07-22:
ping @mhei

predators46, 2020-07-22:
> ping @mhei

> ping @mhei

Help...

mhei, 2020-07-22:
Why do you try such an old PHP version? You must provide more information if you want to have support. At least, you should point me to the feed and version you're using and provide a diffconfig etc.

predators46, 2020-07-22:
> Mengapa Anda mencoba versi PHP yang lama? Anda harus memberikan lebih banyak informasi jika Anda ingin mendapat dukungan. Setidaknya, Anda harus mengarahkan saya ke umpan dan versi yang Anda gunakan dan memberikan diffconfig dll.

I prefer php5, I use php5.6-32 from repo chaos calmer. it is very strange I tried it on lede and openwrt 18.01 and it worked. maybe it's clear from the error log to the failure

predators46, 2020-07-22:
> Why do you try such an old PHP version? You must provide more information if you want to have support. At least, you should point me to the feed and version you're using and provide a diffconfig etc.

Php5.6-32 https://github.com/openwrt/packages/tree/for-15.05/lang

predators46, 2020-07-23:
I see all of this is a mistake in the version of mariadb on OpenWrt 19.07 and I compare mariadb on OpenWrt 18.01 very different. Do you have an idea that php5 can be successfully made on OpenWRt 19.07?

predators46, 2020-07-23:
I really need php5 can be used at OpenWRt 19.07. assistance is greatly appreciated

predators46, 2020-07-23:
I use 2 patches here 
1. patch for icu 
2. patch for openssl

https://drive.google.com/file/d/12qlATZ61kFj4VBdiSj7TZPlCFpyZDLJ0/view?usp=drivesdk

https://drive.google.com/file/d/12njoyaFQ7YbqFE_3vOjLz5VHJDoqZ9zt/view?usp=drivesdk

mhei, 2020-07-23:
> Do you have an idea that php5 can be successfully made on OpenWRt 19.07?

I never tried. I could have a look this evening... but please provide the output of running `scripts/diffconfig.sh` so that I can exactly reproduce your desired package selection.

predators46, 2020-07-23:
> > Do you have an idea that php5 can be successfully made on OpenWRt 19.07?
> 
> I never tried. I could have a look this evening... but please provide the output of running `scripts/diffconfig.sh` so that I can exactly reproduce your desired package selection.

Fix. bug versi mariadb 19.07

closed 2020-07-23 by predators46