openwrt/packages #12905: lxc: ERROR: Unable to fetch GPG key from keyserver

alex9434, 2020-07-23:
# Issue template (remove lines from top till here)

Maintainer: @ratkaj 
Environment: OpenWrt SNAPSHOT, r13902-5e86877f36 on arm64 (Raspberry Pi 4B, 4GB)

Description: unable to create lxc container
```
lxc-create -n test -t download
Setting up the GPG keyring
ERROR: Unable to fetch GPG key from keyserver
lxc-create: test: lxccontainer.c: create_run_template: 1626 Failed to create container from template
lxc-create: test: tools/lxc_create.c: main: 319 Failed to create container test
```
Dear @ratkaj , @cshoredaniel , @neheb , @xzpron , @champtar , @antonlacon 
I suspect this might be more an issue with gnupg than lxc. I could not find the maintainer in the Makefile for gnupg and therefore have included the contributors. Therefore I apologize for the long distribution list.

I suspect the issue with lxc-create is caused by gnupg:
```
gpg -vvvv  --keyserver-options=debug  --keyserver "hkp://pool.sks-keyservers.net" --recv-keys "0xBAEFF88C22F6E216"
gpg: using character set `iso-8859-1'
gpg: directory `/root/.gnupg' created
gpg: can't open `/usr/share/gnupg/options.skel': No such file or directory
gpg: keyring `/root/.gnupg/secring.gpg' created
gpg: keyring `/root/.gnupg/pubring.gpg' created
gpg: requesting key 22F6E216 from hkp server pool.sks-keyservers.net
gpg: unable to execute program `/usr/lib/gnupg/gpgkeys_curl': No such file or directory
gpg: no handler for keyserver scheme `hkp'
gpg: keyserver receive failed: keyserver error
```
Any idea how to fix this?

xypron, 2020-07-23:
* Did you try installing the missing program /usr/lib/gnupg/gpgkeys_curl?
* hkp:// uses port 11371. Is this port allowed in your firewall?

You could download the key from https://keyserver.ubuntu.com/pks/lookup?op=get&search=0xbaeff88c22f6e216.

alex9434, 2020-07-30:
Dear all,

I have solved the issue. gpgkeys_curl is included in Package/gnupg-utils, that is by default not included in gnupg and needs to be installed separately. It might make sense to include the depencence on gnupg-utils in the Makefile for lxc-create so that this can be installed automatically.

Thank you,
Alex

closed 2020-07-30 by alex9434