openwrt/packages #12924: mwan3: rtmon_ip4/6 functions use uncommited uci changes

aaronjg, 2020-07-25:
Maintainer: @feckert
Environment: 19.07.3

Description:

mwan3_rtmon_ip4/6 functions call `uci get` which returns uncommitted changes staged using uci. These should 'config_get' functions to get only committed changes.

feckert, 2020-07-25:
The problem her is, if we use `config_get` in `mwan3_rtmon_ip4/6` then we have to load the whole config with `config_load mwan3` every time.
This is very time and memory intensive.

I see her two possibilities to achieve this:

- Add the variable on rtmon [start](https://github.com/openwrt/packages/blob/master/net/mwan3/files/lib/mwan3/mwan3.sh#L667)

or

- Before the rtmon [process](https://github.com/openwrt/packages/blob/master/net/mwan3/files/usr/sbin/mwan3rtmon#L24) loop starts. 

I think the second one is the best, because we to an `config_load mwan3` already.

aaronjg, 2020-07-25:
It looks like the config_load takes about 80ms, which is not great, but not terrible. On the other hand, the calls to `uci`, which we are currently doing, are even slower.

I have a fix for this ready here:
https://github.com/aaronjg/openwrt-packages/tree/mwan3-rtmon-uci-fix

Time to run an `mwan3 restart` is comparable 

new code
```
real    0m 22.97s
user    0m 9.75s
sys     0m 4.61s
```

old code
```
real    0m 23.35s
user    0m 9.69s
sys     0m 4.87s
```

performance checking of `uci get` vs. config_load.
```shell
~ # cat test_config_load.sh; time ./test_config_load.sh 
#!/bin/sh
. /lib/functions.sh
for i in `seq 1 100`; do
        config_load mwan3
        for section in ${CONFIG_SECTIONS}; do
                config_get cfgtype "$section" TYPE
                [ "$cfgtype" != "interface" ] && continue
                tid=$((tid+1))
                config_get family "$section" family ipv4
                config_get enabled "$section" enabled 0
        done
done
real    0m 7.73s
user    0m 6.62s
sys     0m 1.00s

```

```shell

~ # cat test_config_load2.sh; time ./test_config_load2.sh 
#!/bin/sh
for i in `seq 1 100`; do
        idx=0
        while uci get mwan3.@interface[$idx] >/dev/null 2>&1 ; do
                idx=$((idx+1))
                uci -q get mwan3.@interface[$idx].family >/dev/null
                uci -q get mwan3.@interface[$idx].enabled >/dev/null
        done
done
exit 0

real    0m 11.24s
user    0m 8.64s
sys     0m 2.27s

```

aaronjg, 2020-07-26:
Actually, we can do this in a much better way than polling for routing table changes. 

There is an `ip monitor route` function that can alert to routing changes. I updated the P/R to use that functionality. Now routing table changes will be detected and applied as soon as they happen, rather than waiting for the rtmon polling interval to elapse.

but it's a fairly sizeable change so more testing would be greatly appreciated.

jamesmacwhite, 2020-07-26:
@aaronjg Feel free to send the PR my way for UA testing. I've got two physical WANs with dual stack, along with two Wireguard interfaces also with dual stack, so should provide a good environment for testing the routing table codebase.

aaronjg, 2020-07-26:
@jamesmacwhite 

Thanks! That would be great. Can you pull it from my github branch? https://github.com/aaronjg/openwrt-packages/tree/mwan3-rtmon-uci-fix

jamesmacwhite, 2020-07-26:
@aaronjg Thanks! I have applied the latest changes in your PR to my setup. Looks to be all working for me with your routing table changes with rtmon.

I've confirmed some of my routing rules with the different interfaces and everything appears to be OK. I'll leave it running for a while, let me know if you want me to run anything specific for further verification.

aaronjg, 2020-07-26:
There are a couple of things to look for:

1) Are there any `RTNETLINK file exists` messages appearing when interfaces come online/go offline. This used to be a problem with the old rtmon function. It has been resolved and I want to make sure I didn't introduce any regressions.
2) Any issues with IPv6 links staying up? I'm not confident I can test this thoroughly in my setup.
3) How does the time to run `mwan3 restart` compare to before? For me it is about 1/3 the time from before, and would like to confirm that it is still fast on more complex setups.
4) Can you ping an IPv6 address using `-I <devname>`. I wasn't explicitly trying to address this, but I did cleanup the routing table stuff a lot, so perhaps this is working now?

jamesmacwhite, 2020-07-26:
Thanks. Here's my results of the tests:

1. No, I think you actuallt resolved this around 2.8.6 in your first PR for IPv6 improvements, I haven't seen `RTNETLINK file exists` since that version when running mwan3 restart. I know exactly what you mean though. I saw this regularly in older mwan3 versions.

2. IPv6 seems to be working fine. All interfaces are up and have been for around 45 minutes after restarting mwan3 with the newer code. I also just restarted mwan3 again for testing the performance for point 3 and again IPv6 seem fine on all interfaces, mwan3track shows them all as up and working, I've also confirmed with testing some rules and traceroute etc to each interface.

3. I have to say that is actually noticeably slower when running `mwan3 restart` for me, timing it it took nearly a full minute for the execution to complete. I can't say I had actually timed mwan3 under the current 2.8.11 release, but it seems a bit slower for me. It would still take sometime anyway. I imagine the more interfaces you have the longer it will be. In my case I have:

- Primary WAN (wan and wan6)
- Secondary WAN (wanb and wanb6)
- Wireguard VPN 1 (wg and wg6)
- Wireguard VPN 2 (wgb and wgb6)

So a total of 8 interfaces, all the IPv6 interfaces marked as `xxx6` are aliases to main interface, as splitting IPv4 and IPv6 interfaces is required for proper routing/rules with mwan3. I do use NAT66, due to my second WAN not having a prefix delegated and Wireguard being from Mullvad which only provides a single static ULA /128, but I'm not sure that's really a factor.

4. I can ping an IPv6 domain/IP with the interface name for all interfaces except my wan6 interface which is a 6in4. This still fails when using `-I 6in4-wan6`, works when specifying `-I 2001:470:xxxx:xx::2. This may however be a separate issue, because it's being documented a fair bit in terms of sit interfaces.

Hope this is useful!

aaronjg, 2020-07-26:
Thanks.  Could you run `time mwan3 restart` under this branch and the 2.8.11? There is a lot of repeated stuff that mwan3 does in the initialization that could potentially be cut out. It really shouldn't take a full minute to bring everything online and apply the rules. However, this patch is reducing some of the duplicated steps, so if it is slower, I need to figure out why.

Regarding 4 - yeah, this isn't related to the patch was just hopeful some of the cleanup may have fixed things. I'll try set up a 6in4 tunnel to see if I can diagnose what is causing this, but that is a project for another PR.

jamesmacwhite, 2020-07-26:
**Your PR:**

```
root@linksys-wrt3200acm:/usr/sbin# time mwan3 restart
real    0m 57.60s
user    0m 30.53s
sys     0m 6.90s
```
**2.8.11:**

```
root@linksys-wrt3200acm:/tmp# time mwan3 restart
sh: invalid number '6817
20553'
real    0m 59.37s
user    0m 29.11s
sys     0m 6.68s
```

Interestingly, pretty much the same, so it shows how human perception of time can be off! I have a feeling it may have become slower more recently with changes, but I'd imagine it is also relative to the amount of interfaces being tracked.

That invalid number error seems to be a potential bug, although it doesn't seem to be causing any problems from a routing point of view.

aaronjg, 2020-07-26:
Yeah, a minute is really slow. It can depend on how many interfaces you are tracking, how many policies you have, how many rules you have, and what other stuff is going on in the `iface` hotplug scripts. I'm working on some speed improvements for this...

Glad to know that this change is at least is not any slower than before.

jamesmacwhite, 2020-07-26:
There's a slight reduction in total time of execution with your PR and the routing changes in rtmon seem to be holding up on my setup, so hopefully it gives you a solid base to work from, if you want to pursue that. Avoiding having to use a polling interval is a good start, as it makes routing table changes instantaneous, which is a good thing.

I don't know if in addition to the amount of interfaces, having an interface as tun, wireguard or a VLAN WAN has any bearing on the performance as well. Personally, it doesn't bother me too much, given you won't really be calling mwan3 restart constantly and the hotplug events for each interface are more selective, but performance is always good!

Key config info from my setup is:

* 8 Interfaces
* 12 Member Profiles
* 7 Policies
* 19 custom rules + 3 default rules (https sticky, default IPv4 and IPv6)

closed 2020-08-19 by feckert