openwrt/packages #12930: <v4l2-utils> v4l2-ctl segfault when setting uvc controls

mgondium, 2020-07-25:
# v4l2-ctl -c sharpness=16
Segmentation fault

Currently broken in trunk, works in openwrt 18, not sure in openwrt 19.

I've read a similar problem that  has to do with some hacking regarding musl, not sure if it's already fixed and just needs a pull request.

neheb, 2020-07-26:
I bumped the version to the latest.

Please run strace and post the log. I have a feeling it's a uClibc++ issue.

mgondium, 2020-07-26:
Freshly built.
OpenWrt SNAPSHOT, r13952-ac9730c495
```
root@Openwrt:~# strace v4l2-ctl -c sharpness=16
execve("/usr/bin/v4l2-ctl", ["v4l2-ctl", "-c", "sharpness=16"], 0x7fd57438 /* 13 vars */) = 0
set_thread_area(0x77e0aec8)             = 0
set_tid_address(0x77e01ff8)             = 1851
open("/etc/ld-musl-mips-sf.path", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = -1 ENOENT (No such file or directory)
open("/lib/libv4l2.so.0", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = -1 ENOENT (No such file or directory)
open("/usr/local/lib/libv4l2.so.0", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = -1 ENOENT (No such file or directory)
open("/usr/lib/libv4l2.so.0", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = 3
fcntl64(3, F_SETFD, FD_CLOEXEC)         = 0
fstat64(3, {st_mode=S_IFREG|0755, st_size=45816, ...}) = 0
read(3, "\177ELF\1\2\1\0\0\0\0\0\0\0\0\0\0\3\0\10\0\0\0\1\0\0\17\300\0\0\0004"..., 936) = 936
mmap2(NULL, 114688, PROT_READ|PROT_EXEC, MAP_PRIVATE, 3, 0) = 0x77d3e000
mmap2(0x77d55000, 20480, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED, 3, 0x7000) = 0x77d55000
close(3)                                = 0
open("/lib/libuClibc++.so.0", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = -1 ENOENT (No such file or directory)
open("/usr/local/lib/libuClibc++.so.0", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = -1 ENOENT (No such file or directory)
open("/usr/lib/libuClibc++.so.0", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = 3
fcntl64(3, F_SETFD, FD_CLOEXEC)         = 0
fstat64(3, {st_mode=S_IFREG|0755, st_size=190972, ...}) = 0
read(3, "\177ELF\1\2\1\0\0\0\0\0\0\0\0\0\0\3\0\10\0\0\0\1\0\0\352\340\0\0\0004"..., 936) = 936
mmap2(NULL, 258048, PROT_READ|PROT_EXEC, MAP_PRIVATE, 3, 0) = 0x77cfe000
mmap2(0x77d3a000, 12288, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED, 3, 0x2c000) = 0x77d3a000
close(3)                                = 0
open("/lib/libgcc_s.so.1", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = 3
fcntl64(3, F_SETFD, FD_CLOEXEC)         = 0
fstat64(3, {st_mode=S_IFREG|0644, st_size=78080, ...}) = 0
read(3, "\177ELF\1\2\1\0\0\0\0\0\0\0\0\0\0\3\0\10\0\0\0\1\0\0'\320\0\0\0004"..., 936) = 936
mmap2(NULL, 147456, PROT_READ|PROT_EXEC, MAP_PRIVATE, 3, 0) = 0x77cda000
mmap2(0x77cfc000, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED, 3, 0x12000) = 0x77cfc000
close(3)                                = 0
open("/lib/libv4lconvert.so.0", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = -1 ENOENT (No such file or directory)
open("/usr/local/lib/libv4lconvert.so.0", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = -1 ENOENT (No such file or directory)
open("/usr/lib/libv4lconvert.so.0", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = 3
fcntl64(3, F_SETFD, FD_CLOEXEC)         = 0
fstat64(3, {st_mode=S_IFREG|0755, st_size=127644, ...}) = 0
read(3, "\177ELF\1\2\1\0\0\0\0\0\0\0\0\0\0\3\0\10\0\0\0\1\0\0\25\260\0\0\0004"..., 936) = 936
mmap2(NULL, 532480, PROT_READ|PROT_EXEC, MAP_PRIVATE, 3, 0) = 0x77c58000
mmap2(0x77c86000, 344064, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED, 3, 0x1e000) = 0x77c86000
mmap2(0x77c88000, 335872, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_ANONYMOUS, -1, 0) = 0x77c88000
close(3)                                = 0
mprotect(0x77d55000, 4096, PROT_READ)   = 0
mprotect(0x77d3a000, 8192, PROT_READ)   = 0
mprotect(0x77cfc000, 4096, PROT_READ)   = 0
mprotect(0x77c86000, 4096, PROT_READ)   = 0
mprotect(0x459000, 4096, PROT_READ)     = 0
--- SIGSEGV {si_signo=SIGSEGV, si_code=SEGV_MAPERR, si_addr=NULL} ---
+++ killed by SIGSEGV +++
Segmentation fault
root@Openwrt:~# 

```

neheb, 2020-07-26:
strange...I looked through the musl source. All of that output is in musl's ldso.

Can you compile with libstdcpp instead? make menuconfig > Global build settings > Preferred standard C++ library.

mgondium, 2020-07-26:
It looks very similar.
```
root@Openwrt:~# strace v4l2-ctl -c sharpness=16
execve("/usr/bin/v4l2-ctl", ["v4l2-ctl", "-c", "sharpness=16"], 0x7f8d9a58 /* 13 vars */) = 0
set_thread_area(0x77ec6ec8)             = 0
set_tid_address(0x77ebdff8)             = 1538
open("/etc/ld-musl-mips-sf.path", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = -1 ENOENT (No such file or directory)
open("/lib/libv4l2.so.0", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = -1 ENOENT (No such file or directory)
open("/usr/local/lib/libv4l2.so.0", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = -1 ENOENT (No such file or directory)
open("/usr/lib/libv4l2.so.0", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = 3
fcntl64(3, F_SETFD, FD_CLOEXEC)         = 0
fstat64(3, {st_mode=S_IFREG|0755, st_size=45816, ...}) = 0
read(3, "\177ELF\1\2\1\0\0\0\0\0\0\0\0\0\0\3\0\10\0\0\0\1\0\0\17\300\0\0\0004"..., 936) = 936
mmap2(NULL, 114688, PROT_READ|PROT_EXEC, MAP_PRIVATE, 3, 0) = 0x77dfa000
mmap2(0x77e11000, 20480, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED, 3, 0x7000) = 0x77e11000
close(3)                                = 0
open("/lib/libstdc++.so.6", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = -1 ENOENT (No such file or directory)
open("/usr/local/lib/libstdc++.so.6", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = -1 ENOENT (No such file or directory)
open("/usr/lib/libstdc++.so.6", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = 3
fcntl64(3, F_SETFD, FD_CLOEXEC)         = 0
fstat64(3, {st_mode=S_IFREG|0755, st_size=1340480, ...}) = 0
read(3, "\177ELF\1\2\1\0\0\0\0\0\0\0\0\0\0\3\0\10\0\0\0\1\0\7,\240\0\0\0004"..., 936) = 936
mmap2(NULL, 1417216, PROT_READ|PROT_EXEC, MAP_PRIVATE, 3, 0) = 0x77ca0000
mmap2(0x77dea000, 65536, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED, 3, 0x13a000) = 0x77dea000
mmap2(0x77df8000, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_ANONYMOUS, -1, 0) = 0x77df8000
close(3)                                = 0
open("/lib/libgcc_s.so.1", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = 3
fcntl64(3, F_SETFD, FD_CLOEXEC)         = 0
fstat64(3, {st_mode=S_IFREG|0644, st_size=78080, ...}) = 0
read(3, "\177ELF\1\2\1\0\0\0\0\0\0\0\0\0\0\3\0\10\0\0\0\1\0\0'\320\0\0\0004"..., 936) = 936
mmap2(NULL, 147456, PROT_READ|PROT_EXEC, MAP_PRIVATE, 3, 0) = 0x77c7c000
mmap2(0x77c9e000, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED, 3, 0x12000) = 0x77c9e000
close(3)                                = 0
open("/lib/libv4lconvert.so.0", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = -1 ENOENT (No such file or directory)
open("/usr/local/lib/libv4lconvert.so.0", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = -1 ENOENT (No such file or directory)
open("/usr/lib/libv4lconvert.so.0", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = 3
fcntl64(3, F_SETFD, FD_CLOEXEC)         = 0
fstat64(3, {st_mode=S_IFREG|0755, st_size=127644, ...}) = 0
read(3, "\177ELF\1\2\1\0\0\0\0\0\0\0\0\0\0\3\0\10\0\0\0\1\0\0\25\260\0\0\0004"..., 936) = 936
mmap2(NULL, 532480, PROT_READ|PROT_EXEC, MAP_PRIVATE, 3, 0) = 0x77bfa000
mmap2(0x77c28000, 344064, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED, 3, 0x1e000) = 0x77c28000
mmap2(0x77c2a000, 335872, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_ANONYMOUS, -1, 0) = 0x77c2a000
close(3)                                = 0
mprotect(0x77e11000, 4096, PROT_READ)   = 0
mprotect(0x77dea000, 40960, PROT_READ)  = 0
mprotect(0x77c9e000, 4096, PROT_READ)   = 0
mprotect(0x77c28000, 4096, PROT_READ)   = 0
mprotect(0x459000, 4096, PROT_READ)     = 0
brk(NULL)                               = 0x1bd0000
brk(0x1bd5000)                          = 0x1bd5000
--- SIGSEGV {si_signo=SIGSEGV, si_code=SEGV_MAPERR, si_addr=NULL} ---
+++ killed by SIGSEGV +++
Segmentation fault
root@Openwrt:~# 

```

neheb, 2020-07-26:
That last one shows a null pointer being passed to malloc. Lovely...

Some googling reveals this: https://gitlab.alpinelinux.org/alpine/aports/-/issues/7047

It seems when I updated the package I removed the patch. Will investigate...

closed 2020-07-27 by neheb

mgondium, 2020-07-28:
Confirmed. Thanks!