openwrt/packages #12933: ksmbd-server: Filesizes are 0 on windows

fvgithub, 2020-07-26:
Maintainer: @Andy Walsh <andy.walsh44+github@gmail.com>
Environment: TP-Link Archer A7 V5 19.7.3

Description: Filesizes are displayed 0 on a windows 10 machine

I am using ksmbd-server 3.2.1-1 to access the openwrt router's filesystem from windows.
Sharing appears to work, the file content is accessible but the filesizes reported to windows are 0 bytes.
This prevents certain tools from working (sqlite3)

neheb, 2020-07-26:
version 3.2.4 was released recently. Need a PR to update it.

fvgithub, 2020-07-27:
I am rather new to github.
Should I simply wait and check in my openwrt when the new version is available?
Or are you suggesting I take the 3.2.4 release from somewhere?
If the latter, do you have a link to the repo?
Thanks,
Frederic

neheb, 2020-07-27:
either me or @Andy2244 will get to it.

Andy2244, 2020-07-27:
> either me or @Andy2244 will get to it.

For snapshots/master releases i will update, yet i still have no idea what to-do with 19.07, since #12599 did not really give me a answer?

@fvgithub You can try snapshots, which already has a much newer version or switch back to samba4 until i have a working update solution for 19.07.

Neustradamus, 2021-03-17:
@fvgithub: It is solved or not?

fvgithub, 2021-03-18:
Hello,

I have not recceived any notification up until now.

I will need to test this.

I'll get back to you

Regards,

Frederic

fvgithub, 2021-03-20:
@Neustradamus: What version should I test exactly and where do I find it?
I just upgraded to openwrt 19.7.7
However I cannot install ksmbd-server (opkg install ksmbd-server)

Is the package available for my TP-Link Archer A7.
Do I need a particular distfeed?

src/gz openwrt_core http://downloads.openwrt.org/releases/19.07.7/targets/ath79/generic/packages
src/gz openwrt_kmods http://downloads.openwrt.org/releases/19.07.7/targets/ath79/generic/kmods/4.14.221-1-b84a5a29b1d5ae1dc33ccf9ba292ca1d
src/gz openwrt_base http://downloads.openwrt.org/releases/19.07.7/packages/mips_24kc/base
src/gz openwrt_freifunk http://downloads.openwrt.org/releases/19.07.7/packages/mips_24kc/freifunk
src/gz openwrt_luci http://downloads.openwrt.org/releases/19.07.7/packages/mips_24kc/luci
src/gz openwrt_packages http://downloads.openwrt.org/releases/19.07.7/packages/mips_24kc/packages
src/gz openwrt_routing http://downloads.openwrt.org/releases/19.07.7/packages/mips_24kc/routing
src/gz openwrt_telephony http://downloads.openwrt.org/releases/19.07.7/packages/mips_24kc/telephony

Neustradamus, 2021-03-20:
@fvgithub: Thanks for your reply! :)
@neheb, @Andy2244: Please look the comment of @fvgithub.

fvgithub, 2021-03-21:
FYI:

root@wlan:~# opkg install ksmbd-server
Installing ksmbd-server (3.3.4-1) to root...
Downloading http://downloads.openwrt.org/releases/19.07.7/packages/mips_24kc/packages/ksmbd-server_3.3.4-1_mips_24kc.ipk
Collected errors:
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for ksmbd-server:
 *      kmod-crypto-arc4
 * opkg_install_cmd: Cannot install package ksmbd-server.

neheb, 2021-03-21:
That was a mistake that crept in the backport. I think you can install with opkg ---force-depends

fvgithub, 2021-03-22:
This issue is not solved for me

root@wlan:/wpub# ls -la
drwxrwxrwx    2 root     root             0 Mar 22 07:43 .
drwxr-xr-x    1 root     root             0 Mar 22 07:40 ..
-rw-r--r--    1 nobody   nogroup    1200069 Mar 22 07:44 klad_eendraadschema_benedenverdieping.pdf
-rw-r--r--    1 root     root             5 Mar 22 07:40 test.txt

λ dir \\wlan\wpub
 Volume in drive \\wlan\wpub is wpub
 Volume Serial Number is BC3A-C512

 Directory of \\wlan\wpub

15/02/2021  16:22    <DIR>          .
15/02/2021  16:22    <DIR>          ..
21/11/2020  16:10                 0 klad_eendraadschema_benedenverdieping.pdf
22/03/2021  07:40                 0 test.txt
               2 File(s)              0 bytes
               2 Dir(s)       8,491,008 bytes free

root@wlan:/wpub# opkg list | grep -i smb
kmod-fs-ksmbd - 4.14.221+3.3.4-1
ksmbd-server - 3.3.4-1
luci-app-ksmbd - git-21.079.58580-41ab871-1

root@wlan:/wpub# uname -a
Linux wlan 4.14.221 #0 Mon Feb 15 15:22:37 2021 mips GNU/Linux

Model  TP-Link Archer A7 v5
Architecture  Qualcomm Atheros QCA956X ver 1 rev 0
Firmware Version  OpenWrt 19.07.7 r11306-c4a6851c72 / LuCI openwrt-19.07 branch git-21.044.30835-34e0d65

neheb, 2021-03-22:
Right. That's 3.3.4.

Neustradamus, 2021-04-13:
@fvgithub: Solved now with last version?

fvgithub, 2021-04-13:
@Neustradamus :Which version should I test? 
I checked latest stable openwrt but this is still at 19.07.7
I upgraded ksmbd-server to the latest available 3.3.8.
**The issue is not fixed with this version.**

 -----------------------------------------------------
 OpenWrt 19.07.7, r11306-c4a6851c72
 -----------------------------------------------------
root@wlan:~# opkg list | grep -i smb
kmod-fs-ksmbd - 4.14.221+3.3.4-1
ksmbd-server - 3.3.8-1
luci-app-ksmbd - git-21.101.59933-c56d9f2-1

on router:

root@wlan:/wpub# ls -la test.txt
-rw-r--r--    1 root     root            13 Apr 13 06:52 test.txt
root@wlan:/wpub# cat test.txt
FILE CONTENT

on windows:

λ dir \\192.168.4.1\wpub\test.txt
 Volume in drive \\192.168.4.1\wpub is wpub
 Volume Serial Number is BC3A-C512

 Directory of \\192.168.4.1\wpub

13/04/2021  06:52                 0 test.txt
               1 File(s)              0 bytes
               0 Dir(s)       7,004,160 bytes free

C:\Users\fvanc
λ type \\192.168.4.1\wpub\test.txt

neheb, 2021-04-13:
This issue probably belongs upstream.

fvgithub, 2021-04-13:
Good question
It's the root file system but this is overlayfs
/dev/mtdblock5 on /overlay type jffs2 (rw,noatime)
overlayfs:/overlay on / type overlay (rw,noatime,lowerdir=/,upperdir=/overlay/upper,workdir=/overlay/work)

Looks like the overlayfs causes the issue.
I created a second share directly in /overlay directory and this works without an issue

neheb, 2021-04-13:
Either that or jffs2. The latter has caused issues before.

Typical usage of samba uses external USB storage with some proper filesystem.

fvgithub, 2021-04-13:
FYI: My test with a share directly on /overlayfs (which is jffs2) works fine so overlayfs is definately a factor (possibly in combination with jffs2?)
Thanks again for the hint

Neustradamus, 2021-08-01:
19.07.x: https://github.com/openwrt/packages/pull/15108
master: https://github.com/openwrt/packages/pull/16245

Neustradamus, 2021-08-01:
@fvgithub: Can you create a dump/cap of the problem?

Neustradamus, 2021-09-12:
@fvgithub: Any news?

fvgithub, 2021-09-13:
I upgraded my openwrt modules but the issue is still there
I have OpenWrt 19.07.7 r11306-c4a6851c72 with
ksmbd-server    3.3.9-1
luci-app-ksmbd   git-21.252.54297-fc2ff4d-1
FYI: One warning during upgrade 'kmod-crypto-arc4 not available'

On the router it looks fine to me

root@wlan:~# ls -la /wpub/
drwxrwxrwx    2 root     root             0 May  2 20:03 .
drwxr-xr-x    1 root     root             0 Mar 22 07:40 ..
-rw-r--r--    1 root     root           479 May  2 20:03 elecverbruik.txt
-rw-r--r--    1 root     root            13 Apr 13 06:52 test.txt
lrwxrwxrwx    1 root     root            23 Apr 13 08:03 test2.txt -> /overlay/test/test2.txt

On windows the directory content does not even show (file content also empy)

λ dir \\192.168.4.1\wpub
 Volume in drive \\192.168.4.1\wpub is wpub
 Volume Serial Number is BC3A-C512

 Directory of \\192.168.4.1\wpub

22/03/2021  08:40    <DIR>          .
22/03/2021  08:40    <DIR>          ..
               0 File(s)              0 bytes
               2 Dir(s)       1,368,064 bytes free

λ type \\192.168.4.1\wpub\test.txt
λ
λ type \\192.168.4.1\wpub\filedoesnotexist.txt
The system cannot find the file specified.

Note: The share not on the overlayfs filesystem works

If you need specific files, please let me know how I should collect them

Neustradamus, 2021-09-15:
@neheb, @Andy2244, @mmakassikis, @namjaejeon, @hclee, @launius, @sergey-senozhatsky: What do you think?

Neustradamus, 2022-08-01:
@fvgithub: Your problem has been solved?

fvgithub, 2022-08-02:
Hi, I am still on ksmbd-server 3.3.9-1 on openwrt 19.07
The situation has not changed.
I have found an alternative for smbd so I do not require this to work anymore.

Neustradamus, 2022-12-29:
@fvgithub: Have you upgraded your router?

Neustradamus, 2023-08-26:
@mmakassikis: Do not forget it.

Neustradamus, 2023-10-13:
@fvgithub: Are you here?

Can you create a dump/cap of the problem?

The goal is to found a solution for this bug...

Neustradamus, 2023-10-20:
@fvgithub or others, can you create a ticket upstream?
- https://github.com/cifsd-team/ksmbd/issues

It is possible to do a dump/pcap too?

Thanks in advance.

Neustradamus, 2023-12-18:
@neheb and others: For information, there are two new repos in more main ksmbd source code:
- https://github.com/namjaejeon/stable-linux-6.1-ksmbd
- https://github.com/namjaejeon/stable-linux-5.15-ksmbd // 23.05.x

Maybe you can request for 5.10.x (22.03.x), 6.6.x?

For kmsbd-tools too?

Source:
- https://github.com/cifsd-team/ksmbd/issues/599#issuecomment-1855653856

fvgithub, 2025-10-29:
I added the maintainer
I don't know how to send a ticket upstream.

Neustradamus, 2025-10-30:
@fvgithub: I think that you can do it here too: https://github.com/namjaejeon/ksmbd/issues.