openwrt/packages #12938: Nginx problem under master branch!!

zxlhhyccc, 2020-07-26:
[image]

neheb, 2020-07-26:
ping @peter-stadler

peter-stadler, 2020-07-26:
Thank you very much.

peter-stadler, 2020-07-26:
Does commenting `PROVIDES:=nginx-mod-luci-ssl` as in the draft PR solves it? (I cannot test it right now)

neheb, 2020-07-26:
yes

peter-stadler, 2020-07-27:
Thank you :-) I created a new PR as I did not find a way to squash commits on github's online editor (I have only access to my main system in a week again).

garypang13, 2020-07-28:
```
2020-07-28T04:04:42.0442810Z Collected errors:
2020-07-28T04:04:42.0444723Z  * satisfy_dependencies_for: Cannot satisfy the following dependencies for luci-ssl-nginx:
2020-07-28T04:04:42.0446928Z  * 	nginx-util (>= 1.4-2)
2020-07-28T04:04:42.0448520Z  * opkg_install_cmd: Cannot install package luci-ssl-nginx.
2020-07-28T04:04:42.0449776Z make[2]: *** [package/install] Error 255
2020-07-28T04:04:42.0451939Z package/Makefile:65: recipe for target 'package/install' failed
```

peter-stadler, 2020-07-28:
Hmm, right now I can test it only on my router (where it works). I tried to avoid this problem by adding the dependency statement to the same Makefile containing the depended package, too. What is `opkg list | grep nginx-util` showing?

rmandrad, 2020-07-30:
Why are you releasing code you can't test!!!! can you just undo what you have done until you properly test ?

peter-stadler, 2020-07-30:
I did quite some testing before making the PR. I am sorry that I overlooked an error for `make menuconfig`. This problem should be solved now: the change was just to remove the `PROVIDES:=nginx-mod-luci-ssl` (I could not test that change, but I think it should not make a problem elsewhere).

Unfortunately I cannot use my main system (where I do the compilation and testing in a VM) until Monday. I did the installation test again on my router (two days ago) and there I did not see the dependency problem for `nginx-util (>= 1.4-2)` ...

Can you give me some details on the problem: 
1. On which system do you experience it?
2. Does it happen on a fresh install and/or an upgrade of `luci-ssl-nginx`?
3. Can you post the output of `opkg list | grep nginx-util` when you see the problem?

rmandrad, 2020-07-31:
It is now working ... I had to unselect all of the Luci-nginx + Nginx-* web packages and reselect back again ... I was having issues while installing the packages and getting errors with dependencies

peter-stadler, 2020-07-31:
I think I understand it now. Thank you :-) I will make another PR (but I cannot test it before Monday) …

rmandrad, 2020-08-01:
Thank you - note also I had to set to CONFIG_PACKAGE_nginx-mod-luci-ssl=n and go back to menuconfig to put back luci-

verofin, 2020-08-02:
I cloned master in the last 8 hours and when I make my project with luci-nginx I get this error

Collected errors:
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for nginx-ssl:
 * 	nginx-util (>= 1.4-2)
 * opkg_install_cmd: Cannot install package nginx-ssl.
package/Makefile:65: recipe for target 'package/install' failed
make[2]: *** [package/install] Error 255

peter-stadler, 2020-08-03:
Thank you, too. Now, I tested it and marked the PR ready for review.
(For testing it before it is merged into master, you can use the linked branch https://github.com/peter-stadler/packages/tree/patch-2 …)

closed 2020-08-14 by BKPepe