openwrt/packages #12958: modemmanager: 1.14 build fails due to autoreconf forced

aleksander0m, 2020-07-28:
Maintainer: @nickberry17 
Environment: Attempting to build MM 1.14 on top of 19.07 branch

Description: The ModemManager 1.14 release build-depends on autoconf-archive when building from git. This requirement doesn't exist when building from a source tarball because the configure script is already included in the source tarball. In the openwrt build setup, though, the package has `PKG_FIXUP:=autoreconf`, and that ends up triggering a full autoreconf to regenerate the configure script, ignoring the one shipped in the source tarball. This ends up with the build requiring autoconf-archive or otherwise some m4 macros would be undefined:

```
checking whether the mips-openwrt-linux-musl-gcc linker (mips-openwrt-linux-musl-ld) supports shared libraries... yes
checking whether -lc should be explicitly linked in... no
checking dynamic linker characteristics... GNU/Linux ld.so
checking how to hardcode library paths into programs... immediate
checking whether stripping libraries is possible... yes
checking if libtool supports shared libraries... yes
checking whether to build shared libraries... yes
checking whether to build static libraries... no
./configure: line 12855: syntax error near unexpected token `git-directory'
./configure: line 12855: `AX_IS_RELEASE(git-directory)'
Makefile:142: recipe for target '/home/miwire/openwrt/build_dir/target-mips_24kc_musl/ModemManager-1.14.0/.configured_68b329da9893e34099c7d8ad5cb9c940' failed
make[3]: *** [/home/miwire/openwrt/build_dir/target-mips_24kc_musl/ModemManager-1.14.0/.configured_68b329da9893e34099c7d8ad5cb9c940] Error 2
make[3]: Leaving directory '/home/miwire/openwrt/feeds/packages/net/modemmanager'
time: package/feeds/packages/modemmanager/compile#8.95#1.10#10.58
package/Makefile:111: recipe for target 'package/feeds/packages/modemmanager/compile' failed
make[2]: *** [package/feeds/packages/modemmanager/compile] Error 2
make[2]: Leaving directory '/home/miwire/openwrt'
package/Makefile:107: recipe for target '/home/miwire/openwrt/staging_dir/target-mips_24kc_musl/stamp/.package_compile' failed
make[1]: *** [/home/miwire/openwrt/staging_dir/target-mips_24kc_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/miwire/openwrt'
/home/miwire/openwrt/include/toplevel.mk:225: recipe for target 'world' failed
make: *** [world] Error 2
```

neheb, 2020-07-28:
I'd urge upstream to switch to CMake or meson. I'm tired of dealing with autotools.

aleksander0m, 2020-07-28:
This isn't really an issue in autotools, it's a decision of the upstream maintainer (me :D) to depend on autoconf-archive for non-source tarball builds. It should be an easy fix IMO; either always build-depend on autoconf-archive or otherwise avoid forcing the unneeded autoreconf (unless this autoreconf is required for some reason I don't know?)

aleksander0m, 2020-07-28:
The autoreconf was explicitly requested, but it was also anyway implicitly triggered because the Makefile is modifying `configure.ac` in the source tarball like this:
```
        $(SED) 's|^\(GLIB_MKENUMS\)=.*|\1=$(STAGING_DIR_HOSTPKG)/bin/glib-mkenums|' \
                $(PKG_BUILD_DIR)/configure.ac
        $(SED) 's|^\(GDBUS_CODEGEN\)=.*|\1=$(STAGING_DIR_HOSTPKG)/bin/gdbus-codegen|' \
                $(PKG_BUILD_DIR)/configure.ac
```

Any idea why are these updates needed? Both libmbim and libqmi also required glib-mkenums, and I don't see any such update in their corresponding Makefiles.

micmac1, 2020-07-29:
> The autoreconf was explicitly requested, but it was also anyway implicitly triggered because the Makefile is modifying `configure.ac` in the source tarball like this:
>
> ```
>         $(SED) [...]

See fa8ad6e. Builds fail without as the target binaries are used during build. :)

aleksander0m, 2020-07-29:
@micmac1 does that mean we're forcing to use the host package glib tools in this way, instead of using the target package glib tools? And if so, does it really matter much? And if I'm not mistaken libqmi and libmbim also look for glib-mkenums in the same way as ModemManager; why do they not have the same build issue?

micmac1, 2020-07-29:
During build-phase! I imagine using target binaries would usually fail because they're for a different architecture.

Am July 29, 2020 7:14:00 AM UTC schrieb Aleksander Morgado <notifications@github.com>:
> @micmac1 does that mean we're forcing to use the host package glib tools in this way, instead of using the target package glib tools? And if so, does it really matter much? And if I'm not mistaken [...]

aleksander0m, 2020-07-29:
@micmac1 oh, but both glib-mkenums and gdbus-codegen are python tools, there should be no problem in using the generated target programs. It would be totally equivalent to e.g. what the libqmi and libmbim builds do themselves; they also carry their own qmi-codegen and mbim-codegen python tools, and these tools are used to build the source files that will be later on compiled in the target. From my point of view, all glib-mkenums, gdbus-codegen, qmi-codegen and mbim-codegen can be treated in the same way (and we don't have host libqmi or host libmbim packages either)

aleksander0m, 2020-07-29:
Actually... latest libqmi and libmbim ship their own "qmi-mkenums" and "mbim-mkenums" in the last releases, which are an exact copy of the glib-mkenums python program but with a patch I wrote to support reporting "Since: " version tags in the generated methods. This means that both libqmi and libmbim run, during build time, all these python programs (qmi-codegen, mbim-codegen, qmi-mkenums, mbim-mkenums) that are shipped in the source tarball.

I would totally suggest to avoid requiring the use of the glib2 host package glib-mkenums or gdbus-codegen tools, and just use the ones built in the glib2 target, because there will be no difference between them (apart from e.g. prefix related hardcoded paths obviously)

micmac1, 2020-07-29:
Maybe best would be not to rely on pkg-config here and simply call the scripts directly. That would use the host's PATH.

aleksander0m, 2020-07-29:
When building MM against a glib2 package installed in a non-standard directory (e.g. I usually build glib2 git master and install in /opt/test) I just need to provide the correct PKG_CONFIG_PATH so that the correct glib2 headers and libraries are used when building and linking. I also know that the correct glib2 glib-mkenums and gdbus-codegen are used due to having them configured with pkg-config. I mean, there are benefits in having them setup via pkg-config instead of wildly available in the PATH.

My suggestion here would be to decide whether we're fine running code-generator python programs available in the target or not, and based on that decision see how it can be solved:

- If we're fine running code generator python programs available in the target, we should modify MM to avoid the autoreconf and just rely on running glib-mkenums and gdbus-codegen in the target to generate code.
- If we're not fine running code generator python programs available in the target we should keep the MM modification to run glib-mkenums and gdbus-codegen from the host glib2 package BUT then we should also have new libqmi and libmbim host packages so that the qmi-codegen, mbim-codegen, qmi-mkenums, mbim-mkenums programs are run from the host package instead of the target package.

If you ask me, given that the host-package and target-package glib-mkenums and gdbus-codegen programs are going to be virtually the same, I would just run them from the target package and save us a lot of unneeded work.

closed 2020-07-30 by neheb

wicadmin, 2021-12-07:
As ModemManager, libmbim, libqmi etc have now moved to meson, has anyone yet created an OpenWRT Makefile that doesn't fail with meson?