openwrt/packages #12968: List of "problematic" architectures

aparcar, 2020-07-29:
Hi, I'm currently building a CI which tests changed packages for multiple architectures. I'm not familiar with the lower level implementation of these architectures, but understand that a package may compile fine on one but break badly on another architecture. Could someone please help with with a short list of top 5 popular and top 5 breaking architectures to test within the CI?

neheb, 2020-07-29:
ARC is a constant source of breakage since it uses a different libc (uClibc-ng).
PowerPC occasionally breaks, either because of no FPU or because of -lssp_nonshared
ARM64 has issues with PIC.
x86_64 is used by @stintel , who constantly complains of his builds being broken.
some of the older ARM platforms break sometimes.

aparcar, 2020-07-29:
Amazing that's what I wanted to know :)!

neheb, 2020-07-30:
ARM64 fails occasionally with PKG_ASLR_PIE. Not sure if that can be turned on.

It would be great to also compile without deprecated OpenSSL APIs if possible.

aparcar, 2020-07-30:
> It would be great to also compile without deprecated OpenSSL APIs if possible.

Not sure how to do that. I'm currently working on a CI for openwrt.git, however it could be used for packages.git, too.

neheb, 2020-07-30:
It’s a Kconfig option. OPENSSL_WITH_DEPRECATED I believe.

neheb, 2020-09-19:
Not valid anymore I believe.

closed 2020-09-19 by neheb