openwrt/packages #12972: mwan3: Multi-wan copying route problem.

CsHeng, 2020-07-30:
Maintainer: @feckert 
Environment: [OpenWrt 19.07.3 r11063-85e04e9f46](https://downloads.openwrt.org/releases/19.07.3/targets/x86/64/)
mwan3: [mwan3_2.8.12-1_all.ipk](https://downloads.openwrt.org/releases/19.07.3/packages/x86_64/packages/mwan3_2.8.12-1_all.ipk)
Description:

Multi-wan copying route problem. 

by checking the [mwan3.sh source file](https://github.com/openwrt/packages/blob/facf8ea299b185a82ae803d68e2b54a9599d8c14/net/mwan3/files/lib/mwan3/mwan3.sh):

**i found `$IP4 route add table $tid $line` will only add the first line in `ip route list table main` with same subnet/proto kernel**

related mwan3 source code below
```bash
# return true(=0) if has any mwan3 interface enabled
# otherwise return false
mwan3_rtmon_ipv4()
{
	local tid=1
	local idx=0
	local ret=1
	local tbl=""
	mkdir -p /tmp/mwan3rtmon
	($IP4 route list table main  | grep -v "^default\|linkdown" | sort -n; echo empty fixup) >/tmp/mwan3rtmon/ipv4.main
	while uci get mwan3.@interface[$idx] >/dev/null 2>&1 ; do
		idx=$((idx+1))
		tid=$idx
		[ "$(uci get mwan3.@interface[$((idx-1))].family)" = "ipv4" ] && {
			tbl=$($IP4 route list table $tid 2>/dev/null)
			if echo "$tbl" | grep -q ^default; then
				(echo "$tbl"  | grep -v "^default\|linkdown" | sort -n; echo empty fixup) >/tmp/mwan3rtmon/ipv4.$tid
				cat /tmp/mwan3rtmon/ipv4.$tid | grep -v -x -F -f /tmp/mwan3rtmon/ipv4.main | while read line; do
					$IP4 route del table $tid $line
				done
				cat /tmp/mwan3rtmon/ipv4.main | grep -v -x -F -f /tmp/mwan3rtmon/ipv4.$tid | while read line; do
					$IP4 route add table $tid $line
				done
			fi
		}
		if [ "$(uci get mwan3.@interface[$((idx-1))].enabled)" = "1" ]; then
			ret=0
		fi
	done
	rm -f /tmp/mwan3rtmon/ipv4.*
	return $ret
}
```

so, assume that I have 2 pppoe line, and they share the same gateway, it will only add the first one to table 1/2

my real sample:

```bash
root@OpenWrt-GW:~# ip route list table main
default via 113.119.164.1 dev pppoe-adsl1 proto static metric 1
default via 113.119.164.1 dev pppoe-adsl2 proto static metric 2
113.119.164.1 dev pppoe-adsl1 proto kernel scope link src xx.xx.xx.xx
113.119.164.1 dev pppoe-adsl2 proto kernel scope link src xx.xx.xx.xx
192.168.0.0/21 dev br-lan proto kernel scope link src 192.168.2.101
192.168.255.0/24 dev bonding-bond0 proto kernel scope link src 192.168.255.1

root@OpenWrt-GW:~# ip route list table 1
default via 113.119.164.1 dev pppoe-adsl1 metric 1
113.119.164.1 dev pppoe-adsl1 proto kernel scope link src xx.xx.xx.xx
192.168.0.0/21 dev br-lan proto kernel scope link src 192.168.2.101
192.168.255.0/24 dev bonding-bond0 proto kernel scope link src 192.168.255.1

root@OpenWrt-GW:~# ip route list table 2
default via 113.119.164.1 dev pppoe-adsl2 metric 2
113.119.164.1 dev pppoe-adsl1 proto kernel scope link src xx.xx.xx.xx
192.168.0.0/21 dev br-lan proto kernel scope link src 192.168.2.101
192.168.255.0/24 dev bonding-bond0 proto kernel scope link src 192.168.255.1

```

table 1 missing pppoe-adsl2 route

and table 2 missing pppoe-adsl1 route

if I manually add the missing route `ip route add table 1 113.119.164.1 dev pppoe-adsl2 proto kernel scope link src xx.xx.xx.xx`, an error will occur: `RTNETLINK answers: File exists`

**but... should this be `iproute`'s feature or just by design (that we can't add route with same subnet which proto is kernel link) ?**

**will this situation cause any bug or we could just ignore it..**

(
actually I have bond0 ipaddress set to 192.168.2.1 earlier, and when I start mwan3, I f**ked up. (i have luci-proto-bonding to setup bond)
192.168.0.0/21 dev br-lan proto kernel scope link src 192.168.2.101
192.168.0.0/21 dev bonding-bond0 proto kernel scope link src 192.168.2.1
)

please help.

aaronjg, 2020-07-31:
copying my response to #11917 here.

>    but... should this be iproute's feature or just by design (that we can't add route with same subnet which proto is kernel link) ?

Yes, this is by design, you can't add two routes to the same destination with the same metric. How would the kernel know which one to use?

>    will this situation cause any bug or we could just ignore it..

Potentially, yes.
Traffic to 113.119.164.1 that mwan3 wants to route out of pppoe-adsl2 will actually go out of pppoe-adsl1 instead.

I am working on a somewhat extensive refactoring of mwan3 for some speed improvements, and I address this. Code should be ready in a week or two.

If you are brave and want to help beta test, you can try it out here. Please let me know if you find any issues.

https://github.com/aaronjg/openwrt-packages/tree/mwan3-startup-performance

CsHeng, 2020-07-31:
> copying my response to #11917 here.
>
>
> Yes, this is by design, you can't add two routes to the same destination with the same metric. How would the kernel know which one to use?
>
>
> Potentially, yes. [...]

I have above setup and run for up to 48 hours, it seems that the missing route is not a big deal...(traffic is not so equally divided to both line, but causing nothing wrong compare to previous mwan3 version 2.6.8)

Total traffic by statistics below(roughly): (every 24 hours)
> pppoe-adsl1: 400GB download/ 200GB upload first day, and 450GB download/ 180GB upload the second day.
> pppoe-adsl2: 450GB download/ 190GB upload first day, and 420GB download/250GB upload the second day.

btw, my setup is equally load-balance two 1000Mbps/100Mbps pppoe line.

Cause that's my office's router, i'll carefully seeking any chance to try the new package out.

closed 2020-08-19 by feckert