openwrt/packages #12973: mariadb: feature request: support lz4 compression

alex9434, 2020-07-30:
Maintainer: @miska 
Environment: OpenWrt Snapshot

Description: Feature request: support LZ4 compression (and potentially bzip2) in mariadb

Dear @miska,

Thank you very much for taking over as maintainer for mariadb. I would like to compile mariadb with support for LZ4 compression because that speeds-up operations esp. with SDD and SD cards.

I have tried manually to set the options in the build-file from OFF to ON and include the LZ4 and BZIP2 libraries in the build process. Unfortunately that did not work. I get the following error message
```
Package mariadb-server is missing dependencies for the following libraries:
libbz2.so.1.0
liblz4.so.1
```
Any ideas how I can add the missing dependencies to the build? Any chance you would include these features in the build process (maybe even as selectable options)? Do I have to include both libraries in the mariadb build or can I use the system-wide LZ4 and BZIP2 libraries?

Thank you,
alex

neheb, 2020-08-02:
You need to add +libbzip2 and +liblz4 to DEPENDS

alex9434, 2020-08-05:
Thank you, that worked. Closing this now. But maybe you would like to consider adding a selection to the Makefile.

closed 2020-08-05 by alex9434