openwrt/packages #12981: glib2.65 fails to build under macos

ldir-EDB0, 2020-07-31:
glib2 package fails to build since the bump to glib2.65

make[3]: Entering directory '/Users/kevin/wrt/feeds/packages/devel/meson'
make[3]: Leaving directory '/Users/kevin/wrt/feeds/packages/devel/meson'
time: package/feeds/packages/meson/host-compile#0.17#0.16#0.41
make[3]: Entering directory '/Users/kevin/wrt/feeds/packages/libs/glib2'
rm -f /Users/kevin/wrt/build_dir/target-x86_64_musl/glib-2.65.0/.configured_*
rm -f /Users/kevin/wrt/staging_dir/target-x86_64_musl/stamp/.glib2_installed
/Users/kevin/wrt/staging_dir/host/bin/sed -e "s|@CC@|'gcc',|" -e "s|@CXX@|'g++',|" -e "s|@PKGCONFIG@|/Users/kevin/wrt/staging_dir/host/bin/pkg-config|" -e "s|@CFLAGS@|'-O2', '-I/Users/kevin/wrt/staging_dir/host/include', '-I/Users/kevin/wrt/staging_dir/hostpkg/include', '-I/Users/kevin/wrt/staging_dir/target-x86_64_musl/host/include', '-I/Users/kevin/wrt/staging_dir/host/include', '-I/Users/kevin/wrt/staging_dir/hostpkg/include', '-I/Users/kevin/wrt/staging_dir/target-x86_64_musl/host/include',|" -e "s|@CXXFLAGS@|'-I/Users/kevin/wrt/staging_dir/host/include', '-I/Users/kevin/wrt/staging_dir/hostpkg/include', '-I/Users/kevin/wrt/staging_dir/target-x86_64_musl/host/include',|" -e "s|@LDFLAGS@|'-L/Users/kevin/wrt/staging_dir/host/lib', '-L/Users/kevin/wrt/staging_dir/hostpkg/lib', '-L/Users/kevin/wrt/staging_dir/target-x86_64_musl/host/lib', '-Wl,-rpath=/Users/kevin/wrt/staging_dir/hostpkg/lib',|" -e "s|@PREFIX@|/Users/kevin/wrt/staging_dir/hostpkg|" < /Users/kevin/wrt/staging_dir/hostpkg/lib/meson/openwrt-native.txt.in > /Users/kevin/wrt/build_dir/target-x86_64_musl/glib-2.65.0/openwrt-native.txt
/Users/kevin/wrt/staging_dir/host/bin/sed -e "s|@CC@|'x86_64-openwrt-linux-musl-gcc',|" -e "s|@CXX@|'x86_64-openwrt-linux-musl-g++',|" -e "s|@AR@|x86_64-openwrt-linux-musl-gcc-ar|" -e "s|@STRIP@|x86_64-openwrt-linux-musl-strip|" -e "s|@NM@|x86_64-openwrt-linux-musl-gcc-nm|" -e "s|@LD@|x86_64-openwrt-linux-musl-ld|" -e "s|@PKGCONFIG@|/Users/kevin/wrt/staging_dir/host/bin/pkg-config|" -e "s|@CFLAGS@|'-O2', '-pipe', '-march=btver2', '-fno-caller-saves', '-fno-plt', '-fhonour-copts', '-Wno-error=unused-but-set-variable', '-Wno-error=unused-result', '-ffile-prefix-map=/Users/kevin/wrt/build_dir/target-x86_64_musl/glib-2.65.0=glib-2.65.0', '-Wformat', '-Werror=format-security', '-fpic', '-fstack-protector', '-D_FORTIFY_SOURCE=1', '-Wl,-z,now', '-Wl,-z,relro', '-I/Users/kevin/wrt/staging_dir/target-x86_64_musl/usr/lib/libiconv-stub/include', '-I/Users/kevin/wrt/staging_dir/target-x86_64_musl/usr/lib/libintl-stub/include', '-ffunction-sections', '-fdata-sections', '-I/Users/kevin/wrt/staging_dir/target-x86_64_musl/usr/include', '-I/Users/kevin/wrt/staging_dir/toolchain-x86_64_gcc-10.1.0_musl/usr/include', '-I/Users/kevin/wrt/staging_dir/toolchain-x86_64_gcc-10.1.0_musl/include/fortify', '-I/Users/kevin/wrt/staging_dir/toolchain-x86_64_gcc-10.1.0_musl/include', '-I/Users/kevin/wrt/staging_dir/target-x86_64_musl/usr/lib/libiconv-stub/include', '-I/Users/kevin/wrt/staging_dir/target-x86_64_musl/usr/lib/libintl-stub/include',|" -e "s|@CXXFLAGS@|'-O2', '-pipe', '-march=btver2', '-fno-caller-saves', '-fno-plt', '-fhonour-copts', '-Wno-error=unused-but-set-variable', '-Wno-error=unused-result', '-ffile-prefix-map=/Users/kevin/wrt/build_dir/target-x86_64_musl/glib-2.65.0=glib-2.65.0', '-Wformat', '-Werror=format-security', '-fpic', '-fstack-protector', '-D_FORTIFY_SOURCE=1', '-Wl,-z,now', '-Wl,-z,relro', '-I/Users/kevin/wrt/staging_dir/target-x86_64_musl/usr/lib/libiconv-stub/include', '-I/Users/kevin/wrt/staging_dir/target-x86_64_musl/usr/lib/libintl-stub/include', '-ffunction-sections', '-fdata-sections', '-I/Users/kevin/wrt/staging_dir/target-x86_64_musl/usr/include', '-I/Users/kevin/wrt/staging_dir/toolchain-x86_64_gcc-10.1.0_musl/usr/include', '-I/Users/kevin/wrt/staging_dir/toolchain-x86_64_gcc-10.1.0_musl/include/fortify', '-I/Users/kevin/wrt/staging_dir/toolchain-x86_64_gcc-10.1.0_musl/include', '-I/Users/kevin/wrt/staging_dir/target-x86_64_musl/usr/lib/libiconv-stub/include', '-I/Users/kevin/wrt/staging_dir/target-x86_64_musl/usr/lib/libintl-stub/include',|" -e "s|@LDFLAGS@|'-L/Users/kevin/wrt/staging_dir/target-x86_64_musl/usr/lib', '-L/Users/kevin/wrt/staging_dir/target-x86_64_musl/lib', '-L/Users/kevin/wrt/staging_dir/toolchain-x86_64_gcc-10.1.0_musl/usr/lib', '-L/Users/kevin/wrt/staging_dir/toolchain-x86_64_gcc-10.1.0_musl/lib', '-fpic', '-specs=/Users/kevin/wrt/include/hardened-ld-pie.specs', '-znow', '-zrelro', '-L/Users/kevin/wrt/staging_dir/target-x86_64_musl/usr/lib/libiconv-stub/lib', '-Wl,-rpath-link=/Users/kevin/wrt/staging_dir/target-x86_64_musl/usr/lib/libiconv-stub/lib', '-L/Users/kevin/wrt/staging_dir/target-x86_64_musl/usr/lib/libintl-stub/lib', '-Wl,-rpath-link=/Users/kevin/wrt/staging_dir/target-x86_64_musl/usr/lib/libintl-stub/lib', '-Wl,--gc-sections',|" -e "s|@ARCH@|"x86_64"|" -e "s|@CPU@|"64"|" -e "s|@ENDIAN@|little|" < /Users/kevin/wrt/staging_dir/hostpkg/lib/meson/openwrt-cross.txt.in > /Users/kevin/wrt/build_dir/target-x86_64_musl/glib-2.65.0/openwrt-cross.txt
/Users/kevin/wrt/staging_dir/host/bin/python /Users/kevin/wrt/staging_dir/hostpkg/lib/meson/meson.py  --buildtype plain --native-file /Users/kevin/wrt/build_dir/target-x86_64_musl/glib-2.65.0/openwrt-native.txt --cross-file /Users/kevin/wrt/build_dir/target-x86_64_musl/glib-2.65.0/openwrt-cross.txt -Diconv=external -Dselinux=disabled -Dlibmount=disabled -Dinternal_pcre=true -Dman=false -Ddtrace=false -Dsystemtap=false -Dgtk_doc=false -Dbsymbolic_functions=true -Dforce_posix_threads=true -Dfam=false -Dinstalled_tests=false -Dnls=disabled -Doss_fuzz=disabled -Dglib_assert=false -Dglib_checks=false -Dxattr=true -Db_lto=true /Users/kevin/wrt/build_dir/target-x86_64_musl/glib-2.65.0/openwrt-build /Users/kevin/wrt/build_dir/target-x86_64_musl/glib-2.65.0/openwrt-build/..
The Meson build system
Version: 0.55.0
Source dir: /Users/kevin/wrt/build_dir/target-x86_64_musl/glib-2.65.0
Build dir: /Users/kevin/wrt/build_dir/target-x86_64_musl/glib-2.65.0/openwrt-build
Build type: cross build
Using 'PKG_CONFIG_PATH' from environment with value: '/Users/kevin/wrt/staging_dir/target-x86_64_musl/usr/lib/pkgconfig:/Users/kevin/wrt/staging_dir/target-x86_64_musl/usr/share/pkgconfig'
Project name: glib
Project version: 2.65.0

../../../../build_dir/target-x86_64_musl/glib-2.65.0/meson.build:1:0: ERROR: Compiler gcc can not compile programs.

A full log can be found at /Users/kevin/wrt/build_dir/target-x86_64_musl/glib-2.65.0/openwrt-build/meson-logs/meson-log.txt
make[3]: *** [Makefile:143: /Users/kevin/wrt/build_dir/target-x86_64_musl/glib-2.65.0/.configured_a17fb5ef857664f03cd0ce37cc5ea591] Error 1
make[3]: Leaving directory '/Users/kevin/wrt/feeds/packages/libs/glib2'
time: package/feeds/packages/glib2/compile#0.93#0.66#1.82
make[2]: *** [package/Makefile:113: package/feeds/packages/glib2/compile] Error 2
make[2]: Leaving directory '/Users/kevin/wrt'
make[1]: *** [package/Makefile:107: /Users/kevin/wrt/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/Users/kevin/wrt'
make: *** [/Users/kevin/wrt/include/toplevel.mk:235: world] Error 2

cat /Users/kevin/wrt/build_dir/target-x86_64_musl/glib-2.65.0/openwrt-build/meson-logs/meson-log.txt
Build started at 2020-07-31T06:57:54.097880
Main binary: /Applications/Xcode.app/Contents/Developer/usr/bin/python3
Build Options: -Diconv=external -Dselinux=disabled -Dlibmount=disabled -Dinternal_pcre=true -Dman=false -Ddtrace=false -Dsystemtap=false -Dgtk_doc=false -Dbsymbolic_functions=true -Dforce_posix_threads=true -Dfam=false -Dinstalled_tests=false -Dnls=disabled -Doss_fuzz=disabled -Dglib_assert=false -Dglib_checks=false -Dxattr=true -Db_lto=true -Dbuildtype=plain '--cross-file /Users/kevin/wrt/build_dir/target-x86_64_musl/glib-2.65.0/openwrt-cross.txt' '--native-file /Users/kevin/wrt/build_dir/target-x86_64_musl/glib-2.65.0/openwrt-native.txt'
Python system: Darwin
The Meson build system
Version: 0.55.0
Source dir: /Users/kevin/wrt/build_dir/target-x86_64_musl/glib-2.65.0
Build dir: /Users/kevin/wrt/build_dir/target-x86_64_musl/glib-2.65.0/openwrt-build
Build type: cross build
None of 'PKG_CONFIG_PATH_FOR_BUILD' are defined in the environment, not changing global flags.
Using 'PKG_CONFIG_PATH' from environment with value: '/Users/kevin/wrt/staging_dir/target-x86_64_musl/usr/lib/pkgconfig:/Users/kevin/wrt/staging_dir/target-x86_64_musl/usr/share/pkgconfig'
Project name: glib
Project version: 2.65.0
[... cut ...] st/lib -Wl,-rpath=/Users/kevin/wrt/staging_dir/hostpkg/lib
Sanity check compile stdout:

-----
Sanity check compile stderr:
ld: warning: directory not found for option '-L/Users/kevin/wrt/staging_dir/target-x86_64_musl/host/lib'
ld: unknown option: -rpath=/Users/kevin/wrt/staging_dir/hostpkg/lib
clang: error: linker command failed with exit code 1 (use -v to see invocation)

-----

../../../../build_dir/target-x86_64_musl/glib-2.65.0/meson.build:1:0: ERROR: Compiler gcc can not compile programs.

neheb, 2020-07-31:
@ldir-EDB0 I very recently pushed an update. Please test.

ldir-EDB0, 2020-07-31:
Still fails.  Have reverted locally so I can get a build to complete.  Hopefully available for more tests later

neheb, 2020-08-01:
Silently fixed with the latest update. I tested on Fedora 32 by replacing HOSTCC with clang.

ldir-EDB0, 2020-08-01:
Whatever you change it has worked - I got a build through on macos.  Thank you.

castiel652, 2020-08-01:
I still have similar problem compiling on Ubuntu 20.04.

Build started at 2020-08-01T09:59:27.750697
Main binary: /home/sw/openwrt/staging_dir/host/bin/python
Build Options: -Diconv=external -Dselinux=disabled -Dlibmount=disabled -Dinternal_pcre=true -Dman=false -Ddtrace=false -Dsystemtap=false -Dgtk_doc=false -Dbsymbolic_functions=true -Dforce_posix_threads=true -Dfam=false -Dinstalled_tests=false -Dnls=disabled -Doss_fuzz=disabled -Dglib_assert=false -Dglib_checks=false -Dxattr=true -Db_lto=true -Dbuildtype=plain '--cross-file /home/sw/openwrt/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/glib-2.65.0/openwrt-cross.txt' '--native-file /home/sw/openwrt/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/glib-2.65.0/openwrt-native.txt'
Python system: Linux
The Meson build system
Version: 0.55.0
Source dir: /home/sw/openwrt/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/glib-2.65.0
Build dir: /home/sw/openwrt/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/glib-2.65.0/openwrt-build
Build type: cross build
None of 'PKG_CONFIG_PATH_FOR_BUILD' are defined in the environment, not changing global flags.
Using 'PKG_CONFIG_PATH' from environment with value: '/home/sw/openwrt/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/usr/lib/pkgconfig:/home/sw/openwrt/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/usr/share/pkgconfig'
Project name: glib
Project version: 2.65.0
None of 'CFLAGS_FOR_BUILD' are defined in the environment, not changing global flags.
None of 'LDFLAGS_FOR_BUILD' are defined in the environment, not changing global flags.
None of 'CPPFLAGS_FOR_BUILD' are defined in the environment, not changing global flags.
None of 'CC_LD_FOR_BUILD' are defined in the environment, not changing global flags.
Sanity testing C compiler: gcc
Is cross compiler: False.
None of 'CC_LD_FOR_BUILD' are defined in the environment, not changing global flags.
Sanity check compiler command line: gcc /home/sw/openwrt/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/glib-2.65.0/openwrt-build/meson-private/sanitycheckc.c -o /home/sw/openwrt/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/glib-2.65.0/openwrt-build/meson-private/sanitycheckc.exe -O2 -I/home/sw/openwrt/staging_dir/host/include -I/home/sw/openwrt/staging_dir/hostpkg/include -I/home/sw/openwrt/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/host/include -I/home/sw/openwrt/staging_dir/host/include -I/home/sw/openwrt/staging_dir/hostpkg/include -I/home/sw/openwrt/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/host/include -pipe -D_FILE_OFFSET_BITS=64 -L/home/sw/openwrt/staging_dir/host/lib -L/home/sw/openwrt/staging_dir/hostpkg/lib -L/home/sw/openwrt/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/host/lib -liconv -Wl,-rpath,/home/sw/openwrt/staging_dir/hostpkg/lib
Sanity check compile stdout:

-----
Sanity check compile stderr:
/bin/ld: cannot find -liconv
collect2: error: ld returned 1 exit status

-----

../../../../build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/glib-2.65.0/meson.build:1:0: ERROR: Compiler gcc can not compile programs.

zxlhhyccc, 2020-08-01:
> I still have similar problem compiling on Ubuntu 20.04.
>
> Build started at 2020-08-01T09:59:27.750697
> Main binary: /home/sw/openwrt/staging_dir/host/bin/python
> Build Options: -Diconv=external [...]

Building soft routing is the same problem
[image]

schm0, 2020-08-01:
Same Problem on Ubuntu.

Borromini, 2020-08-01:
@ldir-EDB0 I hit this issue with master on Debian 10 too. I worked around it by unlinking iconv in the glib2 Makefile, see also https://github.com/openwrt/packages/commit/ad798995ee5d20f13929700de59ffd548fbb48c9.

neheb, 2020-08-01:
AGH now I understand why I initially had libiconv-full/host in PKG_BUILD_DEPENDS. I will fix when I can.

Borromini, 2020-08-01:
Thanks @neheb

neheb, 2020-08-01:
For those that can't wait, run make package/libiconv-full/host/compile BEFORE glib2

neheb, 2020-08-01:
Turns out, changing = to , in HOST_LDFLAGS actually broke it. The compile succeeds but meson removes that rpath. Backported upstream meson patch to prevent that.

Also added libiconv/host to PKG_BUILD_DEPENDS again.

Sorry for all the inconvenience. The initial goal was to fix uClibc-ng (and glibc) problems but it seems it broke more than it fixed.

closed 2020-08-03 by neheb