openwrt/packages #12988: Pull requests failed with 'realname' checking.

jjm2473, 2020-08-01:
What the 'realname' should I use?

https://github.com/openwrt/packages/pull/12985
https://github.com/openwrt/packages/pull/12986
https://github.com/openwrt/packages/pull/12987

yousong, 2020-08-01:
The check logic is available in .circleci/config.yml .  For the moment, just use Pinyin.

champtar, 2020-08-01:
Hi @jjm2473, sorry about this check.
The idea is to force people to put their real name and not their nickname/handle (signed off by has a legal meaning), and for now we just check for a space as a good enough signal that this a real name.
In your language, is there such a simple rule, ie 2 utf8 chars minimum or ??

jjm2473, 2020-08-01:
@champtar I tried some names, but not work. please check 
https://github.com/openwrt/packages/pull/12985/checks 
and 
https://github.com/openwrt/packages/pull/12991/checks

champtar, 2020-08-01:
We have 2 checks, DCO that check that the "signed off by" match the commit author, and another check that check that there is a space in the author name. In #12991 you changed the SOB line but not the commit author
git config user.name "Liangbin Lian"
git commit --amend --reset-author
git push --force

neheb, 2020-08-01:
> Hi @jjm2473, sorry about this check.
> The idea is to force people to put their real name and not their nickname/handle (signed off by has a legal meaning), and for now we just check for a space as a [...]
Correct

jjm2473, 2020-08-01:
@champtar It works! Thank you!

closed 2020-08-01 by jjm2473

champtar, 2020-08-01:
@jjm2473 we will fix the regexp so you can use your real name

jjm2473, 2020-08-01:
@champtar Pinyin is OK for me ;)