openwrt/packages #12990: Master branch libJPEG deleted after a lot of source packages need to rely on!!

zxlhhyccc, 2020-08-01:
WARNING: Makefile 'package/feeds/telephony/freeswitch/Makefile' has a dependency on 'libjpeg', which does not exist
WARNING: Makefile 'package/feeds/telephony/freeswitch/Makefile' has a dependency on 'libjpeg', which does not exist
WARNING: Makefile 'package/feeds/telephony/freeswitch/Makefile' has a build dependency on 'libjpeg', which does not exist
WARNING: Makefile 'package/feeds/packages/gphoto2/Makefile' has a dependency on 'libjpeg', which does not exist
WARNING: Makefile 'package/feeds/packages/graphicsmagick/Makefile' has a dependency on 'libjpeg', which does not exist
WARNING: Makefile 'package/feeds/packages/graphicsmagick/Makefile' has a build dependency on 'libjpeg', which does not exist
WARNING: Makefile 'package/feeds/packages/gst1-plugins-good/Makefile' has a dependency on 'libjpeg', which does not exist
WARNING: Makefile 'package/feeds/packages/imagemagick/Makefile' has a dependency on 'libjpeg', which does not exist
WARNING: Makefile 'package/feeds/packages/lcd4linux/Makefile' has a dependency on 'libjpeg', which does not exist
WARNING: Makefile 'package/feeds/packages/leptonica/Makefile' has a dependency on 'libjpeg', which does not exist
WARNING: Makefile 'package/feeds/packages/libcups/Makefile' has a dependency on 'libjpeg', which does not exist
WARNING: Makefile 'package/feeds/packages/libextractor/Makefile' has a dependency on 'libjpeg', which does not exist
WARNING: Makefile 'package/feeds/packages/libgd/Makefile' has a dependency on 'libjpeg', which does not exist
WARNING: Makefile 'package/feeds/packages/libgd/Makefile' has a dependency on 'libjpeg', which does not exist
WARNING: Makefile 'package/feeds/packages/libupm/Makefile' has a dependency on 'libjpeg', which does not exist
WARNING: Makefile 'package/feeds/packages/libupm/Makefile' has a dependency on 'libjpeg', which does not exist
WARNING: Makefile 'package/feeds/packages/libupm/Makefile' has a dependency on 'libjpeg', which does not exist
WARNING: Makefile 'package/feeds/packages/libupm/Makefile' has a dependency on 'libjpeg', which does not exist
WARNING: Makefile 'package/feeds/packages/minidlna/Makefile' has a dependency on 'libjpeg', which does not exist
WARNING: Makefile 'package/feeds/packages/mjpg-streamer/Makefile' has a dependency on 'libjpeg', which does not exist
WARNING: Makefile 'package/feeds/packages/motion/Makefile' has a dependency on 'libjpeg', which does not exist
WARNING: Makefile 'package/feeds/packages/opencv/Makefile' has a dependency on 'libjpeg', which does not exist
WARNING: Makefile 'package/feeds/packages/pillow/Makefile' has a dependency on 'libjpeg', which does not exist
WARNING: Makefile 'package/feeds/packages/qemu/Makefile' has a dependency on 'libjpeg', which does not exist
WARNING: Makefile 'package/feeds/packages/qemu/Makefile' has a dependency on 'libjpeg', which does not exist
WARNING: Makefile 'package/feeds/packages/sane-backends/Makefile' has a dependency on 'libjpeg', which does not exist
WARNING: Makefile 'package/feeds/packages/sane-backends/Makefile' has a dependency on 'libjpeg', which does not exist
WARNING: Makefile 'package/feeds/packages/sane-backends/Makefile' has a dependency on 'libjpeg', which does not exist
WARNING: Makefile 'package/feeds/packages/sane-backends/Makefile' has a dependency on 'libjpeg', which does not exist
WARNING: Makefile 'package/feeds/packages/sane-backends/Makefile' has a dependency on 'libjpeg', which does not exist
WARNING: Makefile 'package/feeds/packages/sane-backends/Makefile' has a dependency on 'libjpeg', which does not exist
WARNING: Makefile 'package/feeds/packages/sane-backends/Makefile' has a dependency on 'libjpeg', which does not exist
WARNING: Makefile 'package/feeds/packages/sane-backends/Makefile' has a dependency on 'libjpeg', which does not exist
WARNING: Makefile 'package/feeds/packages/sane-backends/Makefile' has a dependency on 'libjpeg', which does not exist
WARNING: Makefile 'package/feeds/packages/spice/Makefile' has a dependency on 'libjpeg', which does not exist
WARNING: Makefile 'package/feeds/packages/tiff/Makefile' has a dependency on 'libjpeg', which does not exist
WARNING: Makefile 'package/feeds/packages/vips/Makefile' has a dependency on 'libjpeg', which does not exist

Andy2244, 2020-08-01:
Its removed because of #11017 and #10090, basically the new "faster" libjpeg-turbo replaces libjpeg. The package also is a drop-in replacement since it has `PROVIDES:=libjpeg`.

So are your sure you have the latest feeds and installed the package?
`./scripts/feeds install -a`

verofin, 2020-08-01:
Sorry to report the problem still exist in a fresh clone I just did
WARNING: Makefile 'package/feeds/telephony/freeswitch/Makefile' has a build dependency on 'libjpeg', which does not exist
WARNING: Makefile 'package/feeds/packages/graphicsmagick/Makefile' has a build dependency on 'libjpeg', which does not exist

Andy2244, 2020-08-01:
When do you get those warnings, at the feeds update stage or later at the menuconfig or make?

verofin, 2020-08-02:
Git clone -> then feeds update -> error

verofin, 2020-08-02:
Hi Andy, I deleted the openWrt directory and cloned it afresh.
I ran the feed update -a command
Then I ran the feeds install -a command that gave me some warnings including the two mentioned.
At the end I run this same install command a second time and then only get the 2 warnings for libjpeg

neheb, 2020-08-02:
I fixed the second. The first will remain for a while.

verofin, 2020-08-02:
Great that seems to do the trick guys!! Both the warnings are resolved

Regards

micmac1, 2020-08-02:
> I fixed the second. The first will remain for a while.

telephony repo fast. Already pushed update :)

closed 2020-08-03 by neheb