openwrt/packages #12995: glib2.65 fails to build

SqTER-PL, 2020-08-01:
I have a problem with glib 2.65.0 compilation
Log with error below.

```
[429/452] Linking target glib/libglib-2.0.so.0.6500.0
FAILED: glib/libglib-2.0.so.0.6500.0 
arm-openwrt-linux-muslgnueabi-gcc  -o glib/libglib-2.0.so.0.6500.0 glib/libglib-2.0.so.0.6500.0.p/deprecated_gallocator.c.o glib/libglib-2.0.so.0.6500.0.p/deprecated_gcache.c.o glib/libglib-2.0.so.0.6500.0.p/deprecated_gcompletion.c.o glib/libglib-2.0.so.0.6500.0.p/deprecated_grel.c.o glib/libglib-2.0.so.0.6500.0.p/deprecated_gthread-deprecated.c.o glib/libglib-2.0.so.0.6500.0.p/garcbox.c.o glib/libglib-2.0.so.0.6500.0.p/garray.c.o glib/libglib-2.0.so.0.6500.0.p/gasyncqueue.c.o glib/libglib-2.0.so.0.6500.0.p/gatomic.c.o glib/libglib-2.0.so.0.6500.0.p/gbacktrace.c.o glib/libglib-2.0.so.0.6500.0.p/gbase64.c.o glib/libglib-2.0.so.0.6500.0.p/gbitlock.c.o glib/libglib-2.0.so.0.6500.0.p/gbookmarkfile.c.o glib/libglib-2.0.so.0.6500.0.p/gbytes.c.o glib/libglib-2.0.so.0.6500.0.p/gcharset.c.o glib/libglib-2.0.so.0.6500.0.p/gchecksum.c.o glib/libglib-2.0.so.0.6500.0.p/gconvert.c.o glib/libglib-2.0.so.0.6500.0.p/gdataset.c.o glib/libglib-2.0.so.0.6500.0.p/gdate.c.o glib/libglib-2.0.so.0.6500.0.p/gdatetime.c.o glib/libglib-2.0.so.0.6500.0.p/gdir.c.o glib/libglib-2.0.so.0.6500.0.p/genviron.c.o glib/libglib-2.0.so.0.6500.0.p/gerror.c.o glib/libglib-2.0.so.0.6500.0.p/gfileutils.c.o glib/libglib-2.0.so.0.6500.0.p/ggettext.c.o glib/libglib-2.0.so.0.6500.0.p/ghash.c.o glib/libglib-2.0.so.0.6500.0.p/ghmac.c.o glib/libglib-2.0.so.0.6500.0.p/ghook.c.o glib/libglib-2.0.so.0.6500.0.p/ghostutils.c.o glib/libglib-2.0.so.0.6500.0.p/giochannel.c.o glib/libglib-2.0.so.0.6500.0.p/gkeyfile.c.o glib/libglib-2.0.so.0.6500.0.p/glib-init.c.o glib/libglib-2.0.so.0.6500.0.p/glib-private.c.o glib/libglib-2.0.so.0.6500.0.p/glist.c.o glib/libglib-2.0.so.0.6500.0.p/gmain.c.o glib/libglib-2.0.so.0.6500.0.p/gmappedfile.c.o glib/libglib-2.0.so.0.6500.0.p/gmarkup.c.o glib/libglib-2.0.so.0.6500.0.p/gmem.c.o glib/libglib-2.0.so.0.6500.0.p/gmessages.c.o glib/libglib-2.0.so.0.6500.0.p/gnode.c.o glib/libglib-2.0.so.0.6500.0.p/goption.c.o glib/libglib-2.0.so.0.6500.0.p/gpattern.c.o glib/libglib-2.0.so.0.6500.0.p/gpoll.c.o glib/libglib-2.0.so.0.6500.0.p/gprimes.c.o glib/libglib-2.0.so.0.6500.0.p/gqsort.c.o glib/libglib-2.0.so.0.6500.0.p/gquark.c.o glib/libglib-2.0.so.0.6500.0.p/gqueue.c.o glib/libglib-2.0.so.0.6500.0.p/grand.c.o glib/libglib-2.0.so.0.6500.0.p/grcbox.c.o glib/libglib-2.0.so.0.6500.0.p/grefcount.c.o glib/libglib-2.0.so.0.6500.0.p/grefstring.c.o glib/libglib-2.0.so.0.6500.0.p/gregex.c.o glib/libglib-2.0.so.0.6500.0.p/gscanner.c.o glib/libglib-2.0.so.0.6500.0.p/gsequence.c.o glib/libglib-2.0.so.0.6500.0.p/gshell.c.o glib/libglib-2.0.so.0.6500.0.p/gslice.c.o glib/libglib-2.0.so.0.6500.0.p/gslist.c.o glib/libglib-2.0.so.0.6500.0.p/gstdio.c.o glib/libglib-2.0.so.0.6500.0.p/gstrfuncs.c.o glib/libglib-2.0.so.0.6500.0.p/gstring.c.o glib/libglib-2.0.so.0.6500.0.p/gstringchunk.c.o glib/libglib-2.0.so.0.6500.0.p/gtestutils.c.o glib/libglib-2.0.so.0.6500.0.p/gthread.c.o glib/libglib-2.0.so.0.6500.0.p/gthreadpool.c.o glib/libglib-2.0.so.0.6500.0.p/gtimer.c.o glib/libglib-2.0.so.0.6500.0.p/gtimezone.c.o glib/libglib-2.0.so.0.6500.0.p/gtranslit.c.o glib/libglib-2.0.so.0.6500.0.p/gtrashstack.c.o glib/libglib-2.0.so.0.6500.0.p/gtree.c.o glib/libglib-2.0.so.0.6500.0.p/guniprop.c.o glib/libglib-2.0.so.0.6500.0.p/gutf8.c.o glib/libglib-2.0.so.0.6500.0.p/gunibreak.c.o glib/libglib-2.0.so.0.6500.0.p/gunicollate.c.o glib/libglib-2.0.so.0.6500.0.p/gunidecomp.c.o glib/libglib-2.0.so.0.6500.0.p/gurifuncs.c.o glib/libglib-2.0.so.0.6500.0.p/gutils.c.o glib/libglib-2.0.so.0.6500.0.p/guuid.c.o glib/libglib-2.0.so.0.6500.0.p/gvariant.c.o glib/libglib-2.0.so.0.6500.0.p/gvariant-core.c.o glib/libglib-2.0.so.0.6500.0.p/gvariant-parser.c.o glib/libglib-2.0.so.0.6500.0.p/gvariant-serialiser.c.o glib/libglib-2.0.so.0.6500.0.p/gvarianttypeinfo.c.o glib/libglib-2.0.so.0.6500.0.p/gvarianttype.c.o glib/libglib-2.0.so.0.6500.0.p/gversion.c.o glib/libglib-2.0.so.0.6500.0.p/gwakeup.c.o glib/libglib-2.0.so.0.6500.0.p/gprintf.c.o glib/libglib-2.0.so.0.6500.0.p/glib-unix.c.o glib/libglib-2.0.so.0.6500.0.p/gspawn.c.o glib/libglib-2.0.so.0.6500.0.p/giounix.c.o glib/libglib-2.0.so.0.6500.0.p/gthread-posix.c.o glib/libcharset/libcharset.a.p/localcharset.c.o glib/gnulib/libgnulib.a.p/asnprintf.c.o glib/gnulib/libgnulib.a.p/printf.c.o glib/gnulib/libgnulib.a.p/printf-args.c.o glib/gnulib/libgnulib.a.p/printf-parse.c.o glib/gnulib/libgnulib.a.p/printf-frexp.c.o glib/gnulib/libgnulib.a.p/printf-frexpl.c.o glib/gnulib/libgnulib.a.p/vasnprintf.c.o glib/gnulib/libgnulib.a.p/xsize.c.o glib/pcre/libpcre.a.p/pcre_byte_order.c.o glib/pcre/libpcre.a.p/pcre_chartables.c.o glib/pcre/libpcre.a.p/pcre_compile.c.o glib/pcre/libpcre.a.p/pcre_config.c.o glib/pcre/libpcre.a.p/pcre_dfa_exec.c.o glib/pcre/libpcre.a.p/pcre_exec.c.o glib/pcre/libpcre.a.p/pcre_fullinfo.c.o glib/pcre/libpcre.a.p/pcre_get.c.o glib/pcre/libpcre.a.p/pcre_globals.c.o glib/pcre/libpcre.a.p/pcre_jit_compile.c.o glib/pcre/libpcre.a.p/pcre_newline.c.o glib/pcre/libpcre.a.p/pcre_ord2utf8.c.o glib/pcre/libpcre.a.p/pcre_string_utils.c.o glib/pcre/libpcre.a.p/pcre_study.c.o glib/pcre/libpcre.a.p/pcre_tables.c.o glib/pcre/libpcre.a.p/pcre_valid_utf8.c.o glib/pcre/libpcre.a.p/pcre_version.c.o glib/pcre/libpcre.a.p/pcre_xclass.c.o -L/Kompilacja/openwrt/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/usr/lib -L/Kompilacja/openwrt/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/lib -L/Kompilacja/openwrt/staging_dir/toolchain-arm_cortex-a15+neon-vfpv4_gcc-8.4.0_musl_eabi/usr/lib -L/Kompilacja/openwrt/staging_dir/toolchain-arm_cortex-a15+neon-vfpv4_gcc-8.4.0_musl_eabi/lib -L/Kompilacja/openwrt/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/usr/lib/libiconv-full/lib -L/Kompilacja/openwrt/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/usr/lib/libintl-full/lib -flto -Wl,--as-needed -Wl,--no-undefined -shared -fPIC -Wl,--start-group -Wl,-soname,libglib-2.0.so.0 -znow -zrelro -Wl,-rpath-link=/Kompilacja/openwrt/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/usr/lib/libiconv-full/lib -Wl,-rpath-link=/Kompilacja/openwrt/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/usr/lib/libintl-full/lib -Wl,--gc-sections -liconv -Wl,-z,nodelete -Wl,-Bsymbolic-functions -pthread -liconv -lm -Wl,--end-group
/Kompilacja/openwrt/staging_dir/toolchain-arm_cortex-a15+neon-vfpv4_gcc-8.4.0_musl_eabi/bin/../lib/gcc/arm-openwrt-linux-muslgnueabi/8.4.0/../../../../arm-openwrt-linux-muslgnueabi/bin/ld: /Kompilacja/openwrt/tmp/ccj4PgHM.ltrans9.ltrans.o: in function `_g_dgettext_should_translate':
<artificial>:(.text+0x2f0c): undefined reference to `libintl_textdomain'
/Kompilacja/openwrt/staging_dir/toolchain-arm_cortex-a15+neon-vfpv4_gcc-8.4.0_musl_eabi/bin/../lib/gcc/arm-openwrt-linux-muslgnueabi/8.4.0/../../../../arm-openwrt-linux-muslgnueabi/bin/ld: <artificial>:(.text+0x2f1c): undefined reference to `libintl_gettext'
/Kompilacja/openwrt/staging_dir/toolchain-arm_cortex-a15+neon-vfpv4_gcc-8.4.0_musl_eabi/bin/../lib/gcc/arm-openwrt-linux-muslgnueabi/8.4.0/../../../../arm-openwrt-linux-muslgnueabi/bin/ld: /Kompilacja/openwrt/tmp/ccj4PgHM.ltrans9.ltrans.o: in function `g_dngettext':
<artificial>:(.text+0x3060): undefined reference to `libintl_dngettext'
/Kompilacja/openwrt/staging_dir/toolchain-arm_cortex-a15+neon-vfpv4_gcc-8.4.0_musl_eabi/bin/../lib/gcc/arm-openwrt-linux-muslgnueabi/8.4.0/../../../../arm-openwrt-linux-muslgnueabi/bin/ld: /Kompilacja/openwrt/tmp/ccj4PgHM.ltrans9.ltrans.o: in function `g_dcgettext':
<artificial>:(.text+0x30bc): undefined reference to `libintl_dcgettext'
/Kompilacja/openwrt/staging_dir/toolchain-arm_cortex-a15+neon-vfpv4_gcc-8.4.0_musl_eabi/bin/../lib/gcc/arm-openwrt-linux-muslgnueabi/8.4.0/../../../../arm-openwrt-linux-muslgnueabi/bin/ld: /Kompilacja/openwrt/tmp/ccj4PgHM.ltrans9.ltrans.o: in function `g_dgettext':
<artificial>:(.text+0x3110): undefined reference to `libintl_dgettext'
/Kompilacja/openwrt/staging_dir/toolchain-arm_cortex-a15+neon-vfpv4_gcc-8.4.0_musl_eabi/bin/../lib/gcc/arm-openwrt-linux-muslgnueabi/8.4.0/../../../../arm-openwrt-linux-muslgnueabi/bin/ld: /Kompilacja/openwrt/tmp/ccj4PgHM.ltrans9.ltrans.o: in function `ensure_gettext_initialized':
<artificial>:(.text+0x33e4): undefined reference to `libintl_bindtextdomain'
[... cut ...]
time: package/feeds/packages/glib2/compile#81.72#10.79#40.83
package/Makefile:111: recipe for target 'package/feeds/packages/glib2/compile' failed
make[1]: *** [package/feeds/packages/glib2/compile] Error 2
make[1]: Leaving directory '/Kompilacja/openwrt'
/Kompilacja/openwrt/include/toplevel.mk:233: recipe for target 'package/glib2/compile' failed
make: *** [package/glib2/compile] Error 2
```
I ran a test and restoring the '005-intl-fix.patch' patch fixes the problem.
Can you take a look at it?

neheb, 2020-08-01:
That is just strange.

I'm not going to restore the patch. It caused massive build failure by adding a libintl build dependency that should be there only if full language support is enabled.

Speaking of which, do you have full language support enabled?

SqTER-PL, 2020-08-02:
Yes, I have full language support enabled. I need it.

neheb, 2020-08-02:
Add

$(if $(INTL_FULL),-lintl)

to TARGET_LDFLAGS.

SqTER-PL, 2020-08-02:
It's ok now. Compilation passed without a problem.

SqTER-PL, 2020-08-02:
Can you push the patch to git?
I close the issue. Everything is OK now. 
The compilation was successful.

Thanks for support.

SqTER-PL, 2020-08-02:
I'm still waiting for the patch to get approved and pushed to git.

neheb, 2020-08-02:
As soon as I test it locally.

neheb, 2020-08-03:
Done.

closed 2020-08-03 by neheb