openwrt/packages #13007: unbound: awk: /tmp/resolv.conf.auto: No such file or directory

cwbsw, 2020-08-03:
Maintainer: @EricLuehrsen 
Environment: OpenWrt snapshot

Description:

```
bundle_resolv_conf_servers() {
  local resolvers=$( awk '/nameserver/ { print $2 }' /tmp/resolv.conf.auto )
  UB_LIST_ZONE_SERVERS="$UB_LIST_ZONE_SERVERS $resolvers"
}
```
`unbound.sh` should be updated because `resolv.conf.auto` has been moved to `/tmp/resolv.conf.d/resolv.conf.auto`.

ja-pa, 2020-08-04:
Related commit https://github.com/openwrt/openwrt/commit/cd48d8d3420dd4df03daaa548227ceb487ba7104

EricLuehrsen, 2020-08-07:
You would think moving a key piece of infrastructure would receive a bit more spot light. As is classic OpenWrt culture, the documentation was not updated either https://openwrt.org/docs/guide-user/base-system/dhcp. :frowning_face:

closed 2020-08-07 by neheb

jow-, 2020-08-07:
> documentation was not updated either

Thanks for the heads-up. I modified the page accordingly.

PR #13045 (2020-08-07): unbound: follow resolv.conf.auto to new location
Maintainer: me
Tested: Linksys WRT3200ACM
Description: resolves #13007