openwrt/packages #13016: python-pynacl: compilation failed

feckert, 2020-08-04:
Maintainer: @jmarcet or @jefferyto 
Environment: x86_64, APU3, openwrt master latest commit

Description:
I get the following output if I want to compile docker-compose.
It does work a view weeks ago, but suddenly after updateing the package feed I got the follwoing compil error.
I am not a python god and do not know what the problem is now!
I haven't read into the whole python build system of openwrt.
I don't know where to start. Is this a problem with the host, target or the pip?

Can anyone from the specialists @jefferyto or @commodo help me out?

```
Collecting cffi>=1.4.1
  Using cached cffi-1.14.1.tar.gz (468 kB)
Requirement already satisfied: pycparser in /home/bbworker/bbworker/owrt_master_x86_64/System6/build/openwrt/staging_dir/hostpkg/lib/python3.8/site-packages (from cffi>=1.4.1) (2.20)
Skipping wheel build for cffi, due to binaries being disabled for it.
Installing collected packages: cffi
    Running setup.py install for cffi: started
    Running setup.py install for cffi: finished with status 'done'
Successfully installed cffi-1.14.1
WARNING: The wheel package is not available.
running install
running build
running build_py
creating build
creating build/lib.linux-x86_64-3.8
creating build/lib.linux-x86_64-3.8/nacl
copying src/nacl/exceptions.py -> build/lib.linux-x86_64-3.8/nacl
copying src/nacl/__init__.py -> build/lib.linux-x86_64-3.8/nacl
copying src/nacl/hash.py -> build/lib.linux-x86_64-3.8/nacl
copying src/nacl/utils.py -> build/lib.linux-x86_64-3.8/nacl
copying src/nacl/public.py -> build/lib.linux-x86_64-3.8/nacl
copying src/nacl/encoding.py -> build/lib.linux-x86_64-3.8/nacl
copying src/nacl/hashlib.py -> build/lib.linux-x86_64-3.8/nacl
copying src/nacl/secret.py -> build/lib.linux-x86_64-3.8/nacl
copying src/nacl/signing.py -> build/lib.linux-x86_64-3.8/nacl
creating build/lib.linux-x86_64-3.8/nacl/pwhash
copying src/nacl/pwhash/__init__.py -> build/lib.linux-x86_64-3.8/nacl/pwhash
copying src/nacl/pwhash/_argon2.py -> build/lib.linux-x86_64-3.8/nacl/pwhash
copying src/nacl/pwhash/argon2i.py -> build/lib.linux-x86_64-3.8/nacl/pwhash
copying src/nacl/pwhash/argon2id.py -> build/lib.linux-x86_64-3.8/nacl/pwhash
copying src/nacl/pwhash/scrypt.py -> build/lib.linux-x86_64-3.8/nacl/pwhash
creating build/lib.linux-x86_64-3.8/nacl/bindings
copying src/nacl/bindings/__init__.py -> build/lib.linux-x86_64-3.8/nacl/bindings
copying src/nacl/bindings/crypto_generichash.py -> build/lib.linux-x86_64-3.8/nacl/bindings
copying src/nacl/bindings/crypto_secretbox.py -> build/lib.linux-x86_64-3.8/nacl/bindings
copying src/nacl/bindings/crypto_shorthash.py -> build/lib.linux-x86_64-3.8/nacl/bindings
copying src/nacl/bindings/utils.py -> build/lib.linux-x86_64-3.8/nacl/bindings
copying src/nacl/bindings/randombytes.py -> build/lib.linux-x86_64-3.8/nacl/bindings
copying src/nacl/bindings/crypto_pwhash.py -> build/lib.linux-x86_64-3.8/nacl/bindings
copying src/nacl/bindings/crypto_hash.py -> build/lib.linux-x86_64-3.8/nacl/bindings
copying src/nacl/bindings/sodium_core.py -> build/lib.linux-x86_64-3.8/nacl/bindings
copying src/nacl/bindings/crypto_aead.py -> build/lib.linux-x86_64-3.8/nacl/bindings
copying src/nacl/bindings/crypto_scalarmult.py -> build/lib.linux-x86_64-3.8/nacl/bindings
copying src/nacl/bindings/crypto_sign.py -> build/lib.linux-x86_64-3.8/nacl/bindings
copying src/nacl/bindings/crypto_box.py -> build/lib.linux-x86_64-3.8/nacl/bindings
copying src/nacl/bindings/crypto_kx.py -> build/lib.linux-x86_64-3.8/nacl/bindings
copying src/nacl/bindings/crypto_secretstream.py -> build/lib.linux-x86_64-3.8/nacl/bindings
copying src/nacl/bindings/crypto_core.py -> build/lib.linux-x86_64-3.8/nacl/bindings
warning: build_py: byte-compiling is disabled, skipping.

running build_ext
generating cffi module 'build/temp.linux-x86_64-3.8/_sodium.c'
creating build/temp.linux-x86_64-3.8
building '_sodium' extension
creating build/temp.linux-x86_64-3.8/build
creating build/temp.linux-x86_64-3.8/build/temp.linux-x86_64-3.8
x86_64-openwrt-linux-musl-gcc -Wno-unused-result -Wsign-compare -DNDEBUG -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -ffile-prefix-map=/home/bbworker/bbworker/owrt_master_x86_64/System6/build/openwrt/build_dir/target-x86_64_musl/Python-3.8.5=Python-3.8.5 -Wformat -Werror=format-security -fpic -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -ffile-prefix-map=/home/bbworker/bbworker/owrt_master_x86_64/System6/build/openwrt/build_dir/target-x86_64_musl/Python-3.8.5=Python-3.8.5 -Wformat -Werror=format-security -fpic -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -DNDEBUG -fno-inline -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -ffile-prefix-map=/home/bbworker/bbworker/owrt_master_x86_64/System6/build/openwrt/build_dir/target-x86_64_musl/Python-3.8.5=Python-3.8.5 -Wformat -Werror=format-security -fpic -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -DNDEBUG -fno-inline -I/home/bbworker/bbworker/owrt_master_x86_64/System6/build/openwrt/staging_dir/target-x86_64_musl/usr/include -I/home/bbworker/bbworker/owrt_master_x86_64/System6/build/openwrt/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/usr/include -I/home/bbworker/bbworker/owrt_master_x86_64/System6/build/openwrt/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/include/fortify -I/home/bbworker/bbworker/owrt_master_x86_64/System6/build/openwrt/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/include -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -ffile-prefix-map=/home/bbworker/bbworker/owrt_master_x86_64/System6/build/openwrt/build_dir/target-x86_64_musl/pypi/PyNaCl-1.4.0=PyNaCl-1.4.0 -Wformat -Werror=format-security -fpic -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -I/home/bbworker/bbworker/owrt_master_x86_64/System6/build/openwrt/staging_dir/target-x86_64_musl/usr/include -I/home/bbworker/bbworker/owrt_master_x86_64/System6/build/openwrt/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/usr/include -I/home/bbworker/bbworker/owrt_master_x86_64/System6/build/openwrt/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/include/fortify -I/home/bbworker/bbworker/owrt_master_x86_64/System6/build/openwrt/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/include -I/home/bbworker/bbworker/owrt_master_x86_64/System6/build/openwrt/staging_dir/target-x86_64_musl/usr/include/python3.8 -fPIC -I/home/bbworker/bbworker/owrt_master_x86_64/System6/build/openwrt/staging_dir/target-x86_64_musl/usr/include/python3.8 -c build/temp.linux-x86_64-3.8/_sodium.c -o build/temp.linux-x86_64-3.8/build/temp.linux-x86_64-3.8/_sodium.o
cc1: note: someone does not honour COPTS correctly, passed 4 times
[... 27 lines cut ...]
In file included from /home/bbworker/bbworker/owrt_master_x86_64/System6/build/openwrt/staging_dir/target-x86_64_musl/usr/include/sodium.h:36,
                 from build/temp.linux-x86_64-3.8/_sodium.c:540:
/home/bbworker/bbworker/owrt_master_x86_64/System6/build/openwrt/staging_dir/target-x86_64_musl/usr/include/sodium/crypto_scalarmult_ed25519.h:40:5: note: previous declaration of 'crypto_scalarmult_ed25519_base' was here
 int crypto_scalarmult_ed25519_base(unsigned char *q, const unsigned char *n)
     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
build/temp.linux-x86_64-3.8/_sodium.c:687:7: error: 'crypto_scalarmult_ed25519_noclamp' redeclared as different kind of symbol
 int (*crypto_scalarmult_ed25519_noclamp)(unsigned char *, const unsigned char *,
       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /home/bbworker/bbworker/owrt_master_x86_64/System6/build/openwrt/staging_dir/target-x86_64_musl/usr/include/sodium.h:36,
                 from build/temp.linux-x86_64-3.8/_sodium.c:540:
/home/bbworker/bbworker/owrt_master_x86_64/System6/build/openwrt/staging_dir/target-x86_64_musl/usr/include/sodium/crypto_scalarmult_ed25519.h:35:5: note: previous declaration of 'crypto_scalarmult_ed25519_noclamp' was here
 int crypto_scalarmult_ed25519_noclamp(unsigned char *q, const unsigned char *n,
     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
build/temp.linux-x86_64-3.8/_sodium.c:689:7: error: 'crypto_scalarmult_ed25519_base_noclamp' redeclared as different kind of symbol
 int (*crypto_scalarmult_ed25519_base_noclamp)(unsigned char *, const unsigned char *) = NULL;
[... cut ...]
 int crypto_scalarmult_ed25519_base_noclamp(unsigned char *q, const unsigned char *n)
     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
error: command 'x86_64-openwrt-linux-musl-gcc' failed with exit status 1
Makefile:39: recipe for target '/home/bbworker/bbworker/owrt_master_x86_64/System6/build/openwrt/build_dir/target-x86_64_musl/pypi/PyNaCl-1.4.0/.built' failed
make[4]: *** [/home/bbworker/bbworker/owrt_master_x86_64/System6/build/openwrt/build_dir/target-x86_64_musl/pypi/PyNaCl-1.4.0/.built] Error 1
```

feckert, 2020-08-04:
The only thing I have changed to upstream is, that I have applied my [patch](https://github.com/openwrt/packages/pull/13012) to move the pip-cache to tmp.

commodo, 2020-08-04:
I updated `python3-cffi` recently to 1.4.1
Will check with that.

commodo, 2020-08-04:
> I updated `python3-cffi` recently to 1.4.1
> Will check with that.

wait; that shouldn't matter; i updated cffi to 1.14.1 recently; but that's on the target-side;
this looks host-side;

commodo, 2020-08-04:
so, this patch in libsodium seems to have broken things here:  https://github.com/openwrt/packages/commit/3ef28a4ab0cb60877c4ec68f5ba8c276021a1ab6
removing that patch helps;
@dangowrt   @jmarcet   @damianorenfer   ideas?

we can probably patch this into PyNaCl, but I wouldn't enforce a solution;
if possible i would let people discuss on this;

jefferyto, 2020-08-04:
I _think_ patching out all of the stub definitions in [src/bindings/minimal/crypto_scalarmult.h](https://github.com/pyca/pynacl/blob/efa75cb7057ed4c2f124ae77d3079e4ee9152080/src/bindings/minimal/crypto_scalarmult.h) and always setting `static const int PYNACL_HAS_CRYPTO_SCALARMULT_ED25519 = 1` should work, but I haven't tested this.

I would have preferred variants for libsodium (#12902) but I see @neheb was the one who [suggested](https://github.com/openwrt/packages/pull/12902#issuecomment-662778907) changing the minimal build API :frowning:

neheb, 2020-08-04:
Looks right.

jefferyto, 2020-08-04:
I know - I got into an argument about variants and InstallDev a while ago (which I am not going to link to here :joy:)

I think the best solution is some kind of lib-switching mechanism in the main repo, where dependent packages can specify which variant they want and the right InstallDev files are switched/linked/etc. just for them, but I really don't have time to look into something so obscure right now :joy:

commodo, 2020-08-05:
PR https://github.com/openwrt/packages/pull/13033

closed 2020-08-05 by neheb

feckert, 2020-08-06:
Thanks all for the quick fix :+1: