openwrt/packages #13019: gcc: compiler seems incompatible with linker provided by binutils

flyn-org, 2020-08-04:
Maintainer: @noblepepper 
Environment: x86_64 master

Description:
I use the gcc package to compile C code on my OpenWrt installation. Something has broken the toolchain. I receive errors such as these when I compile:
```
root@OpenWrt # gcc hello.c 
/usr/bin/ld: /usr/lib/gcc/x86_64-openwrt-linux-musl/7.4.0/libssp_nonshared.a(__stack_chk_fail_local.o): unable to initialize decompress status for section .debug_info
/usr/bin/ld: /usr/lib/gcc/x86_64-openwrt-linux-musl/7.4.0/libssp_nonshared.a(__stack_chk_fail_local.o): unable to initialize decompress status for section .debug_info
/usr/bin/ld: /usr/lib/gcc/x86_64-openwrt-linux-musl/7.4.0/libgcc.a(_muldi3.o): unable to initialize decompress status for section .debug_info
/usr/bin/ld: /usr/lib/gcc/x86_64-openwrt-linux-musl/7.4.0/libgcc.a(_muldi3.o): unable to initialize decompress status for section .debug_info
/usr/bin/ld: /usr/lib/gcc/x86_64-openwrt-linux-musl/7.4.0/libgcc.a(_muldi3.o): unable to initialize decompress status for section .debug_info
/usr/bin/ld: /usr/lib/gcc/x86_64-openwrt-linux-musl/7.4.0/libgcc.a(_muldi3.o): unable to initialize decompress status for section .debug_info
```
I was able to overcome this problem by updating binutils. I copied the version information and patches for binutils 2.32 from toolchain/binutils/ into package/devel/binutils/, and I built a newer version of the binutils package. Whereas the toolchain version of binutils provides the option to build a recent release, the package version is still 2.27.

noblepepper, 2020-08-16:
I can't duplicate your error, all works fine for me. Looking at the package repositories for x86_64 and mips_24kc (my environment) binutils version is 2.34.

flyn-org, 2020-08-16:
The binutils update happened recently, and it did indeed fix this issue. See commit 60f595da.

closed 2020-08-16 by flyn-org

seffs, 2021-01-11:
Hello, this issue happens to me in the same environment, just by running `gcc -v`

```
[root@NAS ~]# gcc -v
Using built-in specs.
COLLECT_GCC=gcc
COLLECT_LTO_WRAPPER=/opt/lib/gcc/x86_64-openwrt-linux-gnu/7.4.0/lto-wrapper
Target: x86_64-openwrt-linux-gnu
Configured with: /home/ryzhovau/Entware.x64/build_dir/target-x86_64_glibc-2.27/gcc-7.4.0/configure --target=x86_64-openwrt-linux --host=x86_64-openwrt-linux --build=x86_64-pc-linux-gnu --program-prefix= --program-suffix= --prefix=/opt --exec-prefix=/opt --bindir=/opt/bin --sbindir=/opt/sbin --libexecdir=/opt/lib --sysconfdir=/opt/etc --datadir=/opt/share --localstatedir=/opt/var --mandir=/opt/man --infodir=/opt/info --disable-nls 'CXXFLAGS_FOR_TARGET=-g -O2 -D_GLIBCXX_INCLUDE_NEXT_C_HEADERS' --build=x86_64-pc-linux-gnu --host=x86_64-openwrt-linux-gnu --target=x86_64-openwrt-linux-gnu --enable-languages=c,c++ --with-bugurl=https://dev.openwrt.org/ --with-pkgversion='OpenWrt GCC 7.4.0' --enable-shared --enable-__cxa_atexit --with-default-libstdcxx-abi=gcc4-compatible --enable-target-optspace --with-gnu-ld --with-specs='-Wl,-rpath=/opt/lib -Wl,--dynamic-linker=/opt/lib/ld-linux-x86-64.so.2' --with-boot-ldflags='static-libstdc++ -static-libgcc -Wl,-rpath=/opt/lib -Wl,--dynamic-linker=/opt/lib/ld-linux-x86-64.so.2' --disable-nls --disable-libsanitizer --disable-libvtv --disable-libcilkrts --disable-libmudflap --disable-libmpx --disable-multilib --disable-libgomp --disable-libquadmath --disable-libssp --disable-decimal-float --disable-libstdcxx-pch --with-host-libstdcxx=-lstdc++ --prefix=/opt --libexecdir=/opt/lib --with-local-prefix=/opt --with-stage1-ldflags='-lstdc++ -Wl,-rpath=/opt/lib -Wl,--dynamic-linker=/opt/lib/ld-linux-x86-64.so.2' --with-default-libstdcxx-abi=gcc4-compatible
Thread model: posix
gcc version 7.4.0 (OpenWrt GCC 7.4.0) 
COMPILER_PATH=/opt/lib/gcc/x86_64-openwrt-linux-gnu/7.4.0/:/opt/lib/gcc/x86_64-openwrt-linux-gnu/7.4.0/:/opt/lib/gcc/x86_64-openwrt-linux-gnu/:/opt/lib/gcc/x86_64-openwrt-linux-gnu/7.4.0/:/opt/lib/gcc/x86_64-openwrt-linux-gnu/
LIBRARY_PATH=/opt/lib/gcc/x86_64-openwrt-linux-gnu/7.4.0/:/lib/../lib64/:/usr/lib/../lib64/:/opt/lib/gcc/x86_64-openwrt-linux-gnu/7.4.0/../../../:/lib/:/usr/lib/
COLLECT_GCC_OPTIONS='-v' '-mtune=generic' '-march=x86-64'
 /opt/lib/gcc/x86_64-openwrt-linux-gnu/7.4.0/collect2 -plugin /opt/lib/gcc/x86_64-openwrt-linux-gnu/7.4.0/liblto_plugin.so -plugin-opt=/opt/lib/gcc/x86_64-openwrt-linux-gnu/7.4.0/lto-wrapper -plugin-opt=-fresolution=/tmp/ccwsxGlB.res -plugin-opt=-pass-through=-lgcc_s -plugin-opt=-pass-through=-lc -plugin-opt=-pass-through=-lgcc_s --eh-frame-hdr -m elf_x86_64 -dynamic-linker /lib64/ld-linux-x86-64.so.2 /opt/lib/gcc/x86_64-openwrt-linux-gnu/7.4.0/crt1.o /opt/lib/gcc/x86_64-openwrt-linux-gnu/7.4.0/crti.o /opt/lib/gcc/x86_64-openwrt-linux-gnu/7.4.0/crtbegin.o -L/opt/lib/gcc/x86_64-openwrt-linux-gnu/7.4.0 -L/lib/../lib64 -L/usr/lib/../lib64 -L/opt/lib/gcc/x86_64-openwrt-linux-gnu/7.4.0/../../.. -rpath=/opt/lib --dynamic-linker=/opt/lib/ld-linux-x86-64.so.2 -lgcc_s -lc -lgcc_s /opt/lib/gcc/x86_64-openwrt-linux-gnu/7.4.0/crtend.o /opt/lib/gcc/x86_64-openwrt-linux-gnu/7.4.0/crtn.o
/usr/bin/ld: /opt/lib/gcc/x86_64-openwrt-linux-gnu/7.4.0/libgcc.a(_muldi3.o): unable to initialize decompress status for section .debug_info
/usr/bin/ld: /opt/lib/gcc/x86_64-openwrt-linux-gnu/7.4.0/libgcc.a(_muldi3.o): unable to initialize decompress status for section .debug_info
/usr/bin/ld: /opt/lib/gcc/x86_64-openwrt-linux-gnu/7.4.0/libgcc.a(_muldi3.o): unable to initialize decompress status for section .debug_info
/usr/bin/ld: /opt/lib/gcc/x86_64-openwrt-linux-gnu/7.4.0/libgcc.a(_muldi3.o): unable to initialize decompress status for section .debug_info
/opt/lib/gcc/x86_64-openwrt-linux-gnu/7.4.0/crt1.o: In function `_start':
(.text+0x20): undefined reference to `main'
collect2: error: ld returned 1 exit status
```

How should I proceed?

noblepepper, 2021-01-14:
I can not duplicate this (I only have mips hardware for testing btw).

Are you using image and packages from https://downloads.openwrt.org/snapshots/targets/ ?

In your ouput COMPILER_PATH=/opt/lib/gcc/COMPILER_PATH=/opt/lib/gcc/x86_64-openwrt-linux-gnu// has some things that look very suspicious-

1. I don't get a COMPILER_PATH line in my output
2. Target x86_64-openwrt-linux-gnu/ probably should be x86_64-openwrt-linux-musl
3. Path references /opt, this directory is not usually on openwrt systems.

Cheers

dberardo-com, 2023-07-04:
hi everyone, i stumbled upon this thread while i was trying to compile (install) a python package on an openwrt device and i am getting the same error as highlighted in this issue.

basically i am trying to install the "regex" python package by downloading the source folder containing both .py and .pyc / .c files. by running "python setup.py install" i can see that the gcc compiler is used and that some .c file is compiled properly. then gcc is called with a -shared  flag and this error appears ... what am i doing wrong ? is what i am trying to achieve just impossible and will never work on openwrt or how should i go about it ? 

i am very confused and somewhat very frustrated with C mystical syntax and processes ... it is just so complicated :D