openwrt/packages #13034: open-vm-tools: compilation error

feckert, 2020-08-05:
Maintainer: @kuoruan @neheb 
Environment: x86_64, APU3, OpenWrt master

Description:
I have updated to the newest package master.
I get the following compilation error for open-vm-tools

```
OpenWrt-libtool: link: x86_64-openwrt-linux-musl-gcc -shared  -fPIC -DPIC  .libs/libvmtools_la-i18n.o .libs/libvmtools_la-monotonicTimer.o .libs/libvmtools_la-signalSource.o .libs/libvmtools_la-vmtools.o .libs/libvmtools_la-vmtoolsConfig.o .libs/libvmtools_la-vmtoolsLog.o .libs/libvmtools_la-vmxLogger.o .libs/libvmtools_la-guestSDKLog.o .libs/libvmtools_la-stub-log.o  -Wl,--whole-archive ../lib/lock/.libs/libLock.a ../lib/backdoor/.libs/libBackdoor.a ../lib/asyncsocket/.libs/libAsyncSocket.a ../lib/sslDirect/.libs/libSslDirect.a ../lib/pollGtk/.libs/libPollGtk.a ../lib/poll/.libs/libPoll.a ../lib/dataMap/.libs/libDataMap.a ../lib/hashMap/.libs/libHashMap.a ../lib/dict/.libs/libDict.a ../lib/dynxdr/.libs/libDynxdr.a ../lib/err/.libs/libErr.a ../lib/file/.libs/libFile.a ../lib/glibUtils/.libs/libGlibUtils.a ../lib/guestApp/.libs/libGuestApp.a ../lib/guestRpc/.libs/libGuestRpc.a ../lib/message/.libs/libMessage.a ../lib/netUtil/.libs/libNetUtil.a ../lib/nicInfo/.libs/libNicInfo.a ../lib/panic/.libs/libPanic.a ../lib/procMgr/.libs/libProcMgr.a ../lib/rpcChannel/.libs/libRpcChannel.a ../lib/rpcIn/.libs/libRpcIn.a ../lib/rpcOut/.libs/libRpcOut.a ../lib/rpcVmx/.libs/libRpcVmx.a ../lib/string/.libs/libString.a ../lib/syncDriver/.libs/libSyncDriver.a ../lib/system/.libs/libSystem.a ../lib/stubs/.libs/libStubsCS.a ../lib/unicode/.libs/libUnicode.a ../lib/user/.libs/libUser.a ../lib/vmCheck/.libs/libVmCheck.a ../lib/vmSignal/.libs/libVmSignal.a ../lib/wiper/.libs/libWiper.a ../lib/misc/.libs/libMisc.a ../lib/slashProc/.libs/libSlashProc.a -Wl,--no-whole-archive  -L/home/feckert/workspace/openwrt/LDM-master-64/build/openwrt/staging_dir/target-x86_64_musl/usr/lib -L/home/feckert/workspace/openwrt/LDM-master-64/build/openwrt/staging_dir/target-x86_64_musl/lib -L/home/feckert/workspace/openwrt/LDM-master-64/build/openwrt/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/usr/lib -L/home/feckert/workspace/openwrt/LDM-master-64/build/openwrt/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/lib -ltirpc -lrt -lcrypt -lpthread -lglib-2.0 -liconv -lm  -Os -fstack-protector -Wl,-z -Wl,now -Wl,-z -Wl,relro -Wl,-z -Wl,defs -Wl,-lc -pthread   -pthread -Wl,-soname -Wl,libvmtools.so.0 -o .libs/libvmtools.so.0.0.0
/home/feckert/workspace/openwrt/LDM-master-64/build/openwrt/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/lib/gcc/x86_64-openwrt-linux-musl/8.4.0/../../../../x86_64-openwrt-linux-musl/bin/ld: cannot find -liconv
collect2: error: ld returned 1 exit status
```
So I thought add `include $(INCLUDE_DIR)/nls.mk` should solve this.
But then I get another error

```
make[6]: Entering directory '/home/feckert/workspace/openwrt/LDM-master-64/build/openwrt/build_dir/target-x86_64_musl/open-vm-tools-11.1.0-16036546/services/vmtoolsd'
x86_64-openwrt-linux-musl-gcc -DPACKAGE_NAME=\"open-vm-tools\" -DPACKAGE_TARNAME=\"open-vm-tools\" -DPACKAGE_VERSION=\"11.1.0\" -DPACKAGE_STRING=\"open-vm-tools\ 11.1.0\" -DPACKAGE_BUGREPORT=\"open-vm-tools-devel@lists.sourceforge.net\" -DPACKAGE_URL=\"\" -DX_DISPLAY_MISSING=1 -DSTDC_HEADERS=1 -DHAVE_SYS_TYPES_H=1 -DHAVE_SYS_STAT_H=1 -DHAVE_STDLIB_H=1 -DHAVE_STRING_H=1 -DHAVE_MEMORY_H=1 -DHAVE_STRINGS_H=1 -DHAVE_INTTYPES_H=1 -DHAVE_STDINT_H=1 -DHAVE_UNISTD_H=1 -DPACKAGE=\"open-vm-tools\" -DVERSION=\"11.1.0\" -DHAVE_DLFCN_H=1 -DLT_OBJDIR=\".libs/\" -DHAVE_DLOPEN=1 -DHAVE_ECVT=1 -DHAVE_FCVT=1 -DHAVE_GETIFADDRS=1 -DHAVE_GETAUXVAL=1 -DHAVE_ISSETUGID=1 -DNO_DNET=1 -DHAVE_CRYPT_H=1 -DHAVE_IFADDRS_H=1 -DHAVE_INTTYPES_H=1 -DHAVE_STDINT_H=1 -DHAVE_STDLIB_H=1 -DHAVE_WCHAR_H=1 -DHAVE_NET_IF_H=1 -DHAVE_SYS_AUXV_H=1 -DHAVE_SYS_IO_H=1 -DHAVE_SYS_PARAM_H=1 -DHAVE_SYS_SYSINFO_H=1 -DHAVE_SYS_TYPES_H=1 -DHAVE_SYS_USER_H=1 -DHAVE_SYS_VFS_H=1 -DHAVE_UNWIND_H=1 -DHAVE__BOOL=1 -DHAVE_STDBOOL_H=1 -DHAVE_STRUCT_STAT_ST_RDEV=1 -DHAVE_STRUCT_TIMESPEC_TV_SEC=1 -DTIME_WITH_SYS_TIME=1 -DNO_XSM=1 -DNO_XCOMPOSITE=1 -DNO_MULTIMON=1 -I.  -DVMTOOLS_USE_GLIB -I/home/feckert/workspace/openwrt/LDM-master-64/build/openwrt/staging_dir/target-x86_64_musl/usr/include/glib-2.0 -pthread -I/home/feckert/workspace/openwrt/LDM-master-64/build/openwrt/staging_dir/target-x86_64_musl/usr/lib/glib-2.0/include  -I/home/feckert/workspace/openwrt/LDM-master-64/build/openwrt/staging_dir/target-x86_64_musl/usr/include -I/home/feckert/workspace/openwrt/LDM-master-64/build/openwrt/staging_dir/target-x86_64_musl/usr/include/glib-2.0 -I/home/feckert/workspace/openwrt/LDM-master-64/build/openwrt/staging_dir/target-x86_64_musl/usr/lib/glib-2.0/include -pthread -I/home/feckert/workspace/openwrt/LDM-master-64/build/openwrt/staging_dir/target-x86_64_musl/usr/include -I/home/feckert/workspace/openwrt/LDM-master-64/build/openwrt/staging_dir/target-x86_64_musl/usr/include/glib-2.0 -pthread -I/home/feckert/workspace/openwrt/LDM-master-64/build/openwrt/staging_dir/target-x86_64_musl/usr/lib/glib-2.0/include  -I/home/feckert/workspace/openwrt/LDM-master-64/build/openwrt/staging_dir/target-x86_64_musl/usr/include -I/home/feckert/workspace/openwrt/LDM-master-64/build/openwrt/staging_dir/target-x86_64_musl/usr/include/glib-2.0 -pthread -I/home/feckert/workspace/openwrt/LDM-master-64/build/openwrt/staging_dir/target-x86_64_musl/usr/lib/glib-2.0/include  -I. -DVMTOOLSD_PLUGIN_ROOT=\"/usr/lib/open-vm-tools/plugins\" -I/home/feckert/workspace/openwrt/LDM-master-64/build/openwrt/build_dir/target-x86_64_musl/open-vm-tools-11.1.0-16036546/lib/include -I/home/feckert/workspace/openwrt/LDM-master-64/build/openwrt/build_dir/target-x86_64_musl/open-vm-tools-11.1.0-16036546/lib/include -I/home/feckert/workspace/openwrt/LDM-master-64/build/openwrt/staging_dir/target-x86_64_musl/usr/include -I/home/feckert/workspace/openwrt/LDM-master-64/build/openwrt/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/usr/include -I/home/feckert/workspace/openwrt/LDM-master-64/build/openwrt/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/include/fortify -I/home/feckert/workspace/openwrt/LDM-master-64/build/openwrt/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/include -I/home/feckert/workspace/openwrt/LDM-master-64/build/openwrt/staging_dir/target-x86_64_musl/usr/lib/libiconv-stub/include -I/home/feckert/workspace/openwrt/LDM-master-64/build/openwrt/staging_dir/target-x86_64_musl/usr/lib/libintl-stub/include  -DUSING_AUTOCONF=1 -DOPEN_VM_TOOLS -DNO_ICU -DVMX86_TOOLS -D_FILE_OFFSET_BITS=64 -D_LARGEFILE64_SOURCE -D_XOPEN_SOURCE=500 -D_BSD_SOURCE -D_SVID_SOURCE -D_DEFAULT_SOURCE  -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -ffile-prefix-map=/home/feckert/workspace/openwrt/LDM-master-64/build/openwrt/build_dir/target-x86_64_musl/open-vm-tools-11.1.0-16036546=open-vm-tools-11.1.0-16036546 -Wformat -Werror=format-security -fpic -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -I/home/feckert/workspace/openwrt/LDM-master-64/build/openwrt/staging_dir/target-x86_64_musl/usr/lib/libiconv-stub/include -I/home/feckert/workspace/openwrt/LDM-master-64/build/openwrt/staging_dir/target-x86_64_musl/usr/lib/libintl-stub/include  -Wall -Werror -Wno-pointer-sign -Wno-unused-value -fno-strict-aliasing -Wno-unknown-pragmas -Wno-uninitialized -Wno-deprecated-declarations -Wno-unused-const-variable -Wno-unused-but-set-variable -MT vmtoolsd-cmdLine.o -MD -MP -MF .deps/vmtoolsd-cmdLine.Tpo -c -o vmtoolsd-cmdLine.o `test -f 'cmdLine.c' || echo './'`cmdLine.c
In file included from cmdLine.c:34:
/home/feckert/workspace/openwrt/LDM-master-64/build/openwrt/staging_dir/target-x86_64_musl/usr/include/glib-2.0/glib/gi18n.h:26: error: "_" redefined [-Werror]
 #define  _(String) gettext (String)

In file included from /home/feckert/workspace/openwrt/LDM-master-64/build/openwrt/staging_dir/target-x86_64_musl/usr/include/glib-2.0/glib/gi18n.h:23,
                 from cmdLine.c:34:
/home/feckert/workspace/openwrt/LDM-master-64/build/openwrt/staging_dir/target-x86_64_musl/usr/lib/libintl-stub/include/libintl.h:41: note: this is the location of the previous definition
 #define _(String) (String)

In file included from cmdLine.c:34:
/home/feckert/workspace/openwrt/LDM-master-64/build/openwrt/staging_dir/target-x86_64_musl/usr/include/glib-2.0/glib/gi18n.h:28: error: "N_" redefined [-Werror]
 #define N_(String) (String)

[... cut ...]
https://github.com/openwrt/packages/commit/48870aac0c799de93e40d568c756b1bccef1689d#diff-d13e9dd83dd618f679be82c105439681
https://github.com/openwrt/packages/commit/f23da6d1b8ef1b02a0398d08a84d812553fc64e7#diff-d13e9dd83dd618f679be82c105439681
https://github.com/openwrt/packages/commit/1f3a60acd6ed3e383dcdb896dbb700c3aad0ad3e#diff-d13e9dd83dd618f679be82c105439681
https://github.com/openwrt/packages/commit/7c9a0919a7fc335b8cc09a221941c131b8f84baf#diff-d13e9dd83dd618f679be82c105439681

But that's just a guess

neheb, 2020-08-05:
Looks like Werror needs to be disabled.

phoenixxie7, 2020-08-21:
I have the same problem. I think that the toolchain should add libiconv

hue715, 2020-09-12:
Environment: x86_64,  OpenWrt master

I think that there is something wrong with glib2，so I used two versions (2.58.3 and 2.65.2) to compile and compared the difference between the results.

when I use glib2(2.58.3) , open-vm-tools makefile (compilation OK)：
[image]

when I use glib2（2.65.2），open-vm-tools makefile (compilation error):
[image]

I guess vmtools does not need liconv.

closed 2020-10-09 by feckert