openwrt/packages #13043: ZeroTier outdated dependency

verofin, 2020-08-06:
Description:

Error when installing ZeroTier from luci interface

Version: 1.4.6-5
Size: ~389.1 KB installed
Dependencies:
libpthread INSTALLED
libgcc1 INSTALLED
libstdcpp6 (387.2 KB) NOT INSTALLED
kmod-tun (27.4 KB) NOT INSTALLED
**kernel VERSION INCOMPATIBLE**
ip-full (208.0 KB) NOT INSTALLED | ip-tiny (147.8 KB) NOT INSTALLED
libnl-tiny INSTALLED
libelf1 (41.0 KB) NOT INSTALLED
zlib INSTALLED
libmnl0 (7.3 KB) NOT INSTALLED
libminiupnpc (23.4 KB) NOT INSTALLED
libnatpmp1 (3.6 KB) NOT INSTALLED

The installed version of package kernel is not compatible, require 5.4.52-1-3ac7ead9… while 5.4.52-1-de9d5d73… is installed.

neheb, 2020-08-06:
Typical problem with snapshot builds.

verofin, 2020-08-07:
Thanks for the feedback. I presumed I needed one of the releases instead of the snapshot. I cloned the 19.07.03 branch, but it does not support Raspberry Pi4 yet.  What would be your suggested resolution to this situation?

neheb, 2020-08-07:
Erm, if you compile your own builds, this will not be an issue. Meaning master will work.

neheb, 2020-08-10:
Can this be closed?

closed 2020-08-12 by BKPepe