openwrt/packages #13052: docker-ce: broken, cgroup subsystem 'device' is missing

domomg, 2020-08-08:
Mantainer: @G-M0N3Y-2503 
Environment: x86_64 on baremetal and VMWare, openwrt master

Description:
I've been having some troubles with the latest OpenWRT snapshots, regarding docker-ce.

My last two attempts have been with r-14035 three days ago, and r-14085 yesterday.

They were both compiled with default options, I only selected docker-related stuffs under Utilities.

In both cases I successfully got a working image from the build system, except when I try to run docker-ce on them, it fails because the 'device' cgroup subsystem isn't mounted, and it's also missing from /sys/fs/cgroup.

In both cases, the cgroup_root and most of the subsystems are correctly mounted.

For comparison, I fired up an old VM with r-12719 built back in March, where everything works without a hitch.

r-14085:
[image]

r-12719:
[image]

Any attempt to use the docker CLI utility returns a "Cannot connect to the Docker daemon at unix:///var/run/docker.sock. Is the docker daemon running?"

Trying to start the dockerd daemon manually ends up with failure due to the 'devices' subsystem being unmounted:

[image]

As far as I know, the last snapshot working for me is r13834, built about 3 weeks ago for a Raspberry Pi 3:

[image]

Fixes attempted so far:

- tried with the kernel boot option "cgroup_enable=devices" and, after a reboot,  made the 'devices' directory inside /sys/fs/cgroup and attempted to mount it with no success, I get the following error:
"cgroup1: Unknown subsys name 'devices' "

`mkdir /sys/fs/cgroup/devices
mount -t cgroup devices -o devices /sys/fs/cgroup/devices`

[image]

G-M0N3Y-2503, 2020-08-08:
I could reproduce this even with all the docker kernel options enabled.
It appears that they aren't even in `/proc/cgroups` so I suspect there has been a change in the kernel config.

peterwillcn, 2020-08-08:
https://github.com/docker/cli/issues/2104

G-M0N3Y-2503, 2020-08-08:
Just tested and it's definitely the change to cgroups v2 done in https://github.com/openwrt/openwrt/commit/d1a8217d87bffa33fd7d4562b3ed2f797c14beaf.
I'll look into adding the legacy options for a custom build but it looks like the snapshots will be broken until https://github.com/moby/moby/issues/40360

domomg, 2020-08-08:
> Just tested and it's definitely the change to cgroups v2 done in [openwrt/openwrt@d1a8217](https://github.com/openwrt/openwrt/commit/d1a8217d87bffa33fd7d4562b3ed2f797c14beaf).
> I'll look into adding the legacy options for a custom build but it looks like the snapshots will be broken until [moby/moby#40360](https://github.com/moby/moby/issues/40360)

First of all, thank you for pointing me to the right direction, it didn't occur to me at all to look at the kernel parameters in the config file, you saved me quite a lot of time.

I did a test rolling back to the commit immediately behind the one giving us trouble, https://github.com/openwrt/openwrt/commit/917980fd8a6589d6911797211f1871016f527f8c , compiled from there, and lo and behold, it works.

[image]

I'll stick to this version until there's a fix for this issue, I'm aware it's not a long term solution given cgroup v1 is legacy, but at least for now it works.

Thank you again for your help.

peterwillcn, 2020-08-09:
@G-M0N3Y-2503  https://github.com/openwrt/openwrt/commit/d1a8217d87bffa33fd7d4562b3ed2f797c14beaf
> You need to enable build-time options for legacy-cgroup1-only controllers (such as devices, net_prio, net_cls) in menuconfig, they are no longer included by default.
It'd be sad to encourage the use of legacy functions which prevent using the modern interfaces (which is true for cgroup1 controllers). Hence, as 19.07 came without cgroup1 per default, we will keep it like that for 20.x (and supposedly everything after). cgroup2 is enabled and supports device-firewalling using eBPF, it has been decided that the previous 'devices' cgroup will not be implemented in cgroup2 (hence Docker needs to change and make use of cgroup2 in order to work on mordern systems).

G-M0N3Y-2503, 2020-08-09:
I'll look into making the current version of docker dependent on these legacy kernel options for now, which will mean docker won't be available in snapshots until the version that supports cgroups2 is released.

peterwillcn, 2020-08-09:
I fixed this problem by enabling the following kernel configuration
```
+CONFIG_KERNEL_CGROUP_DEVICE=y
+CONFIG_KERNEL_CGROUP_FREEZER=y
+CONFIG_KERNEL_CGROUP_NET_CLASSID=y
+CONFIG_KERNEL_CGROUP_NET_PRIO=y
+CONFIG_KERNEL_NET_CLS_CGROUP=y
```

najdanovicivan, 2020-10-10:
I've submitted this PR https://github.com/openwrt/openwrt/pull/3504 it should make it possible to use docker-ce on shapshot build for !SMALL_FLASH devices

G-M0N3Y-2503, 2020-10-10:
For awareness, CGroups V2 should be supported in [[Docker Engine] 2020 release](https://github.com/docker/roadmap/issues/35)

G-M0N3Y-2503, 2021-02-03:
It looks like this issue should be closed now that https://github.com/openwrt/packages/pull/13874 has merged

closed 2021-02-03 by feckert

discordianfish, 2023-09-04:
@feckert I might miss something but it looks like 23.05.0-rc3 is missing the devices cgroup controller as well. It's neither exposed via cgroup v1 nor v2.

hubrichc, 2023-11-23:
Firmware Version：OpenWrt 21.02.7 r16847-f8282da11e

Cgroup v2 mount points: 
Cgroup v1 freezer controller: missing
Cgroup v1 clone_children flag: enabled
Cgroup device: missing
Cgroup sched: enabled
Cgroup cpu account: enabled
Cgroup memory controller: enabled
Cgroup cpuset: enabled

kevin-king, 2024-01-18:
+1 Experiencing this issue as well: `Cgroup device: missing` on OpenWrt 23.05.2. Been troubleshooting for a while.  Can't bring up docker socket on openwrt or on lxc container guests.

This gentleman Wabuo:

https://forum.openwrt.org/t/podman-cgroup-struggles-sys-fs-cgroup-devices-missing-need-help/182119/4

had luck enabling all CGROUP flags and compiling the kernel from scratch just 21 days ago...this is beyond my skill level at the moment, but might be my next attempt.

winhex, 2024-01-22:
I have exactly same problem after trying to install lxc. After this dockerd won't start and stuck with error:
`failed to start daemon: Devices cgroup isn't mounted`
I tried to remove lxc and reinstall docker, but it didn't help. But immediately after installing OpenWrt, docker worked well

OpenWrt 23.05.0 with stock kernel 5.15.134
x86/64

Any ideas?

G-M0N3Y-2503, 2024-02-20:
I'm updating docker on 23.05 and it seemed to run a basic image on my x86 VM perhaps that will help.

https://github.com/openwrt/packages/pull/23483

vincejv, 2024-02-20:
Hello @G-M0N3Y-2503 I've been using Docker **very heavily** on my OpenWRT x86 bare metal machine, I never had the issue regarding cgroup devices. I reckon these guys are installing Docker + LXC on the same OpenWRT host, which I also do.

I initially encountered an issue with the docker daemon not starting when LXC was installed, I was told they were not supposed to run on the same system, but I persevered and investigated the issue, installing/uninstalling every single dependency from LXC, and it turns out it was an issue with the outdated OpenWRT Wiki LXC Tutorial (it still talks about lxc cgroup v1) https://openwrt.org/lxc_openwrt_host

It stated to install the ff packages: `cgroupfs-mount cgroup-tools` which LXC did not require these two packages on OpenWRT 23.x, I'm using LXC Ubuntu Jammy with no issues at all and installing them would throw the docker error that users report during system startup.

Also, I'm running Jellyfin + GPU Hardware Acc, which requires mounting docker devices, and it perfectly works without any hitches which makes me think, it's not even an issue with Docker CGroup Devices at all since Intel GPU Hardware accel ran on Jellyfin docker. 
[image]

cc: @discordianfish @kevin-king @winhex 

Also to add, I'm running #23483 on kernel 6.1.78 with the PR docker version, I was about to push a PR but you beat me to it by minutes. I had a minor issue, during the upgrade, some containers not auto-starting, which was also present in earlier docker versions during dockerd upgrades, but it was very trivial, restarting/redeploying containers fixed it, probably an upstream issue.

Tl;dr: Not a docker issue, but an outdated documentation, didn't have the issue even with Docker 24.x.