openwrt/packages #13053: ffmpeg-full not handled properly for GPL option

02015678, 2020-08-08:
Maintainer: @antonlacon and @thess
Environment: armv8, Cortex-A53, Openwrt-19.07

Description:
ffmpeg-full prefer to use libx264 (gpl) than fdk-aac (non-free), which is good.
As for gpl option it depends on package named **libx264**, but in opwnwrt/packages/libs/libx264/Makefile,
it named itself as **x264**, not **libx264**.

As a result, dependency cannot be satisfied, if user wants to use gpl libx264 for compilation.

neheb, 2020-08-10:
Not true:

https://github.com/openwrt/packages/blob/openwrt-19.07/libs/libx264/Makefile#L85

The PKG_NAME in the Makefile is for downloading only.

02015678, 2020-08-16:
Thanks. It turns out that it doesn't work if manual edit config file and add  `CONFIG_PACKAGE_libffmpeg-custom=m` and  `CONFIG_PACKAGE_fdk-aac=y`. When enable ffmpeg-custom in makeconfig interface ALL the required options would be added to config file.

```
 CONFIG_PACKAGE_libffmpeg-custom=m
 #
 # Build Licensing
 #
 CONFIG_FFMPEG_CUSTOM_GPL=y
 CONFIG_FFMPEG_CUSTOM_GPLV3=y
 CONFIG_FFMPEG_CUSTOM_NONFREE=y
 CONFIG_FFMPEG_CUSTOM_PATENTED=y

 #
 # External Libraries
 #
 CONFIG_FFMPEG_CUSTOM_SELECT_libfdk-aac=y
 CONFIG_FFMPEG_CUSTOM_SELECT_libmp3lame=y
```

Anyway ffmpeg-full should be fine, switch to GPL ffmpeg-full instead.

closed 2020-08-16 by 02015678