openwrt/packages #13081: qemu: build server fails to build

flyn-org, 2020-08-11:
Maintainer: @yousong 
Environment: x86_64 master

Description:
The OpenWrt build server cannot presently build the QEMU packages. See https://downloads.openwrt.org/snapshots/faillogs/x86_64/packages/qemu/compile.txt.

yousong, 2020-08-11:
The issue should be fixed now.  The part I am not sure is how the build passed on May, 2020 when I built and run-tested it...

yousong, 2020-08-12:
Copy buildbot error message here for archive purposes

```
Applying ./patches/0006-util-mmap-alloc-fix-missing-MAP_SYNC.patch using plaintext: 
patching file util/mmap-alloc.c

ERROR: sizeof(size_t) doesn't match GLIB_SIZEOF_SIZE_T.
       You probably need to set PKG_CONFIG_LIBDIR
       to point to the right pkg-config files for your
       build target

Makefile:449: recipe for target '/builder/shared-workdir/build/sdk/build_dir/target-x86_64_musl/qemu-5.0.0/.configured_85380a2ed95d41cd179ef4118b5a6fd4' failed
make[3]: *** [/builder/shared-workdir/build/sdk/build_dir/target-x86_64_musl/qemu-5.0.0/.configured_85380a2ed95d41cd179ef4118b5a6fd4] Error 1
time: package/feeds/packages/qemu/compile#7.11#6.89#15.13
```

And the config.log error as I remembered it.  `ld cannot find -liconv`.  No -L/path/to/libiconv-stub in the cc command.

flyn-org, 2020-08-14:
Looks like the build server built the new version. Thanks!

closed 2020-08-14 by flyn-org