openwrt/packages #13085: ttyd: little endian vs. big endian issue

feckert, 2020-08-11:
Maintainer: @tsl0922 
Environment: x86_64 APU3 and lantiq_xrx200, OpenWrt master

Description:

I have a problem with the little endian vs. big endian with ttyd. On my systems.

On x86_64 every works as expected, but on a MIPS platform ttyd does not work. 
Since I am not familiar with the ttyd the question is whether it is the ttyd or the libwebsocket.

The [assumption](https://github.com/openwrt/packages/issues/11632#issuecomment-670078852) is that the problem is the libwebsocket library.

If it is in libwesocket we should update to the newest upstream version [4.0](https://libwebsockets.org/git/libwebsockets/tree/?h=v4.0-stable) of libwebsocket. But on OpenWrt we still have the version [3.2.1](https://github.com/openwrt/packages/blob/master/libs/libwebsockets/Makefile#L11). As metioned by @karlp we have to make at first update the depending packages to work with https://github.com/openwrt/packages/issues/11632#issuecomment-671882438.

Maybe the problem will solve itself. Or we find the possible bug in libwebsocket and fixes this for version 3.2.1. ?

tsl0922, 2020-08-11:
I'm running ttyd on MIPS based router (MT7621), it works fine.

If you want to test ttyd mips with libwebsockets 4.0, you can download a prebuilt static version from the ci artifacts:

https://github.com/tsl0922/ttyd/actions/runs/187335123

feckert, 2020-08-11:
> I'm running ttyd on MIPS based router (MT7621), it works fine.

Did you install this from the Openwrt repository?

> 
> If you want to test ttyd mips with libwebsockets 4.0, you can download a prebuilt static version from the ci artifacts:
> 
> https://github.com/tsl0922/ttyd/actions/runs/187335123

Tahnks but that is unfortunately not an option, because I have to compile it myself. I am build a image with the image builder.

tsl0922, 2020-08-11:
> Did you install this from the Openwrt repository?
No. I compiled it myself, with openwrt-14.07 sdk and libwebsockets-3.2.2, for an openwrt-based third-party firmware.

tsl0922, 2020-08-11:
Just confirmed: I'm running ttyd mipsel version (little endian) on my router.

micmac1, 2020-08-11:
Hi there,

I tried it out on mips big endian. Works fine there, too. This is 19.07.03.

It was the first time I rand ttyd. It started automatically and I just had to open the URL in my browser. Et voila: looks like a root shell?! :) This can't seriously be the default, right? No authentication whatsoever and you get root?! Whoa :)

Kind regards,
Seb

feckert, 2020-08-19:
I verified it too, and it works in openwrt-19.07 for x86_64 and lantiq_xrx200.
But in the master branch id does not work for lantiq_xrx200 with the current installed libwebsocket version 3.2.1
@karlp If I removed the last version bump [commit](https://github.com/openwrt/packages/commit/7066486fbbfa1bb73fee74911a8306e3a534a6fe#diff-ba78d49c46a24be055e798c6a2435eec) and get back to version 3.1.0 ttyd works again on lantiq_xrx200. The version 3.1.0 is also installed on the latest stable branch openwrt-19.07

karlp, 2020-08-19:
my famliarity with lws extends to, "it's fragile as hell"  I'm being pushed to update it, so no, I'm not really considering going back to some other version.  you can try 4.0.20 yourself, see if it at least fixes ttyd for you, (it's still broken on mosquitto for me unfortunately)  AIUI, it's only broken on your particular BE device right?

feckert, 2020-08-19:
> AIUI, it's only broken on your particular BE device right?

What do you mean with `AIUI` and `BE`?

I can only say that if I try it with a MIPS architecture (lantiq_xrx200) it will **not** work with the current libwebsocket version applied to the package feed.
What about ARM I don't know.
The architecture x86_64 works as expected with the current applied version and the version before your version bump.

karlp, 2020-08-19:
I've filed https://github.com/warmcat/libwebsockets/issues/2008 upstream as their own test app doesn't work on MIPS BE.

karlp, 2020-08-20:
bad news, upstream can't reproduce it on qemu-mips with the Malta machine, and I can't reproduce it on qemu-s390x with alpine to get musl either, so getting hard to work with any other suggestions well received

d1ken, 2021-01-08:
Can you test lws 4.1?
I would like to get lws 4.1 from opkg than compile myself, if it works on MIPS.
4.1 needed because JWT support.