openwrt/packages #13088: mwan3: allow multiple ip addresses in a single rule (Feature Request)

aaronjg, 2020-08-11:
Maintainer: @feckert 
cc @jamesmacwhite @brianjmurrell 

It would be nice to allow mwan3 to have multiple ip addresses in a single rule. This can be handled by creating an ipset for rules with multiple ip addresses.

I'm creating this issue for discussion around the feature. I can implement it.

jamesmacwhite, 2020-08-11:
@aaronjg Just to clarify this one. Is this related to being able to group IP addresses under a single rule, rather than having to have to do multiple `src_ip` rules?

I believe dest_ip supports this already, by using a comma for each IP address, but not in src_ip, so ipset would resolve that one.

aaronjg, 2020-08-11:
@jamesmacwhite

I see, yes you are correct. It is all handled by iptables now, so it should also work for multiple src_ip.  Ipset would be a bit faster than having multiple rules, but probably not worth the change on its own.

What isn't working now is a mixed rule for ipv6 vs ipv4. 

I'll update the title to reflect this.

aaronjg, 2020-08-11:
Actually, I think ipset is the way to handle this. Otherwise you get M*N blow up in the number of rules once you start combining multiple src and destination addresses. This is already an issue in the current implementation of mwan3.

jamesmacwhite, 2020-08-11:
@aaronjg Makes sense. I've been seeing the discussions around this. I think it all stems from @brianjmurrell highlighting the issue when not setting the family value for rules that use a specific family of IP in src_ip https://github.com/openwrt/packages/issues/13003. Given mwan3 has been masking the iptables errors for a while and LuCI doesn't actually support or provide the option itself.

I saw there's some discussion around src_ip and src_ip6 and such, but I guess trying to make it as easy and portable as possible would be best.

I guess the ability to make the family more agnostic is the objective here.

aaronjg, 2020-08-11:
yeah, the ipv6 stuff seems to have been tacked on and I keep finding things that are broken/inconsistent about it. 

I have a fix for it here: 4f0089f7d9ec3e10395aff93ed1b9c8eed187322
Two goals here - one is to make it easier to use the same rules between ipv4 and ipv6, the other is to not have to create an iptables rule for each combination of src/dst ip. Keeping the rules down also makes the firewall rule evaluation faster.