openwrt/packages #13095: Please make available the latest version (a security fix release) of SoftEther VPN server for OpenWRT and Entware

Myron-S, 2020-08-12:
I'm a newbie on this particular bit and it's a bit over my head, so I have a request.   Since the 5.01.9672-1 version there has been a security release..

Can someone please compile the lateest release of [SoftEther](https://github.com/SoftEtherVPN/SoftEtherVPN/releases) for the Entware repository as well as for OpenWRT? I don't know how to do this mysellf.  I asked [here](https://github.com/Entware/Entware/issues/456) and was asked to ask here as it's _upstream specific_ and I need to ask the OpenWRT guys.

The version of SoftEther VPN server at [http://bin.entware.net/x64-k3.2/](http://bin.entware.net/x64-k3.2/) are:

```
softethervpn5-bridge_5.01.9672-1_x64-3.2.ipk       16-Jun-2020 09:38                3623
softethervpn5-client_5.01.9672-1_x64-3.2.ipk       16-Jun-2020 09:38                3505
softethervpn5-libs_5.01.9672-1_x64-3.2.ipk         16-Jun-2020 09:38            11387113
softethervpn5-server_5.01.9672-1_x64-3.2.ipk       16-Jun-2020 09:38                3606
```

The system is a Western Digital DL4100 NAS operating on:
`Linux NAS 3.10.38 #1 SMP Thu Aug 29 20:49:31 PDT 2019 Build-gitcdc15cb x86_64 GNU/Linux`

I has opened a request for this at [https://github.com/Entware/Entware/issues/455](https://github.com/Entware/Entware/issues/455).

neheb, 2020-08-13:
Entware problem. Version 9674 is already available here.

closed 2020-08-13 by neheb

Myron-S, 2020-08-31:
It's how to get this onto the Entware repo.   Who had to do this?