openwrt/packages #13101: ddns-scripts  dinamic dns fro  sitelutions.com update URL changed

u4david, 2020-08-13:
Please change   the updater URL link for sitelution.com ddns provider.

it changed from :
http://www.sitelutions.com

to:
https://dnsup.sitelutions.com

If left unchanged logs will report wrrors like :
ERROR : uclient-fetch Error: '8'
or
Failed to redirect to /dnsup?

neheb, 2020-09-19:
the currect URL is:

```
http://www.sitelutions.com/dnsup?user=[USERNAME]&pass=[PASSWORD]&id=[DOMAIN]&ip=[IP]
```

you're saying to change to

```
https://dnsup.sitelutions.com/dnsup?user=[USERNAME]&pass=[PASSWORD]&id=[DOMAIN]&ip=[IP]
```

?

u4david, 2020-09-21:
Yes the domain is changed to this: https://dnsup.sitelutions.com/

Thx, David V.

neheb, 2020-09-22:
So changing the line I quoted in /etc/ddns/services works?

feckert, 2020-09-22:
I am working on a ddns-scripts rework.
Could we change this after I have merged my[ P/R](https://github.com/openwrt/packages/pull/13402)?

u4david, 2020-09-26:
Right now i have work around by using "custom" provider and:

Https://dnsup.sitelutions.com/dnsup.........
Not sure what P/R is but this change is  not urgent.

Ansuel, 2020-10-21:
@feckert can you do this change?

feckert, 2020-10-22:
@Ansuel done
@u4david I have no account for this so please verify if this works for you.

PR #13739 (2020-10-23): ddns-scripts: update sitelutions url

Maintainer: me / @Ansuel 
Compile tested: not needed only script changes
Run tested: no use case for this only url update

Description:
This fixes #13101
closed 2020-10-23 by feckert