openwrt/packages #13106: procd: 1 sec delay in react for config.change event

vs-1310, 2020-08-13:
Maintainer: @ynezz 
Environment: root@OpenWrt:/# cat /etc/openwrt_release
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='SNAPSHOT'
DISTRIB_REVISION='r13926-f94b098'
DISTRIB_TARGET='malta/le'
DISTRIB_ARCH='mipsel_24kc'
DISTRIB_DESCRIPTION='OpenWrt SNAPSHOT r13926-f94b098'
DISTRIB_TAINTS='no-all'

Description: any service in /etc/init.d/* that expects to receive an event from 
the procd (for example /etc/config/network was changed , network config reload needed) 
receives it with a delay of 1 second

how to reproduce :

1) make a new file for testing our new service:
```
cat > /etc/init.d/procd_ubus_events_test<<EOF
#!/bin/sh /etc/rc.common

USE_PROCD=1
START=90

start_service() {
       echo "Done"
}

reload_service() {
        cat /proc/uptime > /tmp/procd_ubus_event_test.log
}

service_triggers() {
       procd_add_reload_trigger "procd_test"
}
EOF
```
2) start of new service:
```
service procd_ubus_events_test start
```
3) do the test of react for ubus config.change event for procd
(the same events sends by /sbin/reload_config to procd)

```
cat /proc/uptime ; ubus call service event "{ \"type\"
: \"config.change\", \"data\": { \"package\": \"procd_test\" }}" ; sleep 5; cat 
/tmp/procd_ubus_event_test.log
```
result:
77618.34 77536.96
77619.48 77537.95
(reload_service() function from  /etc/init.d/procd_ubus_events_test  was runned by procd with 1 sec delay)

jow-, 2020-08-13:
Set `PROCD_RELOAD_DELAY=0` in your init script. It defaults to 1000ms by default. After applying it to your test script:

```
root@jj:~# cat /proc/uptime ; ubus call service event "{ \"type\": \"config.change\", \"data\": { \"package\": \"procd_test\" }}" ; sleep 5; cat 
/tmp/procd_ubus_event_test.log
330378.52 329616.47
330379.63 329617.46
root@jj:~#
```

closed 2020-08-13 by jow-

vs-1310, 2020-08-14:
> Set `PROCD_RELOAD_DELAY=0` in your init script. It defaults to 1000ms by default. After applying it to your test script:
> 
> ```
> service_triggers() {
>         export PROCD_RELOAD_DELAY=0
>         procd_add_reload_trigger "procd_test"
> }
> ```
> 

THANKS A LOT!!

what was the reason/purpose for adding PROCD_RELOAD_DELAY?

 is it safe to set 0 for all services? (in /lib/functions/procd.sh ?)
 this will speed up the reaction to config changes, but what are the disadvantages?

jow-, 2020-08-14:
> what was the reason/purpose for adding PROCD_RELOAD_DELAY?

Some services react to events that happen in bursts, like interface change events (multiple ifaces brought up at once during boot, cascaded interfaces resulting in events in quick succession when the lowest interface goes down etc.)

Adding a 1 second delay allows all these events to be coalesced into just one service reload since procd will initiate the same trigger action just once for all events received during the PROCD_RELOAD_DELAY time window.

Many init scripts have expensive reload operations (e.g. regeneration of native config from uci through shell scripting) - for such init scripts it is better to keep the reload delay to trigger as little reloads as possible.