openwrt/packages #13122: rtl_433_18.12-1_mipsel_24kc.ipk rtl_433 Segmentation fault (tp-link archer C2) mipsel-24kc

polluz, 2020-08-14:
Hi,

package is http://downloads.openwrt.org/releases/19.07.3/packages/mipsel_24kc/packages/rtl_433_18.12-1_mipsel_24kc.ipk

When launching rtl_433 it gives:

root@c2:~# rtl_433
rtl_433 version unknown
Segmentation fault

dmesg tells:

[  105.786215] do_page_fault(): sending SIGSEGV to rtl_433 for invalid write access to 02400008
[  105.794905] epc = 00401d75 in rtl_433[400000+27000]
[  105.799991] ra  = 00401d7b in rtl_433[400000+27000]
[  274.851388] do_page_fault(): sending SIGSEGV to rtl_433 for invalid write access to 02400008
[  274.860239] epc = 00401d75 in rtl_433[400000+27000]
[  274.865246] ra  = 00401d7b in rtl_433[400000+27000]

Could it be fixed? Evenly could be ported one of the latest stable versions?

Thanks

F

BKPepe, 2020-08-14:
I cherry-picked update for rtl_433 from master branch. See commit: https://github.com/openwrt/packages/commit/bef08d8f7ff6a3156d2ee9300e34dc466311d88e

Could you please compile it yourself or wait until buildbots compile it and try it once again to see if the issue persists even with the latest version?

BKPepe, 2020-08-17:
I see that there is already compiled version 20.02. Don't you mind to try it?

polluz, 2020-08-17:
Sure! Where could i fondi it? 

Tks

F

BKPepe, 2020-08-18:
For example here https://downloads.openwrt.org/releases/19.07.3/packages/mipsel_24kc/packages/rtl_433_20.02-1_mipsel_24kc.ipk

polluz, 2020-08-18:
Trieste with jus do a package upgrade. 

This is the output:

BusyBox v1.30.1 () built-in shell (ash)

  _______                     ________        __
 |       |.-----.-----.-----.|  |  |  |.----.|  |_
 |   -   ||  _  |  -__|     ||  |  |  ||   _||   _|
 |_______||   __|_____|__|__||________||__|  |____|
          |__| W I R E L E S S   F R E E D O M
 -----------------------------------------------------
 OpenWrt 19.07.3, r11063-85e04e9f46
 -----------------------------------------------------
root@c2:~# rtl_433
rtl_433 version unknown inputs file rtl_tcp RTL-SDR
Use -h for usage help and see https://triq.org/ for documentation.
FATAL: low memory? calloc() failed in r_init_cfg() from rtl_433-20.02/src/r_api.c:90
root@c2:~#

Thanks so much.

closed 2020-08-18 by BKPepe

polluz, 2020-08-19:
Hi, BKPepe why did you close this is sue? 
Thanls

BKPepe, 2020-08-19:
Your issue was about segmentation fault, which was fixed as you provided a completely different output and it seems related to low memory. I am not using rtl 433, but you should consult it with help as it is said there.

polluz, 2020-08-19:
Thanks, 

i consult the help but cannot find nothing useful, I've also tried to use the same version of the program on another device (tp-link td-w8970) same platform but different specs. same result.
Also tried to launch rtl_433 (same version) on a ubuntu system and without parameters it starts listening to usb rtl dongle.
I think that the row telling to read documentation was thrown before the FATAL calloc().

:-(

Should i open a new issue?

Thanks!

p.

iddq, 2023-03-04:
Hi @polluz,

Can you please help me what is the solution to low memory error?

Thank you