openwrt/packages #13133: mwan3: mwan3track uses wrong ip address when many intrefaces are present

aaronjg, 2020-08-15:
Maintainer: @feckert 
Environment: 19.07

Description:

When multiple interfaces are present, under certain naming conventions, mwan3 can use the wrong IP address. 

If you have interfaces named, eg "ovpn", "ovpn_2", "ovpn_3, and "ovpn_4" with mwan3 balancing between all of them, then mwan3 will use the ip address of "ovpn_4" for both "ovpn" and "ovpn_4".  This is because mwan3 prefers interfaces ending in "_4" (for ipv4) and "_6" for (ipv6)

https://github.com/openwrt/packages/blob/981d1eb83a3b0676c71559ac8b514c4f8ff56e1b/net/mwan3/files/etc/hotplug.d/iface/15-mwan3#L37

feckert, 2020-11-04:
This was fixed with commit https://github.com/openwrt/packages/commit/bbbc6127abf132a92cc73dd23fa88409dc2b9abd
In the current stable branch owrt-19.07 we have to live with the restriction.
Backporting is not possible without a lot of time.

closed 2020-11-04 by feckert

aaronjg, 2020-11-05:
This is actually a separate issue that arises from the (undocumented) feature that allows you to specify an interface name in mwan3 without the openwrt "_4" or "_6" suffix. Unfortunately, I don't really see a way around this without breaking some existing configurations that rely on this.

Maybe when we have dual stack support, we can eliminate the undocumented behavior, which would eliminate this bug and allow us to remove some complexity from the code base.