openwrt/packages #13145: Can PPTP / L2TP VPN traffic be split through mwan3?

openwrt-diy, 2020-08-16:
Recently, when using mwan3 and PPTP / L2TP functions, we found that VPN traffic could not be assigned to other computer terminals in the LAN! I have tried VPN policy routing, but it is not ideal! Hope to find a solution from mwan3!

aaronjg, 2020-08-17:
At the moment no. mwan3 will create a default route which will take precedence over the split routing functions your VPN provider pushes to you. However, It may be possible to add some mwan3 rules to get the same effect. If that doesn't work, I also have a patch that I am beta testing that will enable this. 

Can you please update your issue tracker with the details form the issue template? Specifically, what version of mwan3 are you on and what version of OpenWRT are you running? Also, please post the output of the mwan3 luci diagnostics page. This will help me figure out exactly what is going on.

openwrt-diy, 2020-08-17:
`Interface status:
 interface wan is online 00h:02m:12s, uptime 00h:59m:27s and tracking is active
 interface wanb is online 00h:02m:09s, uptime 00h:35m:43s and tracking is active
 interface wg1 is offline 00h:00m:00s, uptime 00h:31m:48s and tracking is active
 interface l2tp is offline 00h:00m:00s, uptime 00h:02m:12s and tracking is active

Current ipv4 policies:
balanced:
 wanb (47%)
 wan (52%)
l2tp_only:
 unreachable
wan_only:
 wan (100%)
wanb_only:
 wanb (100%)
wg1_only:
 unreachable

Current ipv6 policies:
balanced:
 unreachable
l2tp_only:
 unreachable
wan_only:
 unreachable
wanb_only:
 unreachable
wg1_only:
 unreachable

Directly connected ipv4 networks:
125.85.190.26
172.26.0.177
192.168.100.0/24
192.168.1.0/24
127.0.0.0/8
10.0.0.1
224.0.0.0/3
139.99.105.5
172.26.0.1
10.0.1.152
10.190.0.0/16

Directly connected ipv6 networks:
fe80::/64
fd5f:42f7:efa3::/64

Active ipv4 user rules:
    0     0 S ip_2_62  all  --  *      *       192.168.100.0/26     0.0.0.0/0            
    0     0 S ip_65_126  all  --  *      *       192.168.100.64/26    0.0.0.0/0            
    0     0 S ip_129_190  all  --  *      *       192.168.100.128/26   0.0.0.0/0            
  436 22629 S ip_193_254  all  --  *      *       192.168.100.192/26   0.0.0.0/0            

Active ipv6 user rules:`

`Software-Version
-------------------------------------------------
OpenWrt - OpenWrt SNAPSHOT r14189-48c3d5c884
LuCI - git-20.219.57394-35e2735

Output of "ip a show"
-------------------------------------------------
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host 
       valid_lft forever preferred_lft forever
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc mq master br-lan state UP group default qlen 1000
[... 364 lines cut ...]
mwan3.l2tp.failure_interval='5'
mwan3.l2tp.recovery_interval='5'
mwan3.l2tp.down='3'
mwan3.l2tp.up='3'
mwan3.l2tp.enabled='1'
mwan3.ip_2_62=rule
mwan3.ip_2_62.src_ip='192.168.100.0/26'
mwan3.ip_2_62.proto='all'
mwan3.ip_2_62.sticky='1'
mwan3.ip_2_62.use_policy='wan_only'
mwan3.ip_65_126=rule
mwan3.ip_65_126.src_ip='192.168.100.64/26'
mwan3.ip_65_126.proto='all'
mwan3.ip_65_126.sticky='1'
mwan3.ip_65_126.use_policy='wanb_only'
mwan3.ip_129_190=rule
mwan3.ip_129_190.src_ip='192.168.100.128/26'
mwan3.ip_129_190.proto='all'
mwan3.ip_129_190.sticky='1'
mwan3.ip_129_190.use_policy='wg1_only'
mwan3.ip_193_254=rule
mwan3.ip_193_254.src_ip='192.168.100.192/26'
mwan3.ip_193_254.proto='all'
mwan3.ip_193_254.sticky='1'
mwan3.ip_193_254.use_policy='l2tp_only'`

openwrt-diy, 2020-08-17:
`Software-Version
-------------------------------------------------
OpenWrt - OpenWrt SNAPSHOT r14189-48c3d5c884
LuCI - git-20.219.57394-35e2735

Output of "ip a show"
-------------------------------------------------
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host 
       valid_lft forever preferred_lft forever
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc mq master br-lan state UP group default qlen 1000
    link/ether 00:e0:67:1c:96:24 brd ff:ff:ff:ff:ff:ff
3: eth1: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc mq master br-lan state DOWN group default qlen 1000
    link/ether 00:e0:67:1c:96:25 brd ff:ff:ff:ff:ff:ff
4: eth2: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc mq master br-lan state DOWN group default qlen 1000
    link/ether 00:e0:67:1c:96:26 brd ff:ff:ff:ff:ff:ff
5: eth3: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc mq master br-lan state DOWN group default qlen 1000
    link/ether 00:e0:67:1c:96:27 brd ff:ff:ff:ff:ff:ff
6: eth4: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc mq state UP group default qlen 1000
    link/ether 00:e0:67:1c:96:28 brd ff:ff:ff:ff:ff:ff
    inet 192.168.1.8/24 brd 192.168.1.255 scope global eth4
       valid_lft forever preferred_lft forever
    inet6 fe80::2e0:67ff:fe1c:9628/64 scope link 
       valid_lft forever preferred_lft forever
7: eth5: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc mq state UP group default qlen 1000
    link/ether 00:e0:67:1c:96:29 brd ff:ff:ff:ff:ff:ff
    inet6 fe80::2e0:67ff:fe1c:9629/64 scope link 
       valid_lft forever preferred_lft forever
8: gre0@NONE: <NOARP> mtu 1476 qdisc noop state DOWN group default qlen 1000
    link/gre 0.0.0.0 brd 0.0.0.0
9: gretap0@NONE: <BROADCAST,MULTICAST> mtu 1462 qdisc noop state DOWN group default qlen 1000
    link/ether 00:00:00:00:00:00 brd ff:ff:ff:ff:ff:ff
10: erspan0@NONE: <BROADCAST,MULTICAST> mtu 1450 qdisc noop state DOWN group default qlen 1000
    link/ether 00:00:00:00:00:00 brd ff:ff:ff:ff:ff:ff
11: br-lan: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue state UP group default qlen 1000
    link/ether 00:e0:67:1c:96:24 brd ff:ff:ff:ff:ff:ff
    inet 192.168.100.1/24 brd 192.168.100.255 scope global br-lan
       valid_lft forever preferred_lft forever
    inet6 fd5f:42f7:efa3::1/60 scope global noprefixroute 
       valid_lft forever preferred_lft forever
    inet6 fe80::2e0:67ff:fe1c:9624/64 scope link 
       valid_lft forever preferred_lft forever
12: pppoe-wanb: <POINTOPOINT,MULTICAST,NOARP,UP,LOWER_UP> mtu 1492 qdisc fq_codel state UNKNOWN group default qlen 3
    link/ppp 
    inet 172.26.0.177 peer 172.26.0.1/32 scope global pppoe-wanb
       valid_lft forever preferred_lft forever
13: wg1: <POINTOPOINT,NOARP,UP,LOWER_UP> mtu 1420 qdisc noqueue state UNKNOWN group default qlen 1000
    link/none 
    inet 10.190.1.121/16 brd 10.190.255.255 scope global wg1
       valid_lft forever preferred_lft forever
25: l2tp-l2tp: <POINTOPOINT,MULTICAST,NOARP,UP,LOWER_UP> mtu 1450 qdisc fq_codel state UNKNOWN group default qlen 3
    link/ppp 
    inet 10.0.1.80 peer 10.0.0.1/32 scope global l2tp-l2tp
       valid_lft forever preferred_lft forever

Output of "ip route show"
-------------------------------------------------
default via 192.168.1.1 dev eth4 proto static src 192.168.1.8 metric 10 
default via 172.26.0.1 dev pppoe-wanb proto static metric 20 
10.0.0.1 dev l2tp-l2tp proto kernel scope link src 10.0.1.80 
10.190.0.0/16 dev wg1 proto static scope link metric 30 
125.85.190.26 via 192.168.1.1 dev eth4 proto static metric 10 
139.99.105.5 via 192.168.1.1 dev eth4 proto static metric 10 
172.26.0.1 dev pppoe-wanb proto kernel scope link src 172.26.0.177 
192.168.1.0/24 dev eth4 proto static scope link metric 10 
192.168.100.0/24 dev br-lan proto kernel scope link src 192.168.100.1

[... 152 lines cut ...]
   15   780 SET        all  --  *      *       0.0.0.0/0            0.0.0.0/0            mark match ! 0xfc00/0xfc00 del-set mwan3_sticky_ip_193_254 src,src
   15   780 SET        all  --  *      *       0.0.0.0/0            0.0.0.0/0            mark match ! 0xfc00/0xfc00 add-set mwan3_sticky_ip_193_254 src,src

Chain mwan3_rule_ip_2_62 (1 references)
 pkts bytes target     prot opt in     out     source               destination         
    0     0 MARK       all  --  *      *       0.0.0.0/0            0.0.0.0/0            mark match 0x0/0x3f00 MARK xset 0x100/0x3f00
    0     0 MARK       all  --  *      *       0.0.0.0/0            0.0.0.0/0            mark match 0x100/0x3f00 ! match-set mwan3_sticky_ip_2_62 src,src MARK and 0xffffc0ff
    0     0 mwan3_policy_wan_only  all  --  *      *       0.0.0.0/0            0.0.0.0/0            mark match 0x0/0x3f00
    0     0 SET        all  --  *      *       0.0.0.0/0            0.0.0.0/0            mark match ! 0xfc00/0xfc00 del-set mwan3_sticky_ip_2_62 src,src
    0     0 SET        all  --  *      *       0.0.0.0/0            0.0.0.0/0            mark match ! 0xfc00/0xfc00 add-set mwan3_sticky_ip_2_62 src,src

Chain mwan3_rule_ip_65_126 (1 references)
 pkts bytes target     prot opt in     out     source               destination         
    0     0 MARK       all  --  *      *       0.0.0.0/0            0.0.0.0/0            mark match 0x0/0x3f00 MARK xset 0x200/0x3f00
[... cut ...]
 pkts bytes target     prot opt in     out     source               destination         
    0     0 mwan3_rule_ip_2_62  all  --  *      *       192.168.100.0/26     0.0.0.0/0            mark match 0x0/0x3f00
    0     0 mwan3_rule_ip_65_126  all  --  *      *       192.168.100.64/26    0.0.0.0/0            mark match 0x0/0x3f00
    0     0 mwan3_rule_ip_129_190  all  --  *      *       192.168.100.128/26   0.0.0.0/0            mark match 0x0/0x3f00
   15   780 mwan3_rule_ip_193_254  all  --  *      *       192.168.100.192/26   0.0.0.0/0            mark match 0x0/0x3f00`

openwrt-diy, 2020-08-17:
`/$ uci show mwan3
mwan3.wan_only=policy
mwan3.wan_only.last_resort='unreachable'
mwan3.wan_only.use_member='wan_m1_w9'
mwan3.wanb_only=policy
mwan3.wanb_only.last_resort='unreachable'
mwan3.wanb_only.use_member='wanb_m1_w8'
mwan3.wg1_only=policy
mwan3.wg1_only.use_member='wg1_m1_w7'
mwan3.wg1_only.last_resort='unreachable'
mwan3.l2tp_only=policy
mwan3.l2tp_only.use_member='t2tp_m1_w6'
mwan3.l2tp_only.last_resort='unreachable'
mwan3.balanced=policy
mwan3.balanced.use_member='wan_m2_w9' 'wanb_m2_w8' 'wg1_m2_w7' 't2tp_m2_w6'
mwan3.balanced.last_resort='unreachable'
mwan3.wan_m1_w9=member
mwan3.wan_m1_w9.interface='wan'
mwan3.wan_m1_w9.metric='1'
mwan3.wan_m1_w9.weight='9'
mwan3.wan_m2_w9=member
mwan3.wan_m2_w9.interface='wan'
mwan3.wan_m2_w9.metric='2'
mwan3.wan_m2_w9.weight='9'
mwan3.wan_m3_w1=member
mwan3.wan_m3_w1.interface='wan'
mwan3.wan_m3_w1.metric='3'
mwan3.wan_m3_w1.weight='1'
mwan3.wanb_m1_w8=member
mwan3.wanb_m1_w8.interface='wanb'
mwan3.wanb_m1_w8.metric='1'
mwan3.wanb_m1_w8.weight='8'
mwan3.wanb_m2_w8=member
mwan3.wanb_m2_w8.interface='wanb'
mwan3.wanb_m2_w8.metric='2'
mwan3.wanb_m2_w8.weight='8'
mwan3.wanb_m3_w1=member
mwan3.wanb_m3_w1.interface='wanb'
mwan3.wanb_m3_w1.metric='3'
mwan3.wanb_m3_w1.weight='1'
mwan3.wg1_m1_w7=member
mwan3.wg1_m1_w7.interface='wg1'
mwan3.wg1_m1_w7.metric='1'
mwan3.wg1_m1_w7.weight='7'
mwan3.wg1_m2_w7=member
mwan3.wg1_m2_w7.interface='wg1'
mwan3.wg1_m2_w7.metric='2'
mwan3.wg1_m2_w7.weight='7'
mwan3.wg1_m3_w1=member
mwan3.wg1_m3_w1.interface='wg1'
mwan3.wg1_m3_w1.metric='3'
mwan3.wg1_m3_w1.weight='1'
mwan3.t2tp_m1_w6=member
mwan3.t2tp_m1_w6.interface='l2tp'
mwan3.t2tp_m1_w6.metric='1'
mwan3.t2tp_m1_w6.weight='6'
mwan3.t2tp_m2_w6=member
mwan3.t2tp_m2_w6.interface='l2tp'
mwan3.t2tp_m2_w6.metric='2'
mwan3.t2tp_m2_w6.weight='6'
mwan3.t2tp_m3_w1=member
mwan3.t2tp_m3_w1.interface='l2tp'
mwan3.t2tp_m3_w1.metric='3'
mwan3.t2tp_m3_w1.weight='1'
mwan3.globals=globals
mwan3.globals.mmx_mask='0x3F00'
mwan3.globals.rtmon_interval='5'
mwan3.wan=interface
mwan3.wan.enabled='1'
mwan3.wan.track_ip='8.8.4.4' '8.8.8.8' '208.67.222.222' '208.67.220.220'
mwan3.wan.family='ipv4'
mwan3.wan.reliability='2'
mwan3.wan.count='1'
mwan3.wan.timeout='2'
mwan3.wan.interval='5'
mwan3.wan.down='3'
mwan3.wan.up='8'
mwan3.wan.initial_state='online'
mwan3.wan.track_method='ping'
mwan3.wan.size='56'
mwan3.wan.max_ttl='60'
mwan3.wan.check_quality='0'
mwan3.wan.failure_interval='5'
mwan3.wan.recovery_interval='5'
mwan3.wanb=interface
mwan3.wanb.track_ip='8.8.4.4' '8.8.8.8' '208.67.222.222' '208.67.220.220'
mwan3.wanb.family='ipv4'
mwan3.wanb.reliability='1'
mwan3.wanb.count='1'
mwan3.wanb.timeout='2'
mwan3.wanb.interval='5'
mwan3.wanb.down='3'
mwan3.wanb.up='8'
mwan3.wanb.initial_state='online'
mwan3.wanb.track_method='ping'
mwan3.wanb.size='56'
mwan3.wanb.max_ttl='60'
mwan3.wanb.check_quality='0'
mwan3.wanb.failure_interval='5'
mwan3.wanb.recovery_interval='5'
mwan3.wanb.enabled='1'
mwan3.wg1=interface
mwan3.wg1.initial_state='online'
mwan3.wg1.family='ipv4'
mwan3.wg1.track_ip='8.8.4.4' '8.8.8.8'
mwan3.wg1.track_method='ping'
mwan3.wg1.reliability='1'
mwan3.wg1.count='1'
mwan3.wg1.size='56'
mwan3.wg1.max_ttl='60'
mwan3.wg1.check_quality='0'
mwan3.wg1.timeout='2'
mwan3.wg1.interval='5'
mwan3.wg1.failure_interval='5'
mwan3.wg1.recovery_interval='5'
mwan3.wg1.down='3'
mwan3.wg1.up='3'
mwan3.wg1.enabled='1'
mwan3.l2tp=interface
mwan3.l2tp.initial_state='online'
mwan3.l2tp.family='ipv4'
mwan3.l2tp.track_ip='8.8.4.4' '8.8.8.8'
mwan3.l2tp.track_method='ping'
mwan3.l2tp.reliability='1'
mwan3.l2tp.count='1'
mwan3.l2tp.size='56'
mwan3.l2tp.max_ttl='60'
mwan3.l2tp.check_quality='0'
mwan3.l2tp.timeout='2'
mwan3.l2tp.interval='5'
mwan3.l2tp.failure_interval='5'
mwan3.l2tp.recovery_interval='5'
mwan3.l2tp.down='3'
mwan3.l2tp.up='3'
mwan3.l2tp.enabled='1'
mwan3.ip_2_62=rule
mwan3.ip_2_62.src_ip='192.168.100.0/26'
mwan3.ip_2_62.proto='all'
mwan3.ip_2_62.sticky='1'
mwan3.ip_2_62.use_policy='wan_only'
mwan3.ip_65_126=rule
mwan3.ip_65_126.src_ip='192.168.100.64/26'
mwan3.ip_65_126.proto='all'
mwan3.ip_65_126.sticky='1'
mwan3.ip_65_126.use_policy='wanb_only'
mwan3.ip_129_190=rule
mwan3.ip_129_190.src_ip='192.168.100.128/26'
mwan3.ip_129_190.proto='all'
mwan3.ip_129_190.sticky='1'
mwan3.ip_129_190.use_policy='wg1_only'
mwan3.ip_193_254=rule
mwan3.ip_193_254.src_ip='192.168.100.192/26'
mwan3.ip_193_254.proto='all'
mwan3.ip_193_254.sticky='1'
mwan3.ip_193_254.use_policy='l2tp_only'`

openwrt-diy, 2020-08-17:
`/$ uci show network
network.loopback=interface
network.loopback.ifname='lo'
network.loopback.proto='static'
network.loopback.ipaddr='127.0.0.1'
network.loopback.netmask='255.0.0.0'
network.globals=globals
network.globals.ula_prefix='fd5f:42f7:efa3::/48'
network.lan=interface
network.lan.type='bridge'
network.lan.proto='static'
network.lan.ipaddr='192.168.100.1'
network.lan.netmask='255.255.255.0'
network.lan.ip6assign='60'
network.lan.ifname='eth0 eth1 eth2 eth3'
network.wan=interface
network.wan.proto='dhcp'
network.wan.ifname='eth4'
network.wan.metric='10'
network.wanb=interface
network.wanb.ifname='eth5'
network.wanb.proto='pppoe'
network.wanb.username='85952540177'
network.wanb.password='9876543210'
network.wanb.ipv6='auto'
network.wanb.metric='20'
network.wg1=interface
network.wg1.proto='wireguard'
network.wg1.private_key='yKvj7LNh7****w2tIeavUh+/NypbJZZ9***SZeMAlg='
network.wg1.addresses='10.190.1.121/16'
network.wg1.metric='30'
network.@wireguard_wg1[0]=wireguard_wg1
network.@wireguard_wg1[0].public_key='Yk1nVzQz5Mm0PR*****kX2uKe8NYRxY='
network.@wireguard_wg1[0].allowed_ips='0.0.0.0/0'
network.@wireguard_wg1[0].endpoint_host='bug.com.wg-sg04.globalssh.xyz'
network.@wireguard_wg1[0].endpoint_port='51820'
network.@wireguard_wg1[0].persistent_keepalive='25'
network.l2tp=interface
network.l2tp.proto='l2tp'
network.l2tp.server='abcde.com'
network.l2tp.username='idwos@dzna'
network.l2tp.password='200300'
network.l2tp.ipv6='auto'
network.l2tp.defaultroute='0'
network.l2tp.metric='40'`

openwrt-diy, 2020-08-17:
`/$ uci show firewall
firewall.@defaults[0]=defaults
firewall.@defaults[0].input='ACCEPT'
firewall.@defaults[0].output='ACCEPT'
firewall.@defaults[0].forward='REJECT'
firewall.@defaults[0].synflood_protect='1'
firewall.@zone[0]=zone
firewall.@zone[0].name='lan'
firewall.@zone[0].input='ACCEPT'
firewall.@zone[0].output='ACCEPT'
firewall.@zone[0].forward='ACCEPT'
firewall.@zone[0].network='lan'
firewall.@zone[1]=zone
firewall.@zone[1].name='wan'
firewall.@zone[1].input='REJECT'
firewall.@zone[1].output='ACCEPT'
firewall.@zone[1].forward='REJECT'
firewall.@zone[1].masq='1'
firewall.@zone[1].mtu_fix='1'
firewall.@zone[1].network='wan wanb wg1 l2tp'
firewall.@forwarding[0]=forwarding
firewall.@forwarding[0].src='lan'
firewall.@forwarding[0].dest='wan'
firewall.@rule[0]=rule
firewall.@rule[0].name='Allow-DHCP-Renew'
firewall.@rule[0].src='wan'
firewall.@rule[0].proto='udp'
firewall.@rule[0].dest_port='68'
firewall.@rule[0].target='ACCEPT'
firewall.@rule[0].family='ipv4'
firewall.@rule[1]=rule
firewall.@rule[1].name='Allow-Ping'
firewall.@rule[1].src='wan'
firewall.@rule[1].proto='icmp'
firewall.@rule[1].icmp_type='echo-request'
firewall.@rule[1].family='ipv4'
firewall.@rule[1].target='ACCEPT'
firewall.@rule[2]=rule
firewall.@rule[2].name='Allow-IGMP'
firewall.@rule[2].src='wan'
firewall.@rule[2].proto='igmp'
firewall.@rule[2].family='ipv4'
firewall.@rule[2].target='ACCEPT'
firewall.@rule[3]=rule
firewall.@rule[3].name='Allow-DHCPv6'
firewall.@rule[3].src='wan'
firewall.@rule[3].proto='udp'
firewall.@rule[3].src_ip='fc00::/6'
firewall.@rule[3].dest_ip='fc00::/6'
firewall.@rule[3].dest_port='546'
firewall.@rule[3].family='ipv6'
firewall.@rule[3].target='ACCEPT'
firewall.@rule[4]=rule
firewall.@rule[4].name='Allow-MLD'
firewall.@rule[4].src='wan'
firewall.@rule[4].proto='icmp'
firewall.@rule[4].src_ip='fe80::/10'
firewall.@rule[4].icmp_type='130/0' '131/0' '132/0' '143/0'
firewall.@rule[4].family='ipv6'
firewall.@rule[4].target='ACCEPT'
firewall.@rule[5]=rule
firewall.@rule[5].name='Allow-ICMPv6-Input'
firewall.@rule[5].src='wan'
firewall.@rule[5].proto='icmp'
firewall.@rule[5].icmp_type='echo-request' 'echo-reply' 'destination-unreachable' 'packet-too-big' 'time-exceeded' 'bad-header' 'unknown-header-type' 'router-solicitation' 'neighbour-solicitation' 'router-advertisement' 'neighbour-advertisement'
firewall.@rule[5].limit='1000/sec'
firewall.@rule[5].family='ipv6'
firewall.@rule[5].target='ACCEPT'
firewall.@rule[6]=rule
firewall.@rule[6].name='Allow-ICMPv6-Forward'
firewall.@rule[6].src='wan'
firewall.@rule[6].dest='*'
firewall.@rule[6].proto='icmp'
firewall.@rule[6].icmp_type='echo-request' 'echo-reply' 'destination-unreachable' 'packet-too-big' 'time-exceeded' 'bad-header' 'unknown-header-type'
firewall.@rule[6].limit='1000/sec'
firewall.@rule[6].family='ipv6'
firewall.@rule[6].target='ACCEPT'
firewall.@rule[7]=rule
firewall.@rule[7].name='Allow-IPSec-ESP'
firewall.@rule[7].src='wan'
firewall.@rule[7].dest='lan'
firewall.@rule[7].proto='esp'
firewall.@rule[7].target='ACCEPT'
firewall.@rule[8]=rule
firewall.@rule[8].name='Allow-ISAKMP'
firewall.@rule[8].src='wan'
firewall.@rule[8].dest='lan'
firewall.@rule[8].dest_port='500'
firewall.@rule[8].proto='udp'
firewall.@rule[8].target='ACCEPT'
firewall.@rule[9]=rule
firewall.@rule[9].name='Support-UDP-Traceroute'
firewall.@rule[9].src='wan'
firewall.@rule[9].dest_port='33434:33689'
firewall.@rule[9].proto='udp'
firewall.@rule[9].family='ipv4'
firewall.@rule[9].target='REJECT'
firewall.@rule[9].enabled='false'
firewall.@include[0]=include
firewall.@include[0].path='/etc/firewall.user'`

openwrt-diy, 2020-08-17:
@feckert @aaronjg

I'm sorry! It took me so long to reply!
I just submitted the configuration parameters of the router on it. Please have time to check it! thank you!

aaronjg, 2020-08-17:
No worries. So mwan3 does not currently work well on the snapshot build. This is at least part of your problem. I am working on a P/R to fix this and will tag you on the PR as soon as it is ready.

There may also be some issues with the split routing - it looks like you are trying to route traffic to 10.0.0.1 through l2tp-l2tp and traffic to 10.190.0.0/16 through wg1? Is that correct?

```
10.0.0.1 dev l2tp-l2tp proto kernel scope link src 10.0.1.80
10.190.0.0/16 dev wg1 proto static scope link metric 30
```

Anyway, getting this working on snapshot is priority #1, and then I can help with the split routing if there are still issues.

aaronjg, 2020-08-17:
It looks like you may have left some confidential info in the post. Can you please delete your penultimate post with the detailed network config?

openwrt-diy, 2020-08-18:
> It looks like you may have left some confidential info in the post. Can you please delete your penultimate post with the detailed network config?

L2TP and WG1 traffic is right! Network account parameters I did some hiding before posting, using some asterisks instead. Now I try to split the route.

aaronjg, 2020-08-19:
Is it correct that the only traffic out of L2TP is for 10.0.0.1 and the only traffic you want out of the wireguard interface is traffic to 10.190.0.0/16?

What are you trying to do with eth4 and pppoe-wanb? Load balance between them? Failover? Something else?

What are you trying to accomplish with all the /26 subnets?

mwan3 currently has issues on the snapshot release, but your configuration also does not look right.

openwrt-diy, 2020-08-20:
> Is it correct that the only traffic out of L2TP is for 10.0.0.1 and the only traffic you want out of the wireguard interface is traffic to 10.190.0.0/16?
>
> What are you trying to do with eth4 and [...]

My thoughts are as follows:
**IP network segment: 192.168.100.2-192.168.100.62 diverted to WAN through MWAN3
IP network segment: 192.168.100.65-192.168.100.126 shunt to WANB through MWAN3
IP network segment: 192.168.100.129-192.168.100.190 shunt to WG1 through MWAN3
IP network segment: 192.168.100.193-192.168.100.254 shunt to L2TP through MWAN3**

openwrt-diy, 2020-08-20:
@aaronjg  Today, I used the official openwrt-19.07-snapshot to recompile the firmware, but I also had some problems!
[image]
[image]
[image]
[image]
[image]
[image]

aaronjg, 2020-08-20:
mwan3 is overkill for what you are trying to do. Just set up routes in the default routing table with appropriate src addresses ranges, and you can get the behavior you need.

openwrt-diy, 2020-08-20:
@aaronjg 
Today, I used openwrt-19.07-snapshot to compile the new firmware. WG1 and L2TP were online and offline repeatedly. The network and firewall were set by default, and there was no change.

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

aaronjg, 2020-08-20:
Yes. mwan3 is currently broken in the snapshot build. If you'd like you can test out the fix here: 
https://github.com/openwrt/packages/pull/13169

This fixed #10712 

However, for the problem you described, mwan3 is not the best solution. If you are just trying to split routes based on src ip, you can do this with just the routing table. mwan3 is only needed if you want to do failover or load balancing. It adds some extra firewall rules, so there is some performance impact associated with using it.

aaronjg, 2020-08-20:
Thanks. Testing and feedback is always appreciated on the P/R. 

If you want to use mwan3, make sure you have default routes for all your WANs in the main routing table.

aaronjg, 2020-09-01:
I saw this was reopened. Short answer is no. There is some issue with L2TP, that causes it not to work with firewall marks on 19.07. This is not an issue with mwan3, but an issue somewhere in an an upstream package. Either, in L2TP, or the kernel itself.

https://github.com/openwrt/packages/issues/13195

It appears to be fixed on the "20.x" snapshot release.

openwrt-diy, 2020-09-01:
> I saw this was reopened. Short answer is no. There is some issue with L2TP, that causes it not to work with firewall marks on 19.07. This is not an issue with mwan3, but an issue somewhere in an an [...]

Sorry, I wanted to fix the error, but then forgot to close it!

openwrt-diy, 2021-01-18:
@aaronjg @jamesmacwhite @feckert
After updating to the stable version and recompiling the functional firmware, the problem is solved! Thank you for your hard work and dedication!

Firmware Version: OpenWrt-19.07.5
Kernel Version: 4.14.209
Software Version: mwan3-2.8.15-1

closed 2021-01-18 by openwrt-diy

jamesmacwhite, 2021-01-18:
Good news, this is likely fixed by the backported SK kernel patches in 19.07.5.