openwrt/packages #13146: mpg123: compile fails, missing headers, should not use cmake

micmac1, 2020-08-16:
Maintainer: @wigyori 
Environment: master

Description:

Hi all,

Since 9fc2785 there are problems with mpg123:

1. missing headers

```
make[8]: Entering directory '/builder/shared-workdir/build/sdk/build_dir/target-x86_64_musl/freeswitch-1.10.4.-release/src/mod/formats/mod_shout'
  CC       mod_shout_la-mod_shout.lo
In file included from mod_shout.c:36:
/builder/shared-workdir/build/sdk/staging_dir/target-x86_64_musl/usr/include/mpg123.h:12:10: fatal error: fmt123.h: No such file or directory
 #include <fmt123.h>
          ^~~~~~~~~~
compilation terminated.
Makefile:712: recipe for target 'mod_shout_la-mod_shout.lo' failed
```

Full log [here](https://downloads.openwrt.org/snapshots/faillogs/x86_64/telephony/freeswitch/compile.txt).

Mind you this is a header that mpg123.h references :) This is funny. Its own stuff is missing.

2. missing soname libs

Usually one would expect shared libs with soname info in their name, like in the past:

```
/usr/lib/libmpg123.so.0.44.10
/usr/lib/libout123.so.0.2.2
```

Not anymore. Now we only get this:

```
/usr/lib/libout123.so
/usr/lib/libmpg123.so
```

This is not good if one choses to use ABI_VERSION. It's not good in general.

Seeing all this I had my doubts that upstream considers cmake the way to go, so searched upstream site and found this bug report: [link](https://sourceforge.net/p/mpg123/bugs/290/)

Quote: "Though I want to stress that **the cmake port is not intended to replace the autotools build for Linux**. It's purpose is mainly to provide portability for esoteric build environments like MSVC and it's there because someone contributed it, like other ports."

If this had been run-tested this would've been detected ;-)

Regards,
Seb

micmac1, 2020-08-16:
Regarding compile fails, there's also build failures of mpg123 for aarch64 or neon-targets it seems.

neheb, 2020-08-17:
hrm locally I'm testing fixing CMake to work properly. My patches are pretty big currently. Going back to autotools is the easiest solution though. Oh well.

closed 2020-08-19 by neheb